espressif/arduino-esp32 #2478: Serial1.read() not functional

AkhileshThorat, 2019-02-19:
Make your question, not a Statement, inclusive. Include all pertinent information:

What you are trying to do?
Describe your system( Hardware, computer, O/S, core version, environment).
Describe what is failing.
Show the shortest possible code that will duplicate the error.
Show the EXACT error message(it doesn't work is not enough).
All of this work on your part shows us that you have worked to solve YOUR problem. The more complete your issue posting is, the more likely someone will volunteer their time to help you.

If you have a Guru Meditation Error or Backtrace, ***please decode it***:
https://github.com/me-no-dev/EspExceptionDecoder

----------------------------- Remove above -----------------------------

### Hardware:
Board:							?ESP32 Dev Module? ?node32? ?ttgo_lora?
Core Installation version:			?1.0.0? ?1.0.1-rc4? ?1.0.1? ?1.0.1-git?
IDE name:						?Arduino IDE? ?Platform.io? ?IDF component?
Flash Frequency:					?40Mhz?
PSRAM enabled:                                    ?no? ?yes?
Upload Speed:					?115200?
Computer OS:                                        ?Windows 10? ?Mac OSX? ?Ubuntu?

### Description:
Describe your problem here

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

//Change the code below by your sketch
#include <Arduino.h>

void setup() {
}

void loop() {
}
```

### Debug Messages:
```
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
```

stickbreaker, 2019-02-19:
Pins 6-12 are used by flash memory. Can't be used by app.

AkhileshThorat, 2019-02-19:
> 
> 
> Pins 6-12 are used by flash memory. Can't be used by app.

I have used Dual Input Output(DIO) as flash mode,that makes pin 9 and 10 available for other .pins 9,10 are used by flash when QIO(Quad I/O) is enabled. I have used pins 9 and 10 in micropython for serial communication. this should work in arduino too, but I'm clueless whether it is about Hardware abstraction layer for arduino or some other stuff

stickbreaker, 2019-02-19:
Did you physically disconnect the Flash from those lines?

AkhileshThorat, 2019-02-19:
> 
> 
> Did you physically disconnect the Flash from those lines?

No, I haven't.
Earlier I have tested code on a custom made Micropython (https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo),without disconnecting flash from those serial lines,it works fine.

nikisalli, 2019-05-19:
@AkhileshThorat any news on this? i'm having the same probem. it looks like the rx line beheaves as an output thus having low impedance. i tried manually setting it as an input but it didn't work

AkhileshThorat, 2019-05-20:
> 
> 
> @AkhileshThorat any news on this? i'm having the same probem. it looks like the rx line beheaves as an output thus having low impedance. i tried manually setting it as an input but it didn't work

@nikisalli  Espressif recommends not to use pins 6 -11 for any other purpose/applications.
They're connected to SPI flash memory onboard and may result in guru meditation error if used. It's better you use other pins for your purpose, you can use any digital pin as serial(refer pin map of arduino-esp32).
If you're running out of GPIO's it's better to wait till new ESP32-S2 is available in market. It has Improved MCU functionalities and maybe some extra GPIO's for folks (who used external MCU for other purposes).

nikisalli, 2019-05-20:
the problem is that i have a pcb with an uart module connected to pins 9 and 10 and i'd like prefer to use it instead of ordering new pcbs :/

AkhileshThorat, 2019-05-20:
> 
> 
> the problem is that i have a pcb with an uart module connected to pins 9 and 10 and i'd like prefer to use it instead of ordering new pcbs :/

It's very uncertain that ESP32 won't throw error if you use 9,10 in a specific way. Serial works on 9 and 10 but only when used in very restrictive manner. like don't use it too often(keep atleast 15 seconds interval in between communication ). Always check for programming mode(it should be DIO and not QIO, QIO uses pin 9 and 10 for uploading code). Reset your ESP32 whenever you find your application is not running anymore. 
Next time keep a habit of verifying functional behavior before making a PCB.

nikisalli, 2019-05-20:
ok i'll make a new one then

AkhileshThorat, 2019-08-01:
There were many controversies surrounding using reserved IO pins , I think such pins should not be made accessible outside the module . Such pins should be kept inside the module only for memory interfacing or other interfaces.

closed 2019-08-01 by AkhileshThorat

manuprendlair, 2019-08-28:
Hello, just a little comment on this issue.
I just got this problem (panic core) when using esp32 wroom uart1 for the first time.
And when searching on google I found many people facing this issue!

why?

- Firstly because we found on Espressif document that gpio 9 and 10 are referenced for uart1.
- secondly because default arduino serial1 use these pins!

I found this thread very instructive about this problem: https://www.esp32.com/viewtopic.php?t=1250

So yes if you put flash mode option to DIO, you could use these pin.

Finally, I think it's not a good things that default value for serial1 or 2 generate a panic core ( QIO mode is the default flash mode for ESP32 dev module).
 Initially serial1 was note using these pin by defaulI. I found on an issue (but not able to find it again) that somebody ask to change pin to default pin use in Espressif pinout. Me-no-dev change the default pins to 9 10 (it was logic until you got this issue).

So I suggest that default pin was changed to pin that not expose to a core panic depending the flas config mode.

Regards.

Emmanuel FIANCETTE
France.