espressif/arduino-esp32 #2479: 'class WiFiClass' has no member named 'hostname'

GeorgeFlorian, 2019-02-19:
Hello ! I just copy/paste this [example](https://github.com/me-no-dev/ESPAsyncWebServer/blob/master/examples/simple_server/simple_server.ino)
This is what I get from compiling.

```
Arduino: 1.8.8 (Linux), Board: "ESP32 Dev Module, Disabled, Default, 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 115200, None"

/home/george/New Volume/Arduino Sketches/foo2/foo2.ino: In function 'void setup()':
foo2:37:25: error: 'class WiFiClass' has no member named 'hostname'
     Serial.println(WiFi.hostname());
                         ^
Multiple libraries were found for "WiFi.h"
 Used: /home/george/Arduino/hardware/espressif/esp32/libraries/WiFi
 Not used: /opt/arduino-1.8.8/libraries/WiFi
exit status 1
'class WiFiClass' has no member named 'hostname'

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
```

This is the code:

```
//
// A simple server implementation showing how to:
//  * serve static messages
//  * read GET and POST parameters
//  * handle missing pages / 404s
//

#include <Arduino.h>
#include <WiFi.h>
#include <AsyncTCP.h>
#include <ESPAsyncWebServer.h>

AsyncWebServer server(80);

const char* ssid = "YOUR_SSID";
const char* password = "YOUR_PASSWORD";

const char* PARAM_MESSAGE = "message";

void notFound(AsyncWebServerRequest *request) {
    request->send(404, "text/plain", "Not found");
}

void setup() {

    Serial.begin(115200);
    WiFi.mode(WIFI_STA);
    WiFi.begin(ssid, password);
    if (WiFi.waitForConnectResult() != WL_CONNECTED) {
        Serial.printf("WiFi Failed!\n");
        return;
    }

    Serial.print("IP Address: ");
    Serial.println(WiFi.localIP());
    Serial.print("Hostname: ");
    Serial.println(WiFi.hostname());

    server.on("/", HTTP_GET, [](AsyncWebServerRequest *request){
        request->send(200, "text/plain", "Hello, world");
    });

    // Send a GET request to <IP>/get?message=<message>
    server.on("/get", HTTP_GET, [] (AsyncWebServerRequest *request) {
        String message;
        if (request->hasParam(PARAM_MESSAGE)) {
            message = request->getParam(PARAM_MESSAGE)->value();
        } else {
            message = "No message sent";
        }
        request->send(200, "text/plain", "Hello, GET: " + message);
    });

    // Send a POST request to <IP>/post with a form field message set to <message>
    server.on("/post", HTTP_POST, [](AsyncWebServerRequest *request){
        String message;
        if (request->hasParam(PARAM_MESSAGE, true)) {
            message = request->getParam(PARAM_MESSAGE, true)->value();
        } else {
            message = "No message sent";
        }
        request->send(200, "text/plain", "Hello, POST: " + message);
    });

    server.onNotFound(notFound);

    server.begin();
}
void loop() {}
```

atanisoft, 2019-02-19:
it appears the example is broken, you should replace WiFi.hostname() with WiFi.getHostname() when compiling on the esp32.

GeorgeFlorian, 2019-02-19:
> it appears the example is broken, you should replace WiFi.hostname() with WiFi.getHostname() when compiling on the esp32.

Yea ! This fixed it !

But how should I have known this ? Where does it say ?

How can I not make a similar mistake in the future ? Because there isn't much documentation for ESP32, and the one I found uses syntax for ESP8266 and it's basically useless for a new guy like me.

atanisoft, 2019-02-19:
it should be updated in the example you have copied from. I see you already opened an issue there as well, I'd suggest close this one and track it there instead.

closed 2019-02-19 by GeorgeFlorian