espressif/arduino-esp32 #2483: get NTP server from DHCP

tomtobback, 2019-02-21:
Hi, i am trying to have my ESP32s receive an NTP server from DHCP on my OpenWRT router. I am testing this using the native time functions as in the ESP32>Time>SimpleTime example.
When i use the router's IP address as "ntpServer" i get the router's time as expected. 
I understand that as part of DHCP, the router can also provide an NTP server address to the ESP32. LWIP seems to offer this feature, it is mentioned in https://github.com/espressif/arduino-esp32/blob/master/tools/sdk/include/lwip/lwip/opt.h line875:
```
/**
 * LWIP_DHCP_GETS_NTP==1: Request NTP servers with discover/select. For each
 * response packet, an callback is called, which has to be provided by the port:
 * void dhcp_set_ntp_servers(u8_t num_ntp_servers, ip_addr_t* ntp_server_addrs);
*/
#if !defined LWIP_DHCP_GET_NTP_SRV || defined __DOXYGEN__
#define LWIP_DHCP_GET_NTP_SRV           0
#endif
```
When i set this LWIP_DHCP_GET_NTP_SRV to 1 and compile, i expect it to work but it seems it does not ('failed to obtain time'). I have looked around for this function 'dhcp_set_ntp_servers' but cannot find it, except in https://github.com/espressif/arduino-esp32/blob/master/tools/sdk/include/lwip/lwip/dhcp.h 
```
#if LWIP_DHCP_GET_NTP_SRV
/** This function must exist, in other to add offered NTP servers to
 * the NTP (or SNTP) engine.
 * See LWIP_DHCP_MAX_NTP_SERVERS */
extern void dhcp_set_ntp_servers(u8_t num_ntp_servers, const ip4_addr_t* ntp_server_addrs);
#endif /* LWIP_DHCP_GET_NTP_SRV */
```
But that does not seem to define the function.
[i am using the Arduino IDE]
Has anyone tried this, or any advice?

me-no-dev, 2019-02-21:
LwIP is precompiled with the said options when IDF libs are updated. If you want to change such settings, you need to go the "Arduino as IDF component" route :) Instructions in the readme

tomtobback, 2019-02-25:
OK thanks for the reply. I was hoping there was another way, as the changing the LWIP_DHCP_GET_NTP_SRV setting in that opt.h file does make a difference in my sketch (Serial.print(LWIP_DHCP_GET_NTP_SRV)) but i guess it does not activate the necessary functions.
IDF is not an option for me at the moment, so i think the conclusion is this functionality is not implemented yet in arduino-esp32. 
My application needs to use the NTP server sent by DHCP, or, if not available, default to one named in the sketch. 
I'm still confused that some lwip files are read during Arduino compiling (e.g. sntp.h and sntp_opts.h) and others not (e.g. dhcp.h). 
I'm looking at the configTime() function in https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal-time.c and would like to use the sntp_servermode_dhcp() function there (defined in sntp.h), but it gives 'undefined reference' when compiling.

lbernstone, 2019-02-25:
The header files are all used in your compile.  However, the .c is already compiled, so if that code uses a macro defined in the header, it will use the earlier value.  If you really want this to work in Arduino IDE, you can copy the files (header and .c) from esp-idf into your sketch folder and modify them there.  This will replace the definition in the .a libraries. [Arduino as esp component](https://github.com/espressif/arduino-esp32/blob/master/docs/esp-idf_component.md) is actually easier than that for LWIP.

tomtobback, 2019-02-26:
thanks for your suggestion @lbernstone i found your [repo](https://github.com/lbernstone/sntp_opts) and used that sntp.c 
Now i can use those functions in esp32-hal-time.c 
However `sntp_servermode_dhcp(1);` does not seem to have any effect. As far as i understand it should enable function `dhcp_set_ntp_servers` to be called from dhcp.c but for some reason it does not.

lephisto, 2019-05-08:
Is there any Progress? Besides the NTP Server I would like to get Custom DHCP Options for preconfiguration purposes. This is not possible with the Current LWIP Version ..

lbernstone, 2019-05-09:
Get the [sntp.c from esp-idf](https://github.com/espressif/esp-idf/blob/master/components/lwip/apps/sntp/sntp.c). Put it in your sketch folder.  Add a line with `#define SNTP_GET_SERVERS_FROM_DHCP 1`.
LWIP has no code to get vendor options, so you are out of luck on that.

tomtobback, 2019-05-17:
thanks @lbernstone for the suggestion; i have tried that with an updated EPS32 core for Arduino (latest stable version via boards manager) but starting from the SimpleTime example, i get the following compilation errors:
_sntp.h: No such file or directory_
so i changed the `#include "sntp.h" ` in sntp.c to `#include "lwip/apps/sntp.h" `
next error:
_unknown type name 'sntp_sync_mode_t'_
any suggestion how to fix that?

lbernstone, 2019-05-17:
You need the sntp.c that matches your arduino-esp32 version.  For 1.0.2, that is IDF v3.2.  https://github.com/espressif/esp-lwip/blob/fe9a58cb7abdc74e7a2196b7c4e0a29ff9042f88/src/apps/sntp/sntp.c

tomtobback, 2019-05-20:
thanks again @lbernstone  now i get this compilation error:
_sntp.c:625: error: expected identifier or '(' before '{' token_
see screenshot below, maybe it means that function is defined elsewhere already? 
that new sntp.c looks very similar to the one i found in your repo (as mentioned above): that did compile until i upgraded my core to 1.0.2 using the board manager rather than git. now it gives the same error.

[image]

lbernstone, 2019-05-20:
That is clearly not the code from the link above.  Look at how the lines do not match up. https://github.com/espressif/esp-lwip/blob/fe9a58cb7abdc74e7a2196b7c4e0a29ff9042f88/src/apps/sntp/sntp.c#L625

tomtobback, 2019-05-20:
OK it compiles now, but still not clear if `#define SNTP_GET_SERVERS_FROM_DHCP 1` makes a difference. I've verified via `nmap` that my router is providing the DHCP option 42 for NTP servers, and I've switched on `SNTP_DEBUG` but it doesn't mention receiving anything from DHCP. I'm not entirely sure how it should work, as `configTime()` needs an ntpServer as argument.

lbernstone, 2019-05-20:
I can't get my dhcp server to properly provide this option to test, but I would guess that `Serial.println(ipaddr_ntoa(sntp_getserver(0)))` should return an address. configTime is an arduino thing, so it is going to assume default behavior.  You are going to have to go straight to esp-idf functions to make this functionality work.  All this workaround is why me-no-dev and I recommended using arduino as an esp-idf component.

closed 2019-08-14 by stale[bot]

tve, 2019-10-11:
I'm hitting this same issue: I also would like to get the NTP servers from DHCP. Question: is this not supported in arduino-esp32 for a specific reason or simply because no-one has had time& energy to implement it?

dingo35, 2023-07-08:
To all who hit this topic: I got this (ntp-server through dhcp option 42) working on an ESP32:
-first update your toolchain; pio --upgrade, but also a git pull in my ~/.platformio/ framework directory was necessary.
-in the initialization of your wifi, before you bring it up, add:
`sntp_servermode_dhcp(1);                                                    //try to get the ntp server from dhcp
sntp_setservername(1, "europe.pool.ntp.org");                               //fallback server
sntp_init();
`
...and your esp32 will get the ntp server from DHCP, and fallback on the internet server if it fails...