espressif/arduino-esp32 #2485: Reading from file and notifying on BLE simultaneosly

rado99, 2019-02-21:
Hello everyone,
I am trying to send data over BLE via notifications while reading a file, but I found,that I receive only the last 20 bytes from the reading. I think I need to give time to BT to send the message, but I don't know how.
Thank you in advance.
Here is a snippet from my code:
```
char record[30];
char buffer[20];
  while(file.available()){
  
    for(i=0;i<8;i++){
    record[i]=file.read();
    if(count==18){
      pCharacteristic->setValue(buffer);
      pCharacteristic->notify();
      Serial.println("I am notifying");
      delay(5);
      count=0;
      }
    buffer[count]=record[i];
    count++;
    if(record[i]==compareFrom[i])
    match++;
    }
    if(match==8){
      hadRecord++;
      for(int j=8;j<30;j++){
        record[j]=file.read();
        if(count==18){
      pCharacteristic->setValue(buffer);
      pCharacteristic->notify();
      Serial.println("I am notifying");
      delay(5);
      count=0;
      }
    buffer[count]=record[j];
    count++;
      }
}
```
Here is the last debug message:
`[D][BLECharacteristic.cpp:631] setValue(): >> setValue: length=14, data=33392c2039353830312e30312ebe, characteristic UUID=beb5483e-36e1-4688-b7f5-ea07361b26b4
[D][BLECharacteristic.cpp:638] setValue(): << setValue
[D][BLECharacteristic.cpp:484] notify(): >> notify: length: 14
[V][GeneralUtils.cpp:305] hexDump():      00 01 02 03 04 05 06 07 08 09 0a 0b 0c 0d 0e 0f
[V][GeneralUtils.cpp:306] hexDump():      -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
[V][GeneralUtils.cpp:332] hexDump(): 0000 33 39 2c 20 39 35 38 30 31 2e 30 31 2e be        39, 95801.01..
[D][BLECharacteristic.cpp:533] notify(): << notify
[D][BLECharacteristic.cpp:459] handleGATTServerEvent(): << handleGATTServerEvent
[D][BLECharacteristic.cpp:203] handleGATTServerEvent(): >> handleGATTServerEvent: Unknown
`

rado99, 2019-02-26:
I retried to fix it without using any pointers and reading the file in the Callback function, but without success. Any suggestions?

closed 2019-08-15 by stale[bot]

storerjeremy, 2020-05-13:
@rado99 did you ever figure out what the solution was? Having the same problem at the moment

chegewara, 2020-05-14:
I think @tobozo have some success with this matter.

tobozo, 2020-05-14:
@chegewara actually it was a great help using one of your modified versions of BLE library 

[BLE File Transfer](https://github.com/tobozo/ESP32-BLECollector/blob/master/BLEFileSharing.h)
[Modified BLE Library](https://github.com/tobozo/ESP32-BLECollector/releases/download/1.3/BLE.zip)