espressif/arduino-esp32 #2486: Guru Meditation with wifi.h

tomxygen, 2019-02-21:
### Hardware:
Board:							ESP32 Dev Module
Core Installation version:			1.0.1
IDE name:						Arduino IDE
Flash Frequency:					80Mhz
PSRAM enabled:                                    no
Upload Speed:					115200?
Computer OS:                                        MacOS

I have a project based on the ESP32 and a DHT11 sensor that reads the temperature and uploads it to a webpage.
Unfortunately the wifi.h seems to be the responsible for the crash of the board.
This is the error shown in the serial display:

`[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 2 - STA_START
[V][esp32-hal-i2c.c:1442] i2cInit(): num=0 sda=27 scl=26 freq=0
[V][esp32-hal-i2c.c:1643] i2cSetFrequency(): Fifo threshold=3
[V][esp32-hal-i2c.c:1643] i2cSetFrequency(): Fifo threshold=15
Connecting to NETGEAR_11ng
Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed)
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40146d40: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40146d40: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40146d40: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40146d40: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40146d40: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40146d40: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40146d40: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40146d40: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40146d40: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40146d40: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40146d40: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40146d40: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40146d40: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40146d40: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40146d40: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40146d40: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40146d40: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40146d40: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40146d40: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40146d40: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40146d40: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40146d40: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40146d40: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40146d40: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstrucGuru Meditation Error: Core  1 panic'ed (Unhandled debug exception)
Debug exception reason: Stack canary watchpoint triggered () 
[... 347 lines cut ...]
            client.print(String((intTemperature-32)*5/9) + "&deg;C");
            Serial.print("Temperature on the server");
            Serial.println(intTemperature);
            client.print("<br>");
            client.print("<br>");
            client.print("<h1>Current humidity: </h1>" + String(int(intHumidity)) + "%");
            // The HTTP response ends with another blank line:
            client.println();
            // break out of the while loop:
            break;
          } else {    // if you got a newline, then clear currentLine:
            currentLine = "";
          }
        } else if (c != '\r') {  // if you got anything else but a carriage return character,
          currentLine += c;      // add it to the end of the currentLine
        }

      }
    }
    // close the connection:
    client.stop();
    Serial.println("Client Disconnected.");
  }
  
}`

jestebanes, 2019-03-01:
Same trouble here, but with different code

atanisoft, 2019-03-01:
The original poster issue doesn't appear to have anything to do specifically with wifi. The first error in the stack is pointing to an ISR doing something it shouldn't be doing which flash is being accessed. Most likely it is in the DHT or RH_ASK library that is being compiled into the app. The SSD1306 library uses I2C and from the include name looks like the same library I use and it doesn't use an ISR internally.

Try commenting the driver.init() line and see if the issue goes away. If so, open an issue against that library. Similarly for the DHT library..

tomxygen, 2019-03-01:
@atanisoft thank you for spotting out the issue.
Indeed, removing the driver.init() in my code, the issue goes away, while removing dht.begin() does't fix it.

atanisoft, 2019-03-01:
@tomxygen I'd suggest closing this issue and opening a new one for the maintainer of the RH_ASK library (not even sure what that is!).

tomxygen, 2019-03-01:
@atanisoft I’ll close the issue soon.
RH_ASK is a sub-library of the RadioHead library that deals with wireless communication.
ASK stands for radio amplitude shifting, and I use it with this module: 
https://proxy.duckduckgo.com/iu/?u=https%3A%2F%2Ftse2.mm.bing.net%2Fth%3Fid%3DOIP.uw29VZi4mRZhD27-ldb3aAHaE6%26pid%3D15.1&f=1

atanisoft, 2019-03-01:
from what I can read of the RH_ASK library, it is not specifically written to support the ESP32 (though it has support for the esp8266) so it will need to be updated to support the ESP32.

closed 2019-08-15 by stale[bot]