espressif/arduino-esp32 #2487: two bme280 in serie (i2c, ESP32)

jaccr, 2019-02-21:
re:
Board:							ESP32 Dev Module
IDE name:						Arduino IDE
Flash Frequency:					80Mhz
PSRAM enabled:                                      no
Upload Speed:					921600
Computer OS:                                        Windows 7

### Description:
I want connect two devices (bme280) with I2C in ESP32. But when I looked for information, I found a scaner program to know the devices address. I have two devices in serie connect in I2C bus and if I will run scan program it just found one device and anyone is broken. Would can I do to solve this problem?

#include <Wire.h>

void scan(){
Serial.println(" Scanning I2C Addresses");
uint8_t cnt=0;
for(uint8_t i=0;i<128;i++){
  Wire.beginTransmission(i);
  uint8_t ec=Wire.endTransmission(true);
  if(ec==0){
    if(i<16)Serial.print('0');
    Serial.print(i,HEX);
    cnt++;
  }
  else Serial.print("..");
  Serial.print(' ');
  if ((i&0x0f)==0x0f)Serial.println();
  }
Serial.print("Scan Completed, ");
Serial.print(cnt);
Serial.println(" I2C Devices found.");
}

bool i2cReady(uint8_t adr){
uint32_t timeout=millis();
bool ready=false;
while((millis()-timeout<100)&&(!ready)){
  Wire.beginTransmission(adr);
  ready=(Wire.endTransmission()==0);
  }
return ready;
}

void eepromSize(){
Serial.println("Discovering eeprom sizes 0x50..0x57");
uint8_t adr=0x50,i;
uint16_t size;
char buf[64];
while(adr<0x58){
  i=0;
  size = 0x1000; // Start at 4k
  i += sprintf_P(&buf[i],PSTR("0x%02X: "),adr);
  if(i2cReady(adr)) { // EEPROM answered
    uint8_t zeroByte;
    Wire.beginTransmission(adr);
    Wire.write((uint8_t)0); // set address ptr to 0, two bytes High
    Wire.write((uint8_t)0); // set address ptr to 0, two bytes Low
    uint8_t err=Wire.endTransmission();
    if(err==0){// worked
      err=Wire.requestFrom(adr,(uint8_t)1);
      if(err==1){// got the value of the byte at address 0
        zeroByte=Wire.read();
        uint8_t saveByte,testByte;
        do{
          if(i2cReady(adr)){
            Wire.beginTransmission(adr);
            Wire.write(highByte(size)); // set next test address
            Wire.write(lowByte(size));
            Wire.endTransmission();
            err=Wire.requestFrom(adr,(uint8_t)1);
            if(err==1){
              saveByte=Wire.read();
              Wire.beginTransmission(adr);
              Wire.write(highByte(size)); // set next test address
              Wire.write(lowByte(size));
              Wire.write((uint8_t)~zeroByte); // change it
              err=Wire.endTransmission();
              if(err==0){ // changed it
                if(!i2cReady(adr)){
                  i+=sprintf_P(&buf[i],PSTR(" notReady2.\n"));
                  Serial.print(buf);
                  adr++;
                  break;
                  }
                Wire.beginTransmission(adr);
                Wire.write((uint8_t)0); // address 0 byte High
                Wire.write((uint8_t)0); // address 0 byte Low
                err=Wire.endTransmission();
                if(err==0){
                  err=Wire.requestFrom(adr,(uint8_t)1);
                  if(err==1){ // now compare it
                    testByte=Wire.read();
                    }
                  else {
                    testByte=~zeroByte; // error out
                    }
                  }
                else {
                  testByte=~zeroByte;
                  }
                }
              else {
                testByte = ~zeroByte;
                }
              //restore byte
              if(!i2cReady(adr)){
                i+=sprintf_P(&buf[i],PSTR(" notReady4.\n"));
                Serial.print(buf);
                adr++;
                break;
                }
              
              Wire.beginTransmission(adr);
              Wire.write(highByte(size)); // set next test address
              Wire.write(lowByte(size));
              Wire.write((uint8_t)saveByte); // restore it
              Wire.endTransmission();
              }
            else testByte=~zeroByte;
            }
          else testByte=~zeroByte;
          if(testByte==zeroByte){
            size = size <<1;
            }
          }while((testByte==zeroByte)&&(size>0));
        if(size==0) i += sprintf_P(&buf[i],PSTR("64k Bytes"));
        else i+=sprintf_P(&buf[i],PSTR("%dk Bytes"),size/1024);
        if(!i2cReady(adr)){
          i+=sprintf_P(&buf[i],PSTR(" notReady3.\n"));
          Serial.print(buf);
          adr++;
          continue;
          }
        Wire.beginTransmission(adr);
        Wire.write((uint8_t)0); // set address ptr to 0, two bytes High
        Wire.write((uint8_t)0); // set address ptr to 0, two bytes Low
        Wire.write(zeroByte);  //Restore
        err=Wire.endTransmission();
        }
      else i+=sprintf_P(&buf[i],PSTR("Read 0 Failure"));
      }
    else i+=sprintf_P(&buf[i],PSTR("Write Adr 0 Failure"));
      
    }
  else i+=sprintf_P(&buf[i],PSTR("Not Present."));
  Serial.println(buf);
  adr++;
  }
}

void setup(){
Serial.begin(9600);
Wire.begin();
scan();
Serial.println();
eepromSize();
}

void loop(){}

WereCatf, 2019-02-21:
> I want connect two devices (bme280) with I2C in ESP32. But when I looked for information, I found a scaner program to know the devices address. I have two devices in serie connect in I2C bus and if I will run scan program it just found one device and anyone is broken.

You need to change one or the other device's I2C-address, otherwise they will both use the same address. On most modules, you change the address by modifying the solder-bridges on the board.

jaccr, 2019-02-26:
If I change I2C-address in a bme280 code  (one device in 0x76 and other in 0x77) just work device what have 0x76 I2c-address:

#include <dummy.h>
#include "cactus_io_BME280_I2C.h"
#include <Wire.h>

BME280_I2C bmea(0x76);
BME280_I2C bmeb(0x78);

void setup()
{
  Serial.begin(9600);
  if(!bmea.begin()){
    Serial.println("Device error!");
    while(1);
  }
  else
    Serial.print("bmea ok");
   if(!bmeb.begin()){
    Serial.println("Device error!");
    while(1);
  } 
  Serial.print("bmeb ok");
}

void loop()
{
  float pressure;
  Serial.println("Primer sensor");
 bmea.readSensor();
  //get and print temperatures
  Serial.print("Temp: ");
  Serial.print(bmea.getTemperature_C());
  Serial.println("C");//The unit for  Celsius because original arduino don't support special symbols
  
  //get and print atmospheric pressure data
  Serial.print("Pressure: ");
  Serial.print(pressure = bmea.getPressure_HP());
  Serial.println("HPa");

  //get and print altitude data
  Serial.print("Altitude: ");
 // Serial.print(bmea.calcAltitude(pressure));
  Serial.println("m");

  //get and print humidity data
  Serial.print("Humidity: ");
  Serial.print(bmea.getHumidity());
  Serial.println("%");
////////////////////////////////////////

  Serial.println("Segundo sensor");
  bmeb.readSensor();
  //get and print temperatures
  Serial.print("Temp: ");
  Serial.print(bmeb.getTemperature_C());
  Serial.println("C");//The unit for  Celsius because original arduino don't support special symbols
  
  //get and print atmospheric pressure data
  Serial.print("Pressure: ");
  Serial.print(pressure = bmeb.getPressure_HP ());
  Serial.println("HPa");

  //get and print altitude data
  Serial.print("Altitude: ");
  //Serial.print(bmeb.calcAltitude(pressure));
  Serial.println("m");

  //get and print humidity data
  Serial.print("Humidity: ");
  Serial.print(bmeb.getHumidity());
  Serial.println("%");

  delay(3000);
}

stickbreaker, 2019-02-26:
@jaccr you have to select the address at the sensor.  The specifications are on the bottom of page 32 of the datasheet [BME280 Datasheet](https://ae-bst.resource.bosch.com/media/_tech/media/datasheets/BST-BME280-DS002.pdf).  You have to do a hardware change.

Chuck.

jaccr, 2019-02-27:
Yes, it was that. Thank you.

closed 2019-02-27 by jaccr