espressif/arduino-esp32 #2488: Port "handle empty uri" fix from ESP8622 platform to ESP32 one

FWdeveloper, 2019-02-21:
### Hardware:
Board:							ESP32 Dev Module
Core Installation version:			        1.0.1
IDE name:						Platform.io
Flash Frequency:					80Mhz
PSRAM enabled:                                       no
Upload Speed:					        921600
Computer OS:                                          Windows 7

### Description:
Please port "handle empty uri" fix from ESP8622 platform to ESP32 one:

[https://github.com/esp8266/Arduino/commit/db1cfc77729134a9b5b74047255663c7bd97ac29#diff-15a79083ae1917043eb1d24feccca784](https://github.com/esp8266/Arduino/commit/db1cfc77729134a9b5b74047255663c7bd97ac29#diff-15a79083ae1917043eb1d24feccca784)

### Debug Messages:

Current:

```
POST  HTTP/1.1
Host: 192.168.1.64:39324
User-Agent: ESP32HTTPClient
Connection: keep-alive
Accept-Encoding: identity;q=1,chunked;q=0.1,*;q=0
Content-Type: application/json
Content-Length: 62
[V][HTTPClient.cpp:1074] handleHeaderResponse(): RX: 'HTTP/1.1 400 Bad Request'
```

Expected:

```
POST / HTTP/1.1
Host: 192.168.1.64:39324
User-Agent: ESP32HTTPClient
Connection: keep-alive
Accept-Encoding: identity;q=1,chunked;q=0.1,*;q=0
Content-Type: application/json
Content-Length: 62
[V][HTTPClient.cpp:1076] handleHeaderResponse(): RX: 'HTTP/1.1 200 OK'
```

me-no-dev, 2019-02-21:
Is empty URI in HTTP specs?

FWdeveloper, 2019-02-21:
[https://www.w3.org/Protocols/rfc2616/rfc2616-sec5.html#sec5.1.2](https://www.w3.org/Protocols/rfc2616/rfc2616-sec5.html#sec5.1.2)

> Note that the absolute path cannot be empty; if none is present in the original URI, it MUST be given as "/" (the server root).

FWdeveloper, 2019-03-25:
Hello,

Is there any progress to solve this issue?

stickbreaker, 2019-03-25:
@FWdeveloper The specification you quote:
>Note that the absolute path cannot be empty; if none is present in the original URI, it MUST be given as "/" (the server root).

applies to the client side.  It describes a valid request format.  if the absolute path is empty, it is not a valid request, so the server should respond with a 400 error.

Most current browsers automatically append the root `/` when you just enter a host name. 

But, there are valid cases where no uri is sent to the host:
>       Request-URI    = "*" | absoluteURI | abs_path | authority
>
>The four options for Request-URI are dependent on the nature of the request. The asterisk "*" means that the request does not apply to a particular resource, but to the server itself, and is only allowed when the method used does not necessarily apply to a resource. One example would be
>
>       OPTIONS * HTTP/1.1

Chuck.

closed 2019-03-25 by FWdeveloper