espressif/arduino-esp32 #2492: ESP32-CAM & OVH2640 CamWebServer port forwarding issue

rainalert, 2019-02-22:
### Hardware:
Board:							ESP32-CAM
Core Installation version:			1.0.1
IDE name:						Arduino IDE

I have compile and upload the CamWebServer skecth to ESP32 modul without problem. Application works perfect and I can easily connect modul on port 80 and 81 perfectly in LAN. In order to access to modul form outside of the LAN I make port forwarding. I have forwarded port numbers 8000->80 and 8001->81. When I access to the modul from outside of the LAN (via http://x.x.x.x:8000) html code of the control panel loaded successfully. However i can not access the stream page. When i invesigate the loaded html code stream adress is shown as http://x.x.x.x:8000:81 not http://x.x.x.x:8001 in the html code. If I try http://x.x.x.x:8001/stream i can access the stream without the control panel.

me-no-dev, 2019-02-22:
the panel expects the stream to be on external port 81 and not 8001 :) change the mapping and it will work

closed 2019-02-22 by me-no-dev

robynjayqueerie, 2019-06-05:
@me-no-dev I feel you have been a bit hasty in closing this issue. Similar to @rainalert , I have been trying to access the streaming from outside the LAN, my issue is that I want to set up several of these cameras within the LAN, then access them from outside and the best way is to assign them all different ports.
Like @rainalert i notice that the html page which is generated is locked to port 81. What I can't work out is where that html code is generated and what I can do to change that.
Are you suggesting in your response that port 81 is fixed for streaming and we should just mod the mapping in the project to suit that or are you suggesting there is a way of modifying the project so that the very good and useful control panel page uses a streaming port related to the server publish port.
Do you have any suggestion as to how to modify the original steaming on the control page to do what we want? 
Thanks for your time
Robyn

tuannobick, 2019-08-18:
@me-no-dev  In app_httpd.cpp(Example of ESP32/CameraWebServer) line 648-648 you can change the port like: 
........................
    config.server_port = 353; 
    config.ctrl_port = 353;

In my case, it's work.

LeSamiral, 2020-03-03:
I know this discussion is old, but it can still be useful to someone.
If your code works locally but you only have the web page but not the stream from outside you probably have this issue:
the HTML page say to the browser "hey look! you can find the video stream here http://192.168.x.x:81/stream" correct your code to send your public IP address.

In my case, that was the problem.

chupocro, 2021-04-01:
What I did was I moved the stream server to port 80 by registering stream URL among other URLs with:

`httpd_register_uri_handler(stream_httpd, &stream_uri);`

Then I commented out the code for starting the stream server on port 81:

```
/*
if (httpd_start(&stream_httpd, &config) == ESP_OK) {
    httpd_register_uri_handler(stream_httpd, &stream_uri);
 }
*/
```

If you want to after moving the stream server to port 80 still use the control panel you have to edit the HTML served by the ESP32 which is in:

`const uint8_t index_ov2640_html_gz[] = { ... }`

inside file **camera_index.h**

Use [Cyberchef](https://gchq.github.io/CyberChef/) to decode the byte array into HTML, recipe =

```
From_Hex('Auto')
Gunzip()
```

change the line:

`var streamUrl = baseHost + ':81'`

to

`var streamUrl = baseHost`

and convert the HTML back into the byte array, recipe =

```
Gzip('Dynamic Huffman Coding','','',false)
To_Hex('0x with comma',16)
```

The last thing is to count the bytes end to edit:

`#define index_ov2640_html_gz_len`

and now everything (control panel + still image + stream) works on port 80.

In the same way you can use any other port too - just change the default port, register stream URL among other URLs, remove the code for starting the stream server on another port and remove concatenation of the string `':81'` from HTML code served by ESP32.

BTW, I did that so I can use CameraWebServer with **ngrok** service (running on Raspberry Pi and tunneling traffic towards ESP32 CAM) which works without port forwarding. That way I can use only one tunnel for both still images and video stream and I can still use the control panel.

Anc2, 2021-12-19:
hi please how can i count the bytes end to edit?

chupocro, 2021-12-29:
> hi please how can i count the bytes end to edit?

I misspelled the work "and", it should be "**and** to edit".

You can just copy/paste the hex values in some text editor with line numbers and then multiply the number of rull rows with the number of bytes in each row and then add the number of bytes in the last row or you can enable **raw** view in Cyberchef and read the length in bytes.

JohnnyBaciu, 2025-01-07:
having similar issues. I want all URLs to be on port 3000. i set the config.server_port to be port 3000. The webpage loads but the stream functionality is not working. I have done everything mentioned above: 
I have moved the stream register_uri_handler with the other rigister_uri_handlers
commented out the place where it is initialized on the other port.
I have updated the decoded camera_index array and removed the + "81". 

still doesnt seem to work. ever since i have changed ports, the wepage and still frame capture work a lot slower.