espressif/arduino-esp32 #2494: WiFiClientSecure gets stuck, when send buffers get full

everslick, 2019-02-22:
When we send too fast (faster then the TLS connection allows), eventually the send buffers will fill up and write() returns 0 (indicating, that nothing could be written). Normally this means we just retry until all data is sent, but in this case the TLS socket never recovers from this condition (buffers stay full). The only thing we can do is close the socket and go home.

everslick, 2019-08-01:
still valid, don't close.

everslick, 2019-08-01:
https://github.com/esp8266/Arduino/pull/6104

could be the same bug as on ESP8266. @me-no-dev ?

everslick, 2019-09-30:
still relevant, don't close.

everslick, 2020-02-12:
https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFiClientSecure/src/WiFiClientSecure.cpp#L183

My guess is, when `send_ssl_data()` returns an error we silently close the socket, and do not inform the caller. write should return int, but returns size_t.

everslick, 2020-02-12:
@me-no-dev what do you think?

closed 2020-04-26 by stale[bot]

everslick, 2021-02-17:
Doing more tests with idf-release/v4.2 shows the bug still present. So this could very easily be a issue with WiFiClientSecure. It hink this is serious and worth looking into before next stable release. What can I do to bring this foward besides the what I commented above?

sansillusion, 2021-02-17:
wow I feel bad for you, not even a single aknowledgment of you issue and you have to fight the stale bot to keep it open. I hope you get the attention you need on this issue, unfortunately I can't help.