espressif/arduino-esp32 #2495: ESP32 and BMX055 Wifi i2c

Hellakoe, 2019-02-22:
Hello,

I use the ESP32 to read out the positionsensor Bmx055 from Bosch via i2c, process it and show it on the web server.
The data processing works until I comment in Wifi.softAp(). After that only zeros are shown and i2c doesnt work anymore.
I have already tried the help of @stickbreaker  and added his files to the Esp32 file. Unfortunately it doesnt work.
I also do not get any debug messages.
Do you have any idea? 
I would be so grateful. 
----------------------------- Remove above -----------------------------

### Hardware:
Board:							ESP32 Dev Module
Core Installation/update date:			222/2/19
IDE name:							Arduino IDE
Flash Frequency:					40Mhz
PSRAM enabled:                                          no
Upload Speed:						115200
Computer OS:                                            Windows 10

### Description:
Describe your problem here

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

//Change the code below by your sketch
// Distributed with a free-will license.
// Use it any way you want, profit or free, provided it fits in the licenses of its associated works.
// BMX055
// This code is designed to work with the BMX055_I2CS I2C Mini Module available from ControlEverything.com.
// https://www.controleverything.com/products
#include <Wire.h>
#include <WiFi.h>
const char* ssid     = "ESP32";
const char* password = "123456";// offenes netzwerk,wenn auskommentiert
WiFiServer server(80);
// BMX055 Accl I2C address is 0x18(24)
#define Addr_Accl 0x19
// BMX055 Gyro I2C address is 0x68(104)
#define Addr_Gyro 0x69
// BMX055 Mag I2C address is 0x10(16)
#define Addr_Mag 0x13
int xAccl =0; 
int yAccl =0;
int zAccl =1000;
int xGyro =0;
int yGyro =0;
int zGyro =0;
int zAcclold =0;
int yAcclold =0;
int xAcclold =0;
int xGyroold =0;
int yGyroold =0;
int zGyroold =0;
String header;
//const int output26 = 5;
String MoPinState = "off";
const int output26 = 26; //das war eigentlich unser buzzer für web
int u=0; //damit bei knopfauslösung auch unendlich läuft
//Inhalt: Körner der Sanduhr, 5000ms nur für test! sonst zv = 0
volatile int zv = 0;
//Um den Timer zu konfigurieren, benötigen wir einen Zeiger auf eine Variable vom Typ hw_timer_t
hw_timer_t * timer = NULL;
//Interrupt-Serviceroutine, Körner der Sanduhr werden im Sekundentakt weniger
void IRAM_ATTR onTimer() {
  if(zv > 0)
  {
    zv--;
  }
}
[... 488 lines cut ...]
  
   }
   
   
    // Clear the header variable
    header = "";
    // Close the connection
    client.stop();
    Serial.println("Client disconnected.");
    Serial.println("");
  }
  
  
  //Sanduhrabfrage, je nach Körneranzahl Buzzer aus oder an
  if (zv > 0) 
  {
    digitalWrite(output26, HIGH); // turn buzzer on
  } 
  
  else 
  {
    digitalWrite(output26, LOW); // turn buzzer off
  }
  
  }

stickbreaker, 2019-02-23:
Are you sure you want a binary 'OR' `|` here:
```c++
Serial.println("Gyrokonvertierung Ende");
//Alarmausgabe
 if ((xAccldiff < -350| xAccldiff > 350) || (yAccldiff < -350|| yAccldiff > 350) || (zAccldiff < -450|| zAccldiff > 450)  || (yGyrodiff < -30000|| yGyrodiff > 30000))
   {
```
Why do you read one byte at a time?
```c++
  for (int i = 0; i < 6; i++)
  {
    // Start I2C Transmission
    Wire.beginTransmission(Addr_Mag);
    // Select data register
    Wire.write((66 + i));
    // Stop I2C Transmission
    Wire.endTransmission();
    // Request 1 byte of data
    Wire.requestFrom(Addr_Mag, 1);
 
    // Read 6 bytes of data
    // xMag lsb, xMag msb, yMag lsb, yMag msb, zMag lsb, zMag msb
    if (Wire.available() == 1)
    {
      data[i] = Wire.read();
    }
  }
```
I would rewrite it to this:
```c++
    // Start I2C Transmission
    Wire.beginTransmission(Addr_Mag);
    // Select data register
    Wire.write((66));
    // Stop I2C Transmission
    Wire.endTransmission();
    If(Wire.lastError == I2C_ERROR_OK){ // device answered, accepted data
       Wire.requestFrom(Addr_Mag, 6);
       If (Wire.lastError !=I2C_ERROR_OK){// problem reading data
          Serial.printf("Problem reading Mag data err=%d(%s)\n",Wire.lastError(),Wire.getErrorText(Wire.lastError()));
       }
      else { // good data
         for(uint8_t i = 0;a <6; a++){
           data[i] = Wire.read();
         }
     }
   }
   else { // Problem setting address pointer
       Serial.printf(" Problem communicating with sensor at 0x%02X err=%d(%s)\n",Addr,Mag, Wire.lastError(), Wire.getErrorText(Wire.lastError()));
  }
```

You state you are having problems with I2C communications, but nowhere in you code do you check the I2C error returns?

I would test the error return after every `Wire.endTransmission()`, and `Wire.requestFrom()`.  Try to find where the errors are occurring.  Without knowing the errors and locations there is nothing you can do to solve the problems.

Chuck.

Hellakoe, 2019-02-24:
Hello chuck, 

thank you so much for your help. 

In this line I get the debug message: Invalid use of non-static member function 
> If(Wire.lastError == I2C_ERROR_OK){ // device answered, accepted data

Do you know what could be the reason? 

> Are you sure you want a binary 'OR' `|` here:
yes it was a mistake. it actually should be `||` instead of `|`.  
This means if This or this or this ... happens do this. Right? 

Hella

lbernstone, 2019-02-24:
Wire.lastError() is a function.  You are using it as a variable.
Please note, this is not a forum for assistance with your code.  It is for reporting errors in the arduino-esp32 codebase.  In the future, please trim down your code to a minimal sketch that demonstrates the issue.  A little effort on your part will make it much more likely that you get assistance.

Hellakoe, 2019-02-24:
> Wire.lastError() is a function. You are using it as a variable.
thanks for the hint. 

> Please note, this is not a forum for assistance with your code. It is for reporting errors in the arduino-esp32 codebase
I am a beginner at programming and new on github. didnt know this is a excluding forum only for professional. 

> A little effort on your part will make it much more likely that you get assistance.
I will know it better for the future.

Hellakoe, 2019-02-24:
hmm okay... Thanks for the hint.

I am a beginner at programming and new on github. Will know it for the future.

Hellakoe, 2019-03-04:
@stickbreaker 
I tried to apply your advice and try to solve the problem with a simple source code. Once I comment in `WiFi.softAP (ssid, password);`, the board will not find i2c devices anymore. before that it worked perfectly. I have inserted the shortened text above. So I have found out  that my problem with the web server and i2c is, right?
kind regards Hella

lbernstone, 2019-03-04:
The only interaction b/w the WiFi and i2c is power.  Your power supply is not delivering enough current to drive all your requirements.

stickbreaker, 2019-05-19:
@Hellakoe 
Have you solved you issues?

closed 2019-08-14 by stale[bot]