espressif/arduino-esp32 #2496: Reset button after power up.

cyberrocker, 2019-02-23:
Hello,
I am a first time poster, so i am not sure if this is the correct forum for this issue. Please advise me if it should be somewhere else. I did search for this problem but could not find an easy to understand anser.

I have an ESP32 with wifi code, but after powering up the board to run the code, I need to hit the reset button manually to actually start running. The board will be in a box which will make it impossible for users to reset.

Can some one help with this issue?

Many thanks

closed 2019-08-15 by stale[bot]

JHershey69, 2020-04-08:
I wish this was open again because I have the same question.

lbernstone, 2020-04-08:
This is either a power or a code problem.  If you actually want help, open a new issue and follow the issue template.