espressif/arduino-esp32 #2497: How do I save and read array values with preferences?

xman4242, 2019-02-24:
I would like to understand how to use this library to write arrays to the memory and to read them.
I am using this library with a Heltec WifiKit 32.
I found this thread, however, I am still struggling to understand. 
https://github.com/espressif/arduino-esp32/issues/611
Thanks,
xman4242

boarchuz, 2019-02-24:
If you just want to save and load the whole array, you can simply putbytes("myArray", &myArray, sizeof(myArray));

lbernstone, 2019-02-24:
This will require the the [modification I just submitted](https://github.com/espressif/arduino-esp32/pull/2498), but shows how to cast data into a struct array.
```
#include <Preferences.h>
Preferences prefs;

typedef struct {
  uint8_t hour;
  uint8_t minute;
  uint8_t setting1;
  uint8_t setting2;
} schedule_t;

void setup() {
  Serial.begin(115200);
  prefs.begin("schedule");
  uint8_t content[] = {9, 30, 235, 255, 20, 15, 0, 0};
  prefs.putBytes("schedule", content, 8);
  size_t schLen = prefs.getBytes("schedule", NULL, NULL);
  char buffer[schLen];
  prefs.getBytes("schedule", buffer, schLen);
  if (schLen % sizeof(schedule_t)) {
    log_e("Data is not correct size!");
    return;
  }
  schedule_t *schedule = (schedule_t *) buffer;
  Serial.printf("%02d:%02d %d/%d\n", 
    schedule[1].hour, schedule[1].minute,
    schedule[1].setting1, schedule[1].setting2);
  prefs.putBytes("schedule", schedule, sizeof(schedule));
  buffer[0] = 0;
  prefs.getBytes("schedule", buffer, schLen);
  for (int x=0; x<8; x++) Serial.printf("%02X ", buffer[x]);
  Serial.println(); 
}

void loop() {}
```

xman4242, 2019-02-25:
Thank you all. I think I have it figured out now!

lbernstone, 2019-02-25:
Please close the issue

xman4242, 2019-02-25:
My apologies. It is closed now.

closed 2019-02-25 by xman4242

bastian2001, 2022-12-19:
@guilhermeaiolfi booleans are stored as complete bytes (chars or unsigned chars, idk exactly). If you need to store hundreds of booleans, you could squish them down to bits but otherwise I'd say just store the whole bytes.