espressif/arduino-esp32 #2499: WebServer.send_P cannot handle larger files

WonderWhyWilli, 2019-02-24:
### Hardware:
Board: ESP32 Dev Module (ESP32-WROOM-32D)
Core Installation version: 1.0.1
IDE name: Arduino IDE
Flash Frequency: 80Mhz
PSRAM enabled: no
Upload Speed: OTA
Computer OS: Windows 10

### Description:
Hey, I am trying to send larger files via the WebServer.send_P function,
like chart.min.js, ~160kB, saved as `static const char JS_CHART[] PROGMEM = R"###(***)###";`

But only 46-52kByte get sent each time according to the web browser, resulting in a debug message of the ESP32:
`[D][WiFiClient.cpp:463] connected(): Disconnected: RES: 0, ERR: 128`
and an error message in the developer tools of web browsers:
`GET http://***/js/Chart.min.js net::ERR_CONTENT_LENGTH_MISMATCH`
(The Content-length of the response header matches the size of `JS_CHART[]`)

If I send it in 40kB chunks (so WebServer.send_P and 3x WebServer.sendContent_P) it works every time.

I hope you can help me. Thank you very much in advance!

florian-gh, 2019-03-10:
I had the same problem, and found an error in WebServer.h: the "write()" method can write less than the requested bytes (this is standard behaviour). Here are my changes, although there may be better ways to correct the problem. I'll try to make a pull request...

```
  virtual size_t _currentClientWrite(const char* b, size_t l) {
    size_t totalWritten = 0;
    size_t written;
    do {
      written = _currentClient.write( b + totalWritten, l - totalWritten );
      if (written > 0) {
        totalWritten += written;
      }
    } while (written != -1 && totalWritten < l);
    return totalWritten;
  }
  virtual size_t _currentClientWrite_P(PGM_P b, size_t l) {
    size_t totalWritten = 0;
    size_t written;
    do {
      written = _currentClient.write_P( (PGM_P)(((char *)b) + totalWritten), l - totalWritten );
      if (written > 0) {
        totalWritten += written;
      }
    } while (written != -1 && totalWritten < l);
    return totalWritten;
  }

```

Thorinair, 2019-03-27:
Thank you!! I was struggling with this issue myself. I was trying to send the minified jquery as one of the responses, as I am having my ESP32 run in AP mode, meaning it has no internet. I have spent around 6 hours yesterday trying to find a fix and I finally found this solution today, worked absolutely flawlessly!

WonderWhyWilli, 2019-04-05:
Thank you very much, your solution helped a lot!

closed 2019-04-05 by WonderWhyWilli

vniclos, 2019-07-28:
Thank's great help!