espressif/arduino-esp32 #2503: float to String conversion problem using String concat

rafi021, 2019-02-25:
Dear concern,
I have faced some problem during float to String conversion process. Here is the code

float v=220.0,i=0.25,p=49.5,e=52,f=50.0,pf=0.90,Q=23.97,S=55.0;

String prepare_server_packet()
{ 
   String rf_data;
   char sep = ';';
   rf_data+= String("S")+String(Card_ID);
   rf_data+= sep +String("M")+String(Meter_ID);
   rf_data+= sep +String("V")+String(v);
   rf_data+= sep +String("I")+String(i);
   rf_data+= sep +String("P")+String(p);
   rf_data+= sep +String("S")+String(S);
   rf_data+= sep +String("Q")+String(Q);
   rf_data+= sep +String("E")+String(e);
   rf_data+= sep +String("f")+String(f);
   rf_data+= sep +String("pf")+String(pf);
   rf_data+= sep +String("B")+String(balance);
   rf_data+= sep +String("1_L")+String(digitalRead(Load1_Pin));
   rf_data+= sep +String("2_L")+String(digitalRead(Load2_Pin));
   rf_data+= sep +String("3_L")+String(digitalRead(Load3_Pin)); 
   Serial.println(rf_data);  // display data packet format
   return rf_data; 
}

But unfortunately after first circle data conversion the float values, in second circle its gives me zero values. Debug Message
1st Circle: S19912345678910021;M2;V220.00;I0.25;P49.50;S55.00;Q23.97;E52.00;f50.00;pf0.90;B250.00;1_L0;2_L1;3_L0
2nd Circle:
S19912345678910021;M2;V0.00;I0.00;P0.00;S0.00;Q0.00;E0.00;f0.00;pf803967664128.00;B250.00;1_L0;2_L1;3_L0

I couldn't found  where is the problem. Could you please re-check the WString.cpp and WString.h files for proper variable conversion.

lbernstone, 2019-02-25:
Use char* and printf instead.  Modifying a string over and over (and then passing it between scopes), uses a lot of memory.
```
char rf_data[120];
snprintf(rf_data, 120, "S%ull;M%d;V%3.2f...", Card_ID, Meter_ID, v);
return (String)rf_data;
```

rafi021, 2019-02-25:
Hi lbernstone,
I gave a try as you suggested. But device starts rebooting after 1st circle. Here is the debug message:

S⸮⸮⸮?;M1;V220.00;I0.25;P49.50;S55.00;Q23.97;E52.00;f50.00;pf0.90;B250.00;1_L0;2_L1;3_L
Stack smashing protect failure!

abort() was called at PC 0x400ef7e0 on core 1

Backtrace: 0x4008cfd4:0x3ffb1e90 0x4008d205:0x3ffb1eb0 0x400ef7e0:0x3ffb1ed0 0x400dcc55:0x3ffb1ef0 0x400d39f7:0x3ffb1f40 0x400d4750:0x3ffb1f90 0x400deefd:0x3ffb1fb0 0x4008f265:0x3ffb1fd0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:928
ho 0 tail 12 room 4
load:0x40078000,len:9280
load:0x40080400,len:5848
entry 0x40080698

stickbreaker, 2019-02-25:
@rafi021 The error message is telling you that the stack has been overwritten.

I think the return parameter in @lbernstone's example:
```c++
char rf_data[120];
snprintf(rf_data, 120, "S%ull;M%d;V%3.2f...", Card_ID, Meter_ID, v);
return (String)rf_data;
```
Should be changed to:
```c++
return String(rf_data);
```
I think the direct type cast is the source of the stack overwrite.

Although, I would change your coding to completely remove `Strings`.  I am leery of them because they allocate dynamic memory from Heap, and I am never sure when then release it.  The Memory thrashing/fragmentation side effects have been responsible for many program upsets and data dependent reboots.

Chuck.

rafi021, 2019-02-25:
Thanks a lot for your suggestions. I have tested my code on Arduino Mega. Its run without any problem. Here is compile message:  Chip -> ESP32 with 4MB flash 

**Sketch uses 861766 bytes (65%) of program storage space. Maximum is 1310720 bytes.
Global variables use 48236 bytes (14%) of dynamic memory, leaving 279444 bytes for local variables. Maximum is 327680 bytes.**

Only 14% dynamic memory is used for global variables. I am using OTA+Web Server+WifiMulti+EEPROM +RadioHead libraries. However, i have used String as variable in only two places of my whole code. One inside this function and another one as a global variable to store this message and pass it via RF module. 

I guess may be WString.cpp and WString.h files need some modification. May be its related to String function related buffer or memory management of ESP32 release version.  

Thanks a lot. If you could find any solution for that. Please let me know.

chegewara, 2019-02-25:
`Stack smashing protect failure!`
This line is suggesting that you have problem with stack, not with heap memory. Maybe this will help you to understand it and solve it:

closed 2019-08-15 by stale[bot]