espressif/arduino-esp32 #2505: ESP32 Invalid Static IP Address

marciodias2003, 2019-02-25:
### Hardware:
Board:							ESP32 Dev Module
Core Installation version:			1.0.1
IDE name:						Atmel Studio
Flash Frequency:					80Mhz
PSRAM enabled:                                     no
Upload Speed:					9261600
Computer OS:                                         Windows 10

### Description:
I'm trying to setup a static ip address for ESP32, but I realized that even though I was setting an invalid IP address, it accepts and I do not have any tools to know if I'm actually connected and operational on the network.
My network is in the range 192.168.1.X but if I enter an address in the range 10.1.1.X for example, ESP fires the normal events as if was connected to the network.

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

#include <WiFi.h>

const char* ssid     = "your_network_name";
const char* password = "your_network_password";

IPAddress local_IP(192, 168, 1, 115); //OK
//IPAddress local_IP(10, 1, 1, 115); // Wrong
IPAddress gateway(192, 168, 1, 1); // OK
//IPAddress gateway(10, 1, 1, 1); // Wrong
IPAddress subnet(255, 255, 255, 0);
IPAddress primaryDNS(8, 8, 8, 8); //optional
IPAddress secondaryDNS(8, 8, 4, 4); //optional

void setup()
{
  Serial.begin(115200);

  if (!WiFi.config(local_IP, gateway, subnet, primaryDNS, secondaryDNS)) {
    Serial.println("STA Failed to configure");
  }

  Serial.print("Connecting to ");
  Serial.println(ssid);

  WiFi.begin(ssid, password);

  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }

  Serial.println("");
  Serial.println("WiFi connected!");
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());
  Serial.print("ESP Mac Address: ");
  Serial.println(WiFi.macAddress());
  Serial.print("Subnet Mask: ");
  Serial.println(WiFi.subnetMask());
  Serial.print("Gateway IP: ");
  Serial.println(WiFi.gatewayIP());
  Serial.print("DNS: ");
  Serial.println(WiFi.dnsIP());
}

void loop()
{

}
```

stickbreaker, 2019-02-25:
You are not describing an error. Just because your choice of IP address is non functional does not invalidate the ip.

The wifi stack is operating correctly you just assigned a non route-able address.  If you want it to function you will have to assign a route-able address. That matches the defined network.

Chuck.

marciodias2003, 2019-02-25:
So maybe my question should be:
How to find out if current the static IP address is route-able?

stickbreaker, 2019-02-25:
@marciodias2003 you can't, Static address are by definition network admin assigned.  You have to use DHCP if you are not in control of the network.  The DHCP server will reassign your device the same IP 'if' it remembers your device, and it does not currently have the IP assigned to another device.

mDns is an answer for random IP assignments.  you can use mDns to advertise a fixed name that can be associated with your random IP, but, mDns was designed for small, local networks.  the Host name is limited to `{name}.local` and is only visible on the local network subnet.  The discovery protocol is unroute-able  IP multicast. So, your mDns host name is not visible outside the local subnet.

If you need global visibility you will have to acquire your local DHCP issued IP, then use a Dynamic DNS service like [NoIp](www.noip.com/free) to publish/ link your global name to your temporary IP. 

Chuck.

closed 2019-08-15 by stale[bot]