espressif/arduino-esp32 #2507: Missing datas on HardwareSerial reception

nicolas-rabault, 2019-02-25:
### Hardware:
Board:							ESP32 Dev Module
Core Installation/update date:			1.0.1
IDE name:							Arduino IDE
Flash Frequency:					80Mhz
PSRAM enabled:                                          no
Upload Speed:						921600
Computer OS:                                            Windows, Mac OSX, Ubuntu

### Description:
I'm working on a project with ESP32 on Arduino IDE to create something like a bidirectional pipe between HardwareSerial Serial2@1000000baud and WebSocket or BluetoothSPP. The code also manage connection stuff (WebServer, captive portal, mdns,...) to allow users to easily configure a wifi connection and name its ESP.
You can find my source code on this github : https://github.com/Luos-Robotics/ESP_Luos_pipe

Everything is kind of working, but I have a lot of trouble and strange crash with this code. 
One of them is data loss on reception on Serial2.
In my setup I have a host controler sending Json using serial and the ESP should get those data and send them using Wifi or Bluetooth (or drop it if nobody is connected)
Just to be sure I use a logic analyser to check messages from host integrity, I check 5000 messages and there are all OK.
Here is some normal messages : 

```
{"modules":' '{"gate":' '{},' '"dxl_6":' '{"temperature":' '26.00}}}
{"modules":' '{"gate":' '{},' '"dxl_6":' '{"rot_position":' '45.31,' '"temperature":' '26.00}}}
{"modules":' '{"gate":' '{},' '"dxl_6":' '{}}}

```
If I print messages received from esp I have some of them wrong for example : 

```
{"modules": {"gate": {}, "dxl_6": {"rot{"modules": {"gate": {}, "dxl_6": {"rot{"modules": {"gate": {}, "dxl_6": {}}}
modules": {"gate": {}, "dxl_6": {}}}
{"modules": {"gate": {}, "dxl{"modules": {"gate": {}, "dxl_6": {"rot_position": 45.31}}}
```

This is like sometime the serial reception is disabled or preempted by something else.

Do you already have experienced something similar with ESP?
Do you have any solution to get all those data?

### Sketch:
```cpp
#include <Arduino.h>

void setup() {
    // Setup serial
    Serial2.begin(1000000, SERIAL_8N1, 16, 17);    //Baud rate, parity mode, RX, TX
    Serial.begin(115200);
    delay(100);
    // Start host stream
    Serial2.print("{\"detection\": {}}\r");
}

void loop() {
    static char jsonString[1024] = {0};
    static int i = 0;
    while(Serial2.available()) {
        jsonString[i] = char(Serial2.read());
        if(jsonString[i++] == '\n') {
          // there is a complete message
          Serial.print(jsonString);
          memset(jsonString, 0, 1024);
          i=0;
        }
    }
}
```

### Debug Messages:
There is not

lbernstone, 2019-02-25:
You can try increasing the buffer size (`Serial2.setRxBufferSize`), but at high rates like that, you may need to switch to the [esp-idf uart buffer](https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/peripherals/uart.html) so you can get interrupts or pattern matching.  Are you actually moving Mb/s of data through this channel?

nicolas-rabault, 2019-02-26:
I already try to increase the buffer without success. In the small example the program only manage serial, so the buffer don't have a lot of time to fill.
I will look at the lib you mention, I'm not really familiar with ESP environment yet.
This serial transmission send data full speed 2/3 of the time, so yes there is actually quite a lot of data.
Thank you for your help, I will let you know about my result using this lib.

nicolas-rabault, 2019-02-26:
As I can see there is no way to use the [esp-idf uart buffer](https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/peripherals/uart.html) lib easily into Arduino.
I would like to keep it simple because I want my users to be able to modify it easily without setting up a lot of things.
Do I miss something ? Is it possible to use it into the Arduino IDE?

nicolas-rabault, 2019-02-26:
I tried to do exactly the same code as the minimal one of my first post using the [esp-idf uart buffer](https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/peripherals/uart.html) lib : 

```
#include "uart.h"
void setup() {
  Serial.begin(115200);
  
  // put your setup code here, to run once:
  const int uart_num = UART_NUM_2;
  uart_config_t uart_config = {
      .baud_rate = 1000000,
      .data_bits = UART_DATA_8_BITS,
      .parity = UART_PARITY_DISABLE,
      .stop_bits = UART_STOP_BITS_1,
      .flow_ctrl = UART_HW_FLOWCTRL_DISABLE,
  };
  // Configure UART parameters
  ESP_ERROR_CHECK(uart_param_config((uart_port_t)uart_num, &uart_config));
  // Set UART pins(TX: IO16 (UART2 default), RX: IO17 (UART2 default), RTS: IO18, CTS: IO19)
  ESP_ERROR_CHECK(uart_set_pin((uart_port_t)UART_NUM_2, 17, 16, UART_PIN_NO_CHANGE, UART_PIN_NO_CHANGE));
  // Setup UART buffered IO with event queue
  const int uart_buffer_size = (1024 * 2);
  QueueHandle_t uart_queue;
  // Install UART driver using an event queue here
  ESP_ERROR_CHECK(uart_driver_install((uart_port_t)UART_NUM_2, 1024, 1024, 10, &uart_queue, 0));

  delay(100);
  // Write data to UART.
  const char detect_str[30] = "{\"detection\": {}}\r";
  uart_write_bytes((uart_port_t)UART_NUM_2, (const char*)detect_str, strlen(detect_str));
  Serial.println(detect_str);
 }

void loop() {
  // Read data from UART.
  uart_port_t uart_num = UART_NUM_2;
  static char jsonString[1024];
  static int get_length = 0;
  int length = 0;
  ESP_ERROR_CHECK(uart_get_buffered_data_len(uart_num, (size_t*)&length));
  get_length += uart_read_bytes(uart_num, (uint8_t*)&jsonString[get_length], length, 100);
  for (int i = length; i > 0; i--) {
    if(jsonString[i] == '\n') {
      // there is a complete message
          Serial.print(jsonString);
          memset(jsonString, 0, 1024);
          get_length = 0;
    }
  }
}
```
And I got the same kind of issue, I miss some datas.
Do I make something wrong here, or there is a bug on the lib?
Anyway I will try to use the pattern match interrupt to avoid doing it by myself...

Jeroen88, 2019-02-26:
Looking at the first sketch, I see an issue that, however, does not explain your problem: you do not check for the jsonString[] to overflow.

What might cause the problem, the speed of Serial is an order of 10 smaller than the speed of Serial2. While printing to Serial, Serial2 may overflow.

nicolas-rabault, 2019-02-26:
@lbernstone, I run some calculations about the real datarate of a given configuration (the worst for ESP) and I measure 43600 bytes/s or 348800 bits/s.
In this specific configuration each message is 26 bytes so I get 43600/26 = 1677 interrupt per seconds, which is a lot.

@Jeroen88 yes you right, I don't do it in this minimal sketch because in my big one this I never see it and I want to keep it simple.

> What might cause the problem, the speed of Serial is an order of 10 smaller than the speed of Serial2. While printing to Serial, Serial2 may overflow.

In my main program there is no serial output but there is other stuff running.
So I remove all prints if the received message is ok and now I dont have any missing data!

For sure we receive data faster than we manage them.
Now I have to test it with all my other stuff and make sure to manage serial frequently...

So you both (@lbernstone and @Jeroen88 ) are right, thanks a lot.

closed 2019-02-27 by nicolas-rabault