espressif/arduino-esp32 #2511: Guru Meditation Error: Core  1 panic'ed (LoadStoreError). Exception was unhandled.

GeorgeFlorian, 2019-02-26:
### Hardware:

Board: ESP32 DEVKIT1
Core Installation version: 1.0.1-git [this one](https://github.com/espressif/arduino-esp32)
IDE name: Arduino IDE
Flash Frequency: 80Mhz
PSRAM enabled: no
Upload Speed: 921600
Computer OS: Linux Mint 19.1 Mate

### Description:
I'm trying to write 2 lines in a file and use those lines as a SSID and Password to connect to a WiFi network. It works. But I get a Guru Meditation Error, then it restarts and works. 

I thinks it has something to do with manipulating a `SPIFFS file` that was opened in `FILE_WRITE` mode.
You will see that I've commented twice the following lines, because they too create a Guru Meditation error.

```
// Serial.print("/inputs.txt size: ");
// Serial.println(inputsCreate.size());
```

Now it stops when I'm trying to read the lines from `inputsWrite` file that was opened in `FILE_WRITE` mode. Here:
```
  int i = 0;
    while(inputsWrite.available()){
        String line= inputsRead.readStringUntil('\n');        
        line.trim();
        v[i] = line;
        i++;        
        String string2 = String("Linia ") + i + ": " + line;
        Serial.println(string2);             
    }
```
#### Is this faulty or intentional ? Thank you !

```
#include <Arduino.h>
#include <SPIFFS.h>
#include <WiFi.h>

String v[2];

void setup() {
  Serial.begin(115200);
  delay(2000);

  if(!SPIFFS.begin(true)){
    Serial.println("Couldn't mount SPIFFS. Formatting... ");
  }
  //--- Uncomment to manually reformat Flash Memory
  //SPIFFS.format();

  //--- Check if inputs/txt exists. If not then create one
  //--- Existential if
  if(!SPIFFS.exists("/inputs.txt")) {
    File inputsCreate = SPIFFS.open("/inputs.txt", FILE_WRITE);
    if(!inputsCreate) Serial.println("Couldn't create /inputs.txt");
    Serial.print("Was /inputs.txt created ? ");
    Serial.println(SPIFFS.exists("/inputs.txt"));
    // Serial.print("/inputs.txt size: ");
    // Serial.println(inputsCreate.size());
    inputsCreate.close();
  }
  //--- Check if /inputs.txt is empty (0 bytes) or not (>5)
  //--- println adds /r and /n, which means 2 bytes
  File inputsRead = SPIFFS.open("/inputs.txt");
  if(!inputsRead) {Serial.println("ERROR ! Couldn't open file to read !");}
  if(inputsRead.size() > 5) {
    Serial.println("inputsRead Content: ");

    int i = 0;
    while(inputsRead.available()){
        String line= inputsRead.readStringUntil('\n');  
        line.trim();      
        v[i] = line;
        Serial.print("V[i] length: ");
        Serial.println(v[i].length());
        i++;        
        String string2 = String("Linia ") + i + ": " + line;
        Serial.println(string2);             
    }
    Serial.print("inputsRead size: ");
    Serial.println(inputsRead.size());
    inputsRead.close();
  } else if(inputsRead.size()==0) {
    inputsRead.close();
    File inputsWrite = SPIFFS.open("/inputs.txt", FILE_WRITE);
    if(!inputsWrite) {Serial.println("ERROR ! Couldn't open file to write !");}
    inputsWrite.println("ssid");
    inputsWrite.println("password");
    Serial.println("inputsWrite Content: ");

    int i = 0;
    while(inputsWrite.available()){
        String line= inputsRead.readStringUntil('\n');        
        line.trim();
        v[i] = line;
        i++;        
        String string2 = String("Linia ") + i + ": " + line;
        Serial.println(string2);             
    }
//    Serial.print("inputsWrite size: ");
//    Serial.println(inputsWrite.size());
    inputsWrite.close();
  }

  Serial.print("Outside of `Existential if` /inputs.txt: ");
  Serial.println(SPIFFS.exists("/inputs.txt"));

  File inputs = SPIFFS.open("/inputs.txt");
  Serial.print("/inputs.txt size: ");
  Serial.println(inputs.size()); 
  inputs.close();  

  WiFi.begin(v[0].c_str(),v[1].c_str());  

  while (WiFi.status() != WL_CONNECTED) {    
    delay(500);
    Serial.println("Connecting to WiFi..");
  }
  Serial.println((String)"Connected to " + v[0] + " with IP addres: " + WiFi.localIP().toString());
}

void loop() {
  // put your main code here, to run repeatedly:
}
```

### Debug Messages:
```
12:52:48.867 -> Was /inputs.txt created ? 1
12:52:48.867 -> inputsWrite Content: 
12:52:48.867 -> Linia 1: 
12:52:48.867 -> Linia 2: 
12:52:48.900 -> Guru Meditation Error: Core  1 panic'ed (LoadStoreError). Exception was unhandled.
12:52:48.900 -> Core 1 register dump:
12:52:48.900 -> PC      : 0x40001410  PS      : 0x00060b30  A0      : 0x800d3a42  A1      : 0x3ffb1ea0  
12:52:48.900 -> A2      : 0x3f4002cc  A3      : 0x3ffb206c  A4      : 0x000000ff  A5      : 0x0000ff00  
12:52:48.900 -> A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x00000000  A9      : 0x3ffb1e70  
12:52:48.900 -> A10     : 0x3f4002cc  A11     : 0x00000000  A12     : 0x00000000  A13     : 0x00000000  
12:52:48.933 -> A14     : 0x00ff0000  A15     : 0xff000000  SAR     : 0x0000000a  EXCCAUSE: 0x00000003  
12:52:48.933 -> EXCVADDR: 0x3f4002cc  LBEG    : 0x400013f9  LEND    : 0x4000140d  LCOUNT  : 0xffffffff  
12:52:48.933 -> 
12:52:48.933 -> Backtrace: 0x40001410:0x3ffb1ea0 0x400d3a3f:0x3ffb1eb0 0x400d3a79:0x3ffb1ed0 0x400d1c8e:0x3ffb1ef0 0x400d454f:0x3ffb1fb0 0x4008ebfd:0x3ffb1fd0
12:52:48.998 -> 
12:52:48.998 -> Rebooting...
12:52:48.998 -> ets Jun  8 2016 00:22:57
12:52:48.998 -> 
12:52:48.998 -> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
12:52:48.998 -> configsip: 0, SPIWP:0xee
12:52:48.998 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
12:52:48.998 -> mode:DIO, clock div:1
12:52:48.998 -> load:0x3fff0018,len:4
12:52:48.998 -> load:0x3fff001c,len:1100
12:52:48.998 -> load:0x40078000,len:10208
12:52:48.998 -> load:0x40080400,len:6460
12:52:48.998 -> entry 0x400806a4
12:52:51.324 -> inputsRead Content: 
12:52:51.324 -> V[i] length: 4
12:52:51.324 -> Linia 1: ssid
12:52:51.324 -> V[i] length: 8
12:52:51.324 -> Linia 2: password
12:52:51.324 -> inputsRead size: 16
12:52:51.324 -> Outside of `Existential if` /inputs.txt: 1
12:52:51.324 -> /inputs.txt size: 16
12:52:51.921 -> Connecting to WiFi..
12:52:52.419 -> Connecting to WiFi..
12:52:52.419 -> Connected to ssid with IP addres: 192.168.1.106

```

###Exception Decoder

```
PC: 0x40001410
EXCVADDR: 0x3f4002cc

Decoding stack results
0x400d3a3f: String::copy(char const*, unsigned int) at /home/george/Arduino/hardware/espressif/esp32/cores/esp32/WString.cpp line 200
0x400d3a79: String::operator=(String const&) at /home/george/Arduino/hardware/espressif/esp32/cores/esp32/WString.cpp line 238
0x400d1c8e: setup() at /home/george/New Volume/Arduino Sketches/foo3/foo3.ino line 61
0x400d454f: loopTask(void*) at /home/george/Arduino/hardware/espressif/esp32/cores/esp32/main.cpp line 20
0x4008ebfd: vPortTaskWrapper at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/port.c line 143
```

GeorgeFlorian, 2019-03-04:
I fixed this by making a fs::File for each action used on "/inputs.txt". I don't know which methods work with what FILE_MODE (FILE_READ, FILE_WRITE, FILE_APPEND).
You need to experiment with these, or read the SPIFFS library better than I did.

The code above transformed into this:

```
#include <Arduino.h>
#include <SPIFFS.h>
#include <WiFi.h>

String v[2];

void fileReadLines(File file) {
  int i = 0;
    while(file.available()){
        String line= file.readStringUntil('\n');  
        line.trim();      
        v[i] = line;
        // Serial.print("V[i] length: ");
        // Serial.println(v[i].length());
        i++;        
        String string2 = String("Linia ") + i + ": " + line;
        Serial.println(string2);             
    }    
}

void setup() {
  Serial.begin(115200);
  delay(2000);

  if(!SPIFFS.begin(true)){
    Serial.println("Couldn't mount SPIFFS. Formatting... ");
  }
  //--- Check if inputs/txt exists. If not then create one
  //--- Existential if
 if(!SPIFFS.exists("/inputs.txt")) {
    File inputsCreate = SPIFFS.open("/inputs.txt", FILE_WRITE);
    if(!inputsCreate) Serial.println("Couldn't create /inputs.txt");
    Serial.print("Was /inputs.txt created ? ");
    Serial.println(SPIFFS.exists("/inputs.txt"));
    inputsCreate.close();
    delay(2000);
  }  
  //--- Check if /inputs.txt is empty (0 bytes) or not (>5)
  //--- println adds /r and /n, which means 2 bytes
  File inputsRead = SPIFFS.open("/inputs.txt");
  if(!inputsRead) {Serial.println("ERROR ! Couldn't open file to read !");}
  if(inputsRead.size() > 5) {
    Serial.println("inputsRead Content: ");
    fileReadLines(inputsRead);
    Serial.print("inputsRead size: ");
    Serial.println(inputsRead.size());
    inputsRead.close();
  } else if(inputsRead.size()==0) {
    inputsRead.close();
    File inputsWrite = SPIFFS.open("/inputs.txt", FILE_WRITE);
    if(!inputsWrite) {Serial.println("ERROR ! Couldn't open file to write !");}
    inputsWrite.println("ssid");
    inputsWrite.println("password");
    inputsWrite.close();
    File inputsUse = SPIFFS.open("/inputs.txt");
    Serial.println("inputsUse Content: ");
    fileReadLines(inputsUse);
    Serial.print("inputsUse size: ");
    Serial.println(inputsUse.size());
    inputsUse.close();
  }

  Serial.print("Outside of `Existential if` /inputs.txt: ");
  Serial.println(SPIFFS.exists("/inputs.txt"));

  File inputs = SPIFFS.open("/inputs.txt");
  Serial.print("/inputs.txt size: ");
  Serial.println(inputs.size()); 
  inputs.close();  

  WiFi.begin(v[0].c_str(),v[1].c_str());  

  while (WiFi.status() != WL_CONNECTED) {    
    delay(500);
    Serial.println("Connecting to WiFi..");
  }
  Serial.println((String)"Connected to " + v[0] + " with IP addres: " + WiFi.localIP().toString());
}

void loop() { }
```

closed 2019-03-04 by GeorgeFlorian

matbks, 2019-07-02:
request->send(SPIFFS, "/jquery.min.js", "text/javascript");
When my code tries to read jquery from SPIFFS it crashes, with Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed).
In this project i cannot acess the internet, thats why i need to bring jquery along with esp32.
Can you help me?

GeorgeFlorian, 2019-07-02:
> request->send(SPIFFS, "/jquery.min.js", "text/javascript");
> When my code tries to read jquery from SPIFFS it crashes, with Guru Meditation Error: Core 1 panic'ed (Cache disabled but cached memory [...]

This is how I send my jquery file:
```
  server.on("/jquery-1.12.4.min.js", HTTP_GET, [](AsyncWebServerRequest *request) {
    request->send(SPIFFS, "/jquery-1.12.4.min.js", "text/javascript");
  });
```

Also, make sure that you actually start SPIFFS in the beginning of your code.
This code checks if SPIFFS was mounted and if it wasn't (due to an error) it will delete all files from the flash memory and mount again.
```
  if (!SPIFFS.begin(true)) {
    Serial.println("An Error has occurred while mounting SPIFFS ! Formatting in progress");
    return;
  }
```

matbks, 2019-07-02:
Thank you so much for the answare. Unfortunatelly that didn't workout.

Actually i just discovered that this error only happens when i put the following code line on the setup:
`irrecv.enableIRIn(); // Start the IR receiver`
from 
`#include <IRremote.h>`

Looks like SPIFFS and IRremote can't work together.
I saved the error and decoded it, if you could take a look. Follows:

DECODED ERROR:
```
Decoding stack results
0x400dd11c: IRTimer() at C:\Users\mateu\Documents\Arduino\libraries\ESP32-IRremote\IRremote.cpp line 131
0x40082561: spi_flash_read at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/spi_flash/flash_ops.c line 537
0x400e76d6: esp_partition_read at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/spi_flash/partition.c line 242
0x400eb6dd: spiffs_api_read at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/spiffs/spiffs_api.c line 36
0x400ef873: spiffs_phys_rd at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/spiffs/spiffs/src/spiffs_cache.c line 161
0x400ecfe3: spiffs_obj_lu_find_entry_visitor at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/spiffs/spiffs/src/spiffs_nucleus.c line 166
0x400eeb5e: spiffs_object_find_object_index_header_by_name at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/spiffs/spiffs/src/spiffs_nucleus.c line 1694
0x400ec4d8: SPIFFS_stat at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/spiffs/spiffs/src/spiffs_hydrogen.c line 765
0x400eb0bc: vfs_spiffs_stat at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/spiffs/esp_spiffs.c line 536
0x400e4ba1: esp_vfs_stat at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/vfs/vfs.c line 492
0x400e6911: stat at ../../../.././newlib/libc/syscalls/sysstat.c line 12
0x400d4a22: VFSImpl::open(char const*, char const*) at C:\Users\mateu\Documents\Arduino\hardware\espressif\esp32\libraries\FS\src\vfs_api.cpp line 41
0x400d400a: SPIFFSImpl::exists(char const*) at C:\Users\mateu\Documents\Arduino\hardware\espressif\esp32\libraries\SPIFFS\src\SPIFFS.cpp line 42
0x4016e0c6: fs::FS::exists(char const*) at C:\Users\mateu\Documents\Arduino\hardware\espressif\esp32\libraries\FS\src\FS.cpp line 199
0x400d413b: fs::FS::exists(String const&) at C:\Users\mateu\Documents\Arduino\hardware\espressif\esp32\libraries\FS\src\FS.cpp line 204
0x400d6965: AsyncWebServerRequest::send(fs::FS&, String const&, String const&, bool, std::function ) (C:\Program Files at x86)\Arduino\libraries\ESPAsyncWebServer-master\src\WebRequest.cpp line 771
0x400d2149: std::_Function_handler   >::_M_invoke(const std::_Any_data &,  ) at C:\Users\mateu\Documents\Arduino\WifiLoginIR/WifiLoginIR.ino line 181
0x400d8f75: AsyncCallbackWebHandler::handleRequest(AsyncWebServerRequest*) at c:\users\mateu\documents\arduino\hardware\espressif\esp32\tools\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\5.2.0/functional line 2271
0x400d6bf1: AsyncWebServerRequest::_parseLine() (C:\Program Files at x86)\Arduino\libraries\ESPAsyncWebServer-master\src\WebRequest.cpp line 576
0x400d6cb5: AsyncWebServerRequest::_onData(void*, unsigned int) (C:\Program Files at x86)\Arduino\libraries\ESPAsyncWebServer-master\src\WebRequest.cpp line 121
0x400d6f1d: std::_Function_handler   >::_M_invoke(const std::_Any_data &,  ,  ,  ,  ) (C:\Program Files at x86)\Arduino\libraries\ESPAsyncWebServer-master\src\WebRequest.cpp line 75
0x400d9d4d: AsyncClient::_recv(tcp_pcb*, pbuf*, signed char) at c:\users\mateu\documents\arduino\hardware\espressif\esp32\tools\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\5.2.0/functional line 2271
0x400d9dc9: AsyncClient::_s_recv(void*, tcp_pcb*, pbuf*, signed char) (C:\Program Files at x86)\Arduino\libraries\AsyncTCP-master\src\AsyncTCP.cpp line 1144
0x400da356: _async_service_task(void*) (C:\Program Files at x86)\Arduino\libraries\AsyncTCP-master\src\AsyncTCP.cpp line 145
0x40088685: vPortTaskWrapper at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/port.c line 143
```
ERROR:
```
05:35:37.214 -> PC      : 0x400dd11c  PS      : 0x00060034  A0      : 0x40081800  A1      : 0x3ffbe7a0  
05:35:37.214 -> A2      : 0x00000001  A3      : 0x00000002  A4      : 0x000000ff  A5      : 0x4008b3fc  
05:35:37.214 -> A6      : 0x00000000  A7      : 0x00000001  A8      : 0x8008133c  A9      : 0x3ff5f024  
05:35:37.214 -> A10     : 0x3ffbec3c  A11     : 0x20000000  A12     : 0x00000400  A13     : 0x3ffce6b0  
05:35:37.248 -> A14     : 0x00000026  A15     : 0x3ffc2b68  SAR     : 0x00000017  EXCCAUSE: 0x00000007  
05:35:37.248 -> EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffc  
05:35:37.248 -> Core 1 was running in ISR context:
05:35:37.248 -> EPC1    : 0x40082564  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x400dd11c
05:35:37.248 -> 
05:35:37.248 -> Backtrace: 0x400dd11c:0x3ffbe7a0 0x400817fd:0x3ffbe7c0 0x40082561:0x3ffce6f0 0x400e76d6:0x3ffce760 0x400eb6dd:0x3ffce790 0x400ef873:0x3ffce7c0 0x400ecfe3:0x3ffce7f0 0x400eeb5e:0x3ffce830 0x400ec4d8:0x3ffce870 0x400eb0bc:0x3ffce8a0 0x400e4ba1:0x3ffce900 0x4000bcc5:0x3ffce920 0x400e6911:0x3ffce940 0x400d4a22:0x3ffce960 0x400d400a:0x3ffce9e0 0x4016e0c6:0x3ffcea20 0x400d413b:0x3ffcea40 0x400d6965:0x3ffcea60 0x400d2149:0x3ffceab0 0x400d8f75:0x3ffceb00 0x400d6bf1:0x3ffceb40 0x400d6cb5:0x3ffceb80 0x400d6f1d:0x3ffcebc0 0x400d9d4d:0x3ffcebe0 0x400d9dc9:0x3ffcec20 0x400da356:0x3ffcec40 0x40088685:0x3ffcec70
```
Thank you again.

GeorgeFlorian, 2019-07-02:
Start a new issue !
I have no idea how to help you.