espressif/arduino-esp32 #2514: Why the use of non-standard functionality for serial.print of local time

Gerriko, 2019-02-26:
I am using the time.h core library to get local time.

I am using the "simple time" example but I note that there is non-standard Serial.println formatting functionality included. 

Why so?? 

And, where is this formatting handled as I cannot see the code inside time.h?

In my opinion, this causes more problems than it solves.

For example, I want to display time on a display, such as TFT display or an EPD.  I cannot see how to do this simply. So, is there any way to do this simpler at present?

I found some sort of solution online but all this additional code could readily be resolved if there was a better implementation inside the time library for formatting. 

https://arduino.stackexchange.com/questions/52676/how-do-you-convert-a-formatted-print-statement-into-a-string-variable

lbernstone, 2019-02-26:
It comes from [Print()](https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/Print.cpp#L149), which in turn calls strftime().

CelliesProjects, 2019-03-02:
You can use `strftime` yourself to format time strings.
I use something like the code below to format a string to send it to a TFT or OLED.
````
  struct tm timeinfo;
  getLocalTime( &timeinfo );
  char buff[15];
  strftime( buff, sizeof( buff ), "%T", &timeinfo );
````
See http://www.cplusplus.com/reference/ctime/strftime/

Gerriko, 2019-03-02:
Indeed you can. But in my opinion, to be more in line with the "Arduino
style" those functions should be wrapped inside the library rather than
exposed.

closed 2019-08-15 by stale[bot]