espressif/arduino-esp32 #2516: Wemos Lolin32 unable to connect to upload a sketch

verbessern, 2019-02-26:
### Hardware:
Board:							Wemos Lolin32 (maybe fake because the lolin32 v1.0.0 is overpainted black)
Core Installation/update date:			26.02.2019
IDE name:							Arduino IDE 1.8.8
Flash Frequency:					40/80Mhz
Upload Speed:						115200 and others
Computer OS:                                            Ubuntu 18.04

### Description:
Hi community, I'm trying to upload a program to my new wemos lolin 32, but I can't connect to the micro controller. My usb cable works well with my phone for data transfer, so it should not be the problem. I have tried to ground the GPIO0/2 and upload, but with no success. When i hold the reset button i see a small ~0.03v voltage drop in the 3.3v and 5v pins. "dmesg | grep tty" gives "usb 1-3: cp210x converter now attached to ttyUSB0" so it seems recognizes the device. Arduino IDE also shows in the menu "ports". Around the board 3.3v and 5v are present relative to the ground, also the CHG diode is blinking like 2 times per second, but as i read around, its 'normal', because some boards don't recognize the missing battery (I write this to be explicit, in case its important). I have measured that the 2 pins of the chip entry supply have 3.3v difference, so to the chip arrives electricity. PuTTY connects to the serial port, but stays blank (probably because no program is uploaded yet or worse). Terminal command "esptool.py chip_id" detects the port, but also can't connect. "cu -l /dev/ttyUSB0  -s 9600" connects, but any button freezes the terminal, except "~." for quit. I have tested in other computer with windows 7 older Arduino (don't have exact data), but also no connection is made. How can i detect if the chip is broken? I'm new in hardware. Please help.

### Debug Messages:
Sketch uses 190944 bytes (14%) of program storage space. Maximum is 1310720 bytes.
Global variables use 12632 bytes (3%) of dynamic memory, leaving 315048 bytes for local variables. Maximum is 327680 bytes.
esptool.py v2.6-beta1
Serial port /dev/ttyUSB0
Connecting........_____....._____....._____....._____....._____....._____.....____An error occurred while uploading the sketch
_

A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header

lbernstone, 2019-02-26:
Open a terminal and run `python ~/.arduino15/packages/esp32/hardware/esp32/1.0.1/tools/esptool.py chip_id`.  If this works, then you likely need to hold down a gpio0/boot button while you upload.  If not, then there is some physical problem.  Try another USB cable or port.

lbernstone, 2019-02-27:
put a wire between gpio0 and ground while programming.  Sounds like you have a defective unit.

verbessern, 2019-02-27:
Thank you for your answers, I have done both of this, its written in my issue up, did not help. I will have another cable after 2 days, that was working for sure with esp8266 programming, but the cable seems not to be the problem, because it works with my phone for data transfer. Is there a way to confirm that the chip is broken?  As a last resort, I'm planing to disconnect it from the board and wire it directly, but maybe there is a better way.

1TekPro, 2019-02-27:
Sounds like a coms issue. What bit/baud rates have you tried?

verbessern, 2019-02-27:
I have tried all of them, including 40/80 Mhz. Sometimes when I testing it changes from usb0 to usb1 and back, i always choose what ever is present. "dmesg | grep tty" shows it properly on plug/unplug of the cable, and cp210x is detected correctly.

1TekPro, 2019-02-27:
@verbessern Just for testing, make sure the port settings in device manager (if windows, not sure on mac or linux) are correct. Try manually assigning the device to USB3 and rebooting.

1TekPro, 2019-02-27:
While using the serial monitor at 115200 baud, hold down boot and press reset. Does "waiting for download" show in serial monitor?

verbessern, 2019-02-27:
@1TekPro I don't have /mnt/usb0(1), inside are 5 folders with uuid names, all empty. I get '/dev/ttyUSB0' that is not a directory.  I don't have a boot button. WEMOS Lolin32 CP2104 have only a reset button. I have however tried the "popular" solution to ground GPIO 0 and 2, but did not help.

atanisoft, 2019-02-27:
Can you share a pic of the board you are using?

verbessern, 2019-02-27:
The total effect is like the device does not respond. Does someone know, will the device connect at all if its **not** in 'download mode' (the mode that the button boot is pressed down on some other esp32 modules)? Here is the photo: 
[image]

atanisoft, 2019-02-27:
I'd say it is not an "official" wemos board if it doesn't have the silk screened lolin32 V1.0.0 (which your blurry pic doesn't appear to show).

Can you open a serial terminal with the board attached at 9600 baud and see if you can capture the boot loader messages? remove the wires from GND to 0, 2 for this test just to see that it is booting up at all. If you don't get at least some garbage characters on the serial terminal (ie: wrong baud rate) you may have a bad board. If you do get some output then try adding GND to 0,2 and press the reset button with the serial terminal still open. Check the output from the boot loader for "waiting for download" which is what it should say with GND to 0,2.

1TekPro, 2019-02-27:
Can you take a clearer pic of the wifi chip?

verbessern, 2019-02-27:
@1TekPro here they are. In the board the Lolin32 v1.0.0 is over painted as slightly visible on one of the photos.
[image]
[image]
[image]

verbessern, 2019-02-27:
@atanisoft PuTTY connects to the /dev/ttyUSB0 at 9600, but when i click 'reset' button on the MC nothing happens, but if i unplug the device the terminal closes. The LED 5 lights a little when i touch with my hand the GPIO 5 (discovered by accident). CHG is always blinking 2 times per second or so.

atanisoft, 2019-02-27:
in your updated pics above it looks like the WROOM-32 IC is not soldered for a few pins and some pads seem out of alignment.

Since you didn't get anything from PuTTY that doesn't sound hopeful that the board is functional.

verbessern, 2019-02-27:
@atanisoft i have seen other pictures on the internet for the same board, and this "middle" pins look the same as on my version.

atanisoft, 2019-02-27:
The middle pins are not the ones I'm referring to, the ones on the left and right sides of the chip do not look connected.

verbessern, 2019-02-27:
@atanisoft Yes the chip is alittle to the right, and the right pins and middle pins look well connected, but on the left its hard to tell. I will try to verify that and will edit here.

verbessern, 2019-02-27:
Photoes: [image]
[image]
[image]

closed 2019-02-27 by verbessern

1TekPro, 2019-02-27:
The second pin from the top left side is not connected. This should be the 3.3v input on that board. If you have a multimeter, check for power from the top 2 left pins. If you don't have a meter, an led would work.

verbessern, 2019-02-27:
YES ! IT IS ALIVE!
[image]
Thank you guys really much! Now the terminal shows some characters! Awesome!

1TekPro, 2019-02-27:
Nice solder work there.  :)
I'm glad you got it fixed.

verbessern, 2019-02-27:
Thank you, Its my first board to solder, but i was watching videos in youtube for a month, till i was waiting the controller to arrive. I did that by putting **very little** solder on the top of my iron and just touch the pad. The solder just go sucked into its place, then i repeated on all pins. The beginners luck i guess :).

kn100, 2020-06-23:
I don't want to reopen this issue since I am only posting this here just so others know - to flash one of these 'clone' boards that do not have the EN button like mine didn't, the flashing process looks like this:

1. Plug in device
2. Start flash, wait for the arduino ide to report 'Connecting'...
3. Connect the EN pin to GND briefly using a wire or something
4. The thing starts to flash!
5. The hard reset via RTS pin will fail too, so wait for Arduino IDE to report 'Hard resetting via RTS pin' - and press the reset switch on the device, at which point your serial monitor should spring to life. Congrats!

sethismyfriend, 2021-12-07:
Thanks! That helped me with mine.

JeanBlignaut, 2022-04-06:
Hmm I'm having a similar issue with a "Lolin" d3 with built in OLED - the device is functional since the display goes through a demo loop when powered - however I have trouble even finding the device port under macOS.
I installed ESP-IDF VSCode extension and under windows this reports that the device is not set in the correct mode for programming - now if only I could manage to find the device on macOS where my actual sketch is...

WhiteHatM, 2023-08-05:
[image]
I made a "BOOT" button like this - Placing a button between GPIO0 and GND, whenever IDE shows "Connecting....." I hold this button and the sketch gets uploaded smoothly.