espressif/arduino-esp32 #2517: Partitions

xman4242, 2019-02-26:
Does anyone know which partition the library writes to?

baaridunnasr, 2019-02-27:
[Docs](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/partition-tables.html) on partitions.

Code is usually written to the methods specified in the above link. For me it writes to OTA_0, usually. I don't have a factory partition.

```
# Espressif ESP32 Partition Table
# Name,   Type, SubType, Offset,  Size, Flags
nvs,      data, nvs,     0x9000,  0x4000,
otadata,  data, ota,     0xd000,  0x2000,
phy_init, data, phy,     0xf000,  0x1000,
factory,  0,    0,       0x10000, 1M,
ota_0,    0,    ota_0,  0x110000, 1M,
ota_1,    0,    ota_1,  0x210000, 1M,
```

baaridunnasr, 2019-02-27:
Please ask questions related to usage such as this, [here](https://gitter.im/espressif/arduino-esp32). Such things can be easily researched and doesn't warrant creating an issue. Please close it.

closed 2019-03-01 by xman4242