espressif/arduino-esp32 #2518: Adafruit FONA Guru meditation error: load prohibited

TheIronMechanic, 2019-02-27:
### Hardware:
Board:							ESP-WROOM_32 NodeMCU
Core Installation/update date:			15.0.2019
IDE name:							Arduino IDE
Flash Frequency:					80mhz
PSRAM enabled:                                          no
Upload Speed:					115200
Computer OS:                                            Windows 10

### Description:
I'm using the adafruit Fona Library (latest version), An other sketch using it works fine.
But using this sketch i get a "Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled."

### Sketch:  
```

#include "Adafruit_FONA.h"

#define FONA_RST 4
#define FONA_KEY 8

HardwareSerial *fonaSerial = new HardwareSerial(2);
Adafruit_FONA fona = Adafruit_FONA(FONA_RST);

uint8_t readline(char *buff, uint8_t maxbuff, uint16_t timeout = 0);
uint8_t type;

void setup() {

  Serial.begin(115200);
  fonaSerial->begin(115200);

        uint8_t n = fona.getNetworkStatus();
        Serial.print(F("Network status "));
        Serial.print(n);
        Serial.print(F(": "));
        if (n == 0) Serial.println(F("Not registered"));
        if (n == 1) Serial.println(F("Registered (home)"));
        if (n == 2) Serial.println(F("Not registered (searching)"));
        if (n == 3) Serial.println(F("Denied"));
        if (n == 4) Serial.println(F("Unknown"));
        if (n == 5) Serial.println(F("Registered roaming"));
}

void loop() {
  // put your main code here, to run repeatedly:
  delay(100);

```
}
### Debug Messages:
```
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:952
load:0x40078000,len:6084
load:0x40080000,len:7936
entry 0x40080310
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400e6b2e  PS      : 0x00060530  A0      : 0x800d0ae5  A1      : 0x3ffb1ed0  
A2      : 0x3ffc23ec  A3      : 0x000000c6  A4      : 0x0ffd114c  A5      : 0x00000000  
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x00020000  A9      : 0x3ff44024  
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x00000000  A13     : 0x00000001  
A14     : 0x00060d20  A15     : 0x00000000  SAR     : 0x0000000f  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace: 0x400e6b2e:0x3ffb1ed0 0x400d0ae2:0x3ffb1ef0 0x400d0b75:0x3ffb1f10 0x400d0c0f:0x3ffb1f30 0x400d0c49:0x3ffb1f50 0x400d09e0:0x3ffb1f80 0x400e5677:0x3ffb1fa0

Rebooting...

```
The Exception decoder outputs this:

```
PC: 0x400e6f22: Adafruit_FONA::available() at C:\Users\Lucas\Documents\Arduino\libraries\Adafruit_FONA_Library\Adafruit_FONA.cpp line 1718
EXCVADDR: 0x00000000

Decoding stack results
0x400e6f22: Adafruit_FONA::available() at C:\Users\Lucas\Documents\Arduino\libraries\Adafruit_FONA_Library\Adafruit_FONA.cpp line 1718
0x400d0bae: Adafruit_FONA::flushInput() at C:\Users\Lucas\Documents\Arduino\libraries\Adafruit_FONA_Library\Adafruit_FONA.cpp line 1741
0x400d0c41: Adafruit_FONA::getReply(__FlashStringHelper const*, unsigned short) at C:\Users\Lucas\Documents\Arduino\libraries\Adafruit_FONA_Library\Adafruit_FONA.cpp line 1817
0x400d0cdb: Adafruit_FONA::sendParseReply(__FlashStringHelper const*, __FlashStringHelper const*, unsigned short*, char, unsigned char) at C:\Users\Lucas\Documents\Arduino\libraries\Adafruit_FONA_Library\Adafruit_FONA.cpp line 2050
0x400d0d15: Adafruit_FONA::getNetworkStatus() at C:\Users\Lucas\Documents\Arduino\libraries\Adafruit_FONA_Library\Adafruit_FONA.cpp line 243
0x400d0aa3: setup() at D:\Data\Code\GPS Tracker\TESTB0.133/TESTB0.133.ino line 86
0x400e6137: loopTask(void*) at C:\Users\Lucas\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0\cores\esp32\main.cpp line 15

```
Any idea why?

atanisoft, 2019-02-27:
Maybe open the issue against the Adafruit_FONA library as it doesn't appear to have anything to do with Arduino-esp32 code at all.

stickbreaker, 2019-02-27:
What pins are you using with `Serial2`?

TheIronMechanic, 2019-02-27:
I use pins p14 and p27.

I searched really long on the FONA github and I think I found a solution, but I’m gonna try It tomorrow, it’s 3 am here.
If it resolves my problem I’m going to close the issue tomorrow.

Thanks for your help!

closed 2019-02-27 by TheIronMechanic

PROFEeSp, 2019-03-17:
what was the problem?

RenzoBeux, 2022-03-24:
Hate this people who closes an issue without telling the solution