espressif/arduino-esp32 #2519: Issue with multiple hardware serial ports on Mega 2560

CarlP66, 2019-02-27:
I am currently having a problem very similar to closed problem #665 from Sept 2017 with my Mega 2560. 

Serial0 works fine 100% of the time for communication with a Raspberry Pi using CmdMessenger / PyCmdMessenger at 9600 baud, but when I also communicate with a TFMini Lidar unit on Serial1 at 115200, the Serial0 connection to the Pi intermittently gets screwed up - it works fine for a few transfers, then crashes the Python code on the Raspberry Pi after a timeout.

I believe my IDE and environment is up to date, but can someone be specific about what was updated back in Sept 2017 to fix this.

I have confirmed that my Serial0 and Python code work fine when I comment out the Serial1 reads, and just set t2 & t4 in the code below to integer values (ie t1 = 5; t4 = 10;)

The entire code is quite long, but the declarations and offending function are below:

================ Declarations ================

#include <NewPing.h>
#include <Servo.h>
#include "CmdMessenger.h"

#define CENTER_TRIGGER_PIN 10
#define CENTER_ECHO_PIN 11
#define LEFT_TRIGGER_PIN 8
#define LEFT_ECHO_PIN 9
#define RIGHT_TRIGGER_PIN 12
#define RIGHT_ECHO_PIN 13
#define MAX_DISTANCE 200

NewPing centerSonar(CENTER_TRIGGER_PIN, CENTER_ECHO_PIN, MAX_DISTANCE);
NewPing leftSonar(LEFT_TRIGGER_PIN, LEFT_ECHO_PIN, MAX_DISTANCE);
NewPing rightSonar(RIGHT_TRIGGER_PIN, RIGHT_ECHO_PIN, MAX_DISTANCE);
Servo mytiltservo; // create tilt servo object to control a servo
Servo mypanservo; // create pan servo object to control a servo

const int leftForward = 2;
const int leftBackward = 4;
const int leftEnable =5;
const int rightForward = 3;
const int rightBackward = 7;
const int rightEnable = 6;

int tiltpos = 90; // variable to store the servo position
int panpos =90;
int centerDistance = 100;
int leftDistance = 100;
int rightDistance = 100;

int t1; // byte 3 from lidar
int t2; // byte 4 from lidar
int t3; // byte 5 from lidar
int t4; // byte 6 from lidar

enum {
spin,
anticlockwise,
anticlockwise_until,
clockwise,
clockwise_until,
backup,
forward,
forward_until,
ping_all,
pan_tilt_test,
lidar_read,
pings,
lidar_val,
error,
};

/* Initialize CmdMessenger -- this should match PyCmdMessenger instance */
const int BAUD_RATE = 9600;
const long LIDAR_BAUD_RATE = 115200;
CmdMessenger c = CmdMessenger(Serial,',',';','/');

void setup(){
pinMode(leftForward , OUTPUT);
pinMode(leftBackward , OUTPUT);
pinMode(rightForward , OUTPUT);
pinMode(rightBackward , OUTPUT);
Serial.begin(BAUD_RATE);
Serial1.begin(LIDAR_BAUD_RATE);

attach_callbacks();
}

void attach_callbacks(void) {
c.attach(spin,on_spin);
c.attach(anticlockwise,on_anticlockwise);
c.attach(anticlockwise_until,on_anticlockwise_until_free);
c.attach(clockwise,on_clockwise);
c.attach(clockwise_until,on_clockwise_until_free);
c.attach(backup,on_backup);
c.attach(forward,on_forward);
c.attach(forward_until,on_forward_until_object);
c.attach(ping_all,on_ping_all);
c.attach(pan_tilt_test,on_pan_tilt_test);
c.attach(lidar_read,on_lidar_read);
c.attach(on_unknown_command);
}

===================== Offending Function =========================

void on_lidar_read()
{
// flush lidar buffer
while(Serial1.available() > 0)
{
char t = Serial1.read();
}

// wait for 9 new bytes - ie one full message
while(Serial1.available()<9)
{
delay(5);
}

// read message & return to Raspberry Pi
if((0x59 == Serial1.read()) && (0x59 == Serial1.read())) //Byte 1 & Byte 2
{
t1 = Serial1.read(); //Byte 3
t2 = Serial1.read(); //Byte 4
t3 = Serial1.read(); //Byte 5
t4 = Serial1.read(); //Byte 6
for(int i=0; i<3; i++)
{
Serial1.read(); ////Byte 7,8,9
}

```
    t2 <<= 8;
    t2 += t1;
    t4 <<= 8;
    t4 += t3;

    c.sendCmdStart(lidar_val);
    c.sendCmdBinArg(t2);
    c.sendCmdBinArg(t4);
    c.sendCmdEnd();

}
}

Please help. Thanks.

stickbreaker, 2019-02-27:
@CarlP66 What does this have to do with a ESP32? Are you saying that you are having a problem with ATMega 2560 (atmel) based Arduino board? 

Chuck.

CarlP66, 2019-02-27:
Hi Chuck. My problem is not related to the ESP32, it is an ATMega2560 issue, but the problem is identical to that reported in problem #665 that was, I understand an ATMega 2560 issue that manifsted when using the ESP32 over serial. If you know a better place to report it, or seek support please advise.

stickbreaker, 2019-02-27:
@CarlP66 
Your code does not sync with the input,
```c++
=================== Offending Function =========================
void on_lidar_read()
{
// flush lidar buffer
while(Serial1.available() > 0)
{
char t = Serial1.read();
}

// wait for 9 new bytes - ie one full message
while(Serial1.available()<9)
{
delay(5);
}
```

you need to read Serial input, store into buffer, sync with message in buffer, pass message to handler function.

```c++

enum STAGE{
  stage_none,
  stage_secondByte,
  stage_build,
  stage_good};

#define BUFFLEN 20
char buff[BUFFLEN]; // temp input buffer
static uint8_t head=0, tail=0;
// tail ==head, buffer empty (head+1)%BUFFLEN == tail, buffer is full
// add at head, read from tail

void lidar_in(){
	while(Serial1.available()){
		if((head+1)%BUFFLEN == tail){// bad news, buffer overflow, eat tail
			tail++;
			tail %= BUFFLEN;
		}
		head = (head+1)%BUFFLEN;
		buff[head] = Serial1.read();
// test if complete message is in buffer
		uint8_t msgStart = tail;
		uint8_t msgPos = tail;
		enum STAGE stage=stage_none;
		while(!(stage == stage_good) &&  (msgPos != head)){
			switch(stage){
				case stage_none : 
					if(buff[msgPos] == 0x59) {
						stage = stage_secondByte;
						msgLen = 1;
						tail = msgPos;  // advance tail to possible message start
					}
					break;
				case stage_secondByte :
					if(buff[msgPos] == 0x59){
						stage = stage_build;
						msgLen++;
					}
					else {
						stage = stage_none; // start over
					}	  
					break;
				case stage_build:
					msgLen++;
					if(msgLen ==9) { // full message with 59,59 as start marker
						stage = stage_good;
					}
					else if (msgLen == 3) { // start of data
						msgStart = msgPos;
					}
					break;
				default : ;
			}	 //switch
			if( stage != stage_good){
				msgPos = (msgPos +1 ) %BUFFLEN;
			}
		}
		if( stage == stage_good){ // process message
			uint16_t t2 = buff[msgStart];
			msgStart = (msgStart+1)%BUFFLEN;
			t2 = (buff[msgStart] <<8)+ t2;
			msgStart = (msgStart+1)%BUFFLEN;
			uint16_t t4 = buff[msgStart];
			msgStart = (msgStart+1)%BUFFLEN;
			t4 = (buff[msgStart] <<8)+ t4;
		
			c.sendCmdStart(lidar_val);
			c.sendCmdBinArg(t2);
			c.sendCmdBinArg(t4);
			c.sendCmdEnd();
			tail = (msgPos+1)%BUFFLEN; // consume buffer to inhibit duplicate message detection
		}
	}
}
```

I don't think there is anything wrong with the hardware, I think you need to sync with your message.

Chuck.

CarlP66, 2019-02-27:
Chuck, 
Many thanks. I think you are right. Apologies if this was not the right place to post this.
Thanks, C.

projectgus, 2019-03-25:
Closing as I don't think this has any relation to ESP32.

closed 2019-03-25 by projectgus