espressif/arduino-esp32 #2521: DOIT DEV KIT Continuos resetting

baaridunnasr, 2019-02-28:
### Hardware:
Board:							DOIT ESP32 DEV KIT
Core Installation version:			1.0.1
IDE name:						PIO
Flash Frequency:					40Mhz
PSRAM enabled:                                    No
Upload Speed:					921600
Computer OS:                                        Ubuntu

### Description:
Flashed a really big sketch. The ESP32 continuously reboots while outputting some messages. The sketch is really big. So I have cleanly changed the partitions.

# Name | Type | SubType | Offset | Size | Flags
-- | -- | -- | -- | -- | --
nvs | data | nvs | 0x9000 | 0x5000 |  
otadata | data | ota | 0xe000 | 0x2000 |  
app0 | app | ota_0 | 0x10000 | 0x3F0000 |  

The following is my partitions.

I have used this partition setup before without any issues. And PIO clearly compiles without any issues. I do see that it labels the reset as sofware but it doesn't even cross the setup function! The Serial message "STARTED" is not displayed at all. I have not enabled flash encryption or anything to the best of my knowledge.

### Sketch:
```cpp

void setup()
{
    delay(10000);
    Serial.begin(115200);
    Serial.println("STARTED");
    delay(10000);

    pinMode(LED_BUILTIN, OUTPUT);
    attachInterrupt(digitalPinToInterrupt(interrupt_pin), BT_interrupt_handler, FALLING);
    POST.init();
    data_UPSI_queue = xQueueCreate(40, sizeof(struct data_UPSI));
    xTaskCreatePinnedToCore(&core_1_function_UPSI, "UPSI", 5000000, NULL, 1, &core_1_task_UPSI, 1);
    xTaskCreatePinnedToCore(&core_0_function_comms_and_logs, "Comms and Logs", 5000000, NULL, 1, &core_0_task_comms_and_logs, 0);
    xTaskCreatePinnedToCore(&core_0_function_BT_handler, "BT Interrupt", 5000000, NULL, 2, &core_0_task_BT_handler, 0);
}

void loop()
{
    delay(10);
}
```

### Debug Messages:
```
rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:928
ho 0 tail 12 room 4
load:0x40078000,len:9280
load:0x40080400,len:5860
entry 0x40080698
ets Jun  8 2016 00:22:57

```

stickbreaker, 2019-02-28:
Why do you think you can create 3 tasks with 5MEGABYTES of stack each on a processor with 320kbytes TOTAL of data RAM?

Chuck.

baaridunnasr, 2019-02-28:
Sorry @stickbreaker That is incredibly foolish of me. It was originally the below values. I just added some values out of frustration. 
```cpp
void setup()
{
    delay(10000);
    Serial.begin(115200);
    Serial.println("STARTED");
    delay(10000);

    pinMode(LED_BUILTIN, OUTPUT);
    attachInterrupt(digitalPinToInterrupt(interrupt_pin), BT_interrupt_handler, FALLING);
    POST.init();
    data_UPSI_queue = xQueueCreate(40, sizeof(struct data_UPSI));
    xTaskCreatePinnedToCore(&core_1_function_UPSI, "UPSI", 1000, NULL, 1, &core_1_task_UPSI, 1);
    xTaskCreatePinnedToCore(&core_0_function_comms_and_logs, "Comms and Logs", 1000, NULL, 1, &core_0_task_comms_and_logs, 0);
    xTaskCreatePinnedToCore(&core_0_function_BT_handler, "BT Interrupt", 1000, NULL, 2, &core_0_task_BT_handler, 0);
}

void loop()
{
    delay(10);
}
```

baaridunnasr, 2019-02-28:
I'd just presumed that I would be alerted by IDE if I went above 320KB of RAM.
```
Linking .pioenvs/esp32doit-devkit-v1/firmware.elf
Building .pioenvs/esp32doit-devkit-v1/firmware.bin
Retrieving maximum program size .pioenvs/esp32doit-devkit-v1/firmware.elf
Checking size .pioenvs/esp32doit-devkit-v1/firmware.elf
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [==        ]  18.7% (used 61320 bytes from 327680 bytes)
PROGRAM: [===       ]  33.8% (used 1419346 bytes from 4194304 bytes)
```
This is what PIO reports.

baaridunnasr, 2019-02-28:
@stickbreaker  The error persists, regardless.

stickbreaker, 2019-02-28:
Why do you delay 10 seconds before initing serial?

stickbreaker, 2019-02-28:
Min stack should be 2k...8k based on your call depth and os calls (wifi is a pig)

stickbreaker, 2019-02-28:
The compiler only counts static global RAM allocations. Taskcreate uses heap to allocate at runtime.

baaridunnasr, 2019-02-28:
Initially wrote without delay before Serial Init. But then it wouldn't even reach the Serial.println... As in the error would occur as soon as the flash was over. So I thought what if I tried to put a delay in there and see if I could make the errors stop a bit before appearing. But that was no luck at all...

I'll try to see if I can play around with the stack. But isn't it reasonable to conclude that the code would go though that `Serial.println("STARTED");` before crashing? I could at least eliminate that the program itself isn't starting in some way...

baaridunnasr, 2019-02-28:
Allocated 20K of stack now. It's whopping 60K of total stack now but I guess it should show some working before crashing...

baaridunnasr, 2019-02-28:
Nope. 20K of stack on the tasks doesnt change anything at all. Worst part is that it's not even a core panic! At least then I could say that somewhere the code is wrong... Do you have any things I could check in my code that could help me out? @stickbreaker

baaridunnasr, 2019-02-28:
```cpp
#include "Arduino.h"

void setup()
{
    Serial.begin(115200);
    Serial.println("STARTED");
    delay(10000);

    pinMode(LED_BUILTIN, OUTPUT);
    // attachInterrupt(digitalPinToInterrupt(interrupt_pin), BT_interrupt_handler, FALLING);
    // POST.init();
    // data_UPSI_queue = xQueueCreate(10, sizeof(struct data_UPSI));
    // xTaskCreatePinnedToCore(&core_1_function_UPSI, "UPSI", 50000, NULL, 1, &core_1_task_UPSI, 1);
    // xTaskCreatePinnedToCore(&core_0_function_comms_and_logs, "Comms and Logs", 50000, NULL, 1, &core_0_task_comms_and_logs, 0);
    // xTaskCreatePinnedToCore(&core_0_function_BT_handler, "BT Interrupt", 50000, NULL, 2, &core_0_task_BT_handler, 0);
}

void loop()
{
    delay(1000);
}
```

I have literally commented everything out. All the custom libraries. All the tasks that were declared. Interrupts.

Cleaned up the environment and recompiled everything. But I've only included `Arduino.h` so I guess it should be alright. I STILL have the same problem. But somehow other codes work. 

I'm starting to think this might be a partitioning issue. 
```
board_build.partitions=/home/baaridun/.platformio/packages/framework-arduinoespressif32/tools/partitions/full_storage.bin
board_upload.maximum_size=4194304
```

These are the options I've used to setup my new partition. And I've already shown the partition table...

stickbreaker, 2019-02-28:
Your memory size report talks about 4mB program space, but 0x3f0000 is only 4128768?

stickbreaker, 2019-02-28:
Is it messing up the layout?

baaridunnasr, 2019-02-28:
I just got that. This I believe is the epitome of me being stupid. Thank you for the time. @stickbreaker . Sincerely appreciate your help.

baaridunnasr, 2019-02-28:
@stickbreaker Changed
``` 
board_build.partitions=/home/baaridun/.platformio/packages/framework-arduinoespressif32/tools/partitions/full_storage.bin 
```

```
board_build.partitions=/home/baaridun/.platformio/packages/framework-arduinoespressif32/tools/partitions/full_storage.csv
```
It needed to be a CSV File.
That cleaned up everything.

baaridunnasr, 2019-02-28:
@stickbreaker Sorry to bother you for the last time. But how do I clearly understand errors after decoding stacks? What do I need to look for and where?

closed 2019-02-28 by baaridunnasr

stickbreaker, 2019-02-28:
It is acting like a memory allocation failure. Count up your IRAM_ATTR and global allocations.

stickbreaker, 2019-02-28:
I use me-no-dev's stack decoder under  arduino ide. I dont know how under PIO

stickbreaker, 2019-02-28:
Glad you found it.

baaridunnasr, 2019-02-28:
I have Arduino IDE just to decode stacks using me-no-dev's stack decoder.

```
Decoding stack results
0x40091e54: invoke_abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/panic.c line 155
0x40092085: abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/panic.c line 170
0x40094ee5: xQueueGenericReceive at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/queue.c line 1441
0x401863da: sys_mutex_lock at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/port/esp32/freertos/sys_arch.c line 78
0x401866b6: sys_arch_protect at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/port/esp32/freertos/sys_arch.c line 469
0x4017649c: do_memp_malloc_pool at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/lwip/src/core/memp.c line 302
0x40176505: memp_malloc at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/lwip/src/core/memp.c line 398
0x40187df2: netconn_alloc at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/lwip/src/api/api_msg.c line 742
0x40186bb0: netconn_new_with_proto_and_callback at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/lwip/src/api/api_lib.c line 133
0x40174066: lwip_socket at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/lwip/src/api/sockets.c line 1580
0x400942e5: vPortTaskWrapper at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/port.c line 143
```
This is what I see.

stickbreaker, 2019-02-28:
I would say you exhausted heap.
You had a network socket start a callback, start a memory alloc the die.

mkazimoto, 2019-08-25:
I changed a flash frequency to 40Mhz and the problem is solved.
[image]

tonys0, 2019-10-15:
This actually helped me to solve the problem with continuous reboot.

In Arduino IDE: click Tools, Board, Board Manager. 
After the dialog is loaded type ESP32 and look for:
esp32 by Espressif Systems ...
if it is installed
Press Update button (in my case to version 1.0.4)
if it is not installed, I can not see how you got here to find the continuous reboot problem.
But do not worry, just press Install button.

Try this before you go for 160MHz CPU frequency, 40MHz flash, DIO flash mode, PSRAM disabled etc.
not to mention the resistor and capacitor on EN input.