espressif/arduino-esp32 #2522: mDNS memory leak

AlexTZA, 2019-02-28:
### Hardware:
Board:						Custom board based on ESP32-WROVER
Core Installation version:			1.0.1
IDE name:					Visual Studio (vMicro)
Flash Frequency:				80Mhz
PSRAM enabled:                               yes
Upload Speed:					921600
Computer OS:                                   Windows 10

### Description:

I am trying to use ESPmDNS of v1.0.1 of the Arduino libraries. However, my program periodically queries the mDNS service in case there have been changes to the advertised service. While doing this, I noticed the free heap is constantly reducing in size after the mDNS.queryService call which stops if I comment out this line in my code. I have tried encapsulating the mDNS calls to .begin and .end inside the loop but the "used" RAM is never released which seems problematic, unless I am missing something on the usage side?
In my application, which is more free RAM constrained than the sample below, the device eventually crashes after running out of RAM.

I have reproduced the behaviour in the mDNS-SDExtended example to illustrate the issue.

### Sketch:  
```cpp

/*
  ESP8266 mDNS-SD responder and query sample

  This is an example of announcing and finding services.
  
  Instructions:
  - Update WiFi SSID and password as necessary.
  - Flash the sketch to two ESP8266 boards
  - The last one powered on should now find the other.
 */

#include <WiFi.h>
#include <ESPmDNS.h>
#include "time.h"

const char* ssid     = "ssid";
const char* password = "pwd";

void setup() 
{
    Serial.begin(115200);

    Serial.print("Connecting to Wifi");
    WiFi.begin(ssid, password);
    while (WiFi.status() != WL_CONNECTED) {
        delay(250);
        Serial.print(".");
    }
    Serial.println("");
    Serial.print("Connected to ");
    Serial.println(ssid);
    Serial.print("IP address: ");
    Serial.println(WiFi.localIP());

    if (!MDNS.begin("ESP32_Browser")) {
        Serial.println("Error setting up MDNS responder!");
        while(1){
            delay(1000);
        }
    }
}

void loop() 
{
    Serial.printf("RAM >> Total Heap[%d], Free Heap[%d], Free PSRAM[%d]\r\n", ESP.getHeapSize(),    ESP.getFreeHeap(), ESP.getFreePsram());

    browseService("http", "tcp");
    //delay(1000);
    browseService("arduino", "tcp");
    //delay(1000);
    browseService("workstation", "tcp");
[... 219 lines cut ...]
no services found

Browsing for service _workstation._tcp.local. ... [W][ESPmDNS.cpp:219] queryService(): No results found!
no services found

Browsing for service _smb._tcp.local. ... [W][ESPmDNS.cpp:219] queryService(): No results found!
no services found

Browsing for service _afpovertcp._tcp.local. ... [W][ESPmDNS.cpp:219] queryService(): No results found!
no services found

Browsing for service _ftp._tcp.local. ... [W][ESPmDNS.cpp:219] queryService(): No results found!
no services found

Browsing for service _ipp._tcp.local. ... 1 service(s) found
  1: HPPWCOLOR556 (192.168.0.19:631)

Browsing for service _printer._tcp.local. ... 1 service(s) found
  1:  (0.0.0.0:0)

RAM >> Total Heap[335116], Free Heap[275948], Free PSRAM[4191024]

...etc <Snip>
 
```

me-no-dev, 2019-02-28:
there are fixes for this pending in IDF. next update of the libs this should be fixed :)

atanisoft, 2019-02-28:
@me-no-dev any ideas why total heap went down as well as free heap? I've seen this outside of mDNS and was wondering on it.

closed 2019-03-07 by AlexTZA

g01d10x, 2019-03-10:
@me-no-dev do you know what version of the libraries we should be watching for into Arduino IDE ? Cheers :)