espressif/arduino-esp32 #2524: Unable to connect to WPA2 Enterprise Network

Basstrom87, 2019-03-01:
### Hardware:
Board:							        ESP32 Dev Module
Core Installation version:	1.0.1
IDE name:						        Arduino IDE and Platform.io
Flash Frequency:					  80Mhz?
PSRAM enabled:              no
Upload Speed:					      115200
Computer OS:                MacOS Mojave

### Description:
Can not connect to WPA2 Enterprise Network.

Can see on the RADIUS log it's trying to connect but throws up a EAP Error 23

I'm assuming that it's something to do with the certificate and unsure how it should be implemented so use the certificate or if I have formatted the certificate correctly. There seems to be multiple ways to do it online and unsure which is the correct way.

### Sketch:
```cpp
#include <Arduino.h>
#include <WiFi.h>
#include "esp_wifi.h"
#include "esp_wpa2.h"

#define EAP_IDENTITY "---------"
#define EAP_PASSWORD "---------"
const char* ssid = "SHS_Aruba";

const char* ca_pem = "-----BEGIN CERTIFICATE-----\n"
"MIIFqjCCA5KgAwIBAgIQEdOJY2465aVG9bW90moH4TANBgkqhkiG9w0BAQsFADBH\n"
"MRMwEQYKCZImiZPyLGQBGRYDbGFuMRYwFAYKCZImiZPyLGQBGRYGc2hzd2luMRgw\n"
"FgYDVQQDEw9TYWxpc2J1cnkgSFMgQ0EwHhcNMTYwODMwMDQzNDM4WhcNNDYwODMx\n"
"MDQ0NDM1WjBHMRMwEQYKCZImiZPyLGQBGRYDbGFuMRYwFAYKCZImiZPyLGQBGRYG\n"
"c2hzd2luMRgwFgYDVQQDEw9TYWxpc2J1cnkgSFMgQ0EwggIiMA0GCSqGSIb3DQEB\n"
"AQUAA4ICDwAwggIKAoICAQC1tpr2OB80TDYTyFXoJVbnuw45oSZtFIDOZWjemIhF\n"
"k5UyLX8C8tGlRMlozkVZExSzPyL0Ci/8VHdKpAcJ9rP3R2EAGA7busU1XCiSjNSP\n"
"otLCGO2gFKIItFs/NG/YkgeGjiMhAnfyeo5/6aqZM43raGmhVH8Jt9LH18+QGjKL\n"
"GdDZuhNKtYH+NMT99x6SRuhXB25kOTLt12q0GCJTNT6/g76veIpIfprctJFrCfnC\n"
"VIQGqUuon10nrPLNn/gB5Ke/gLei5NAludEeivoPSQd1UF3dPfiaXVCi2uwcjvB5\n"
"J595XEleN6N+Iw1t0xYtHtJkVF4HZZAFoYbtwfvxrI5YJVPAOcfSPq9joN3z/xQl\n"
"cIKvmpZ/6/A8StsAGviV6qPI/9oKhuWoqT1+qV3nkOPw+BHhVsH2a0Rtm9J1Xi9V\n"
"aUvSKImEGazXNVW+m5U2Hm4S4Q1V3MwGDZ5T5yGfXzLwdFmlUO3/sdTK/vd6+n6+\n"
"XHhWidp9rN5D97G8id6hxNxVGg4HpQI6hA5Y313eMrncKZPtE55tVX2spYCObOsI\n"
"AcskYKqFHLDjqaZEPumlHL5ogliB2STaxt2CgUaeCdKxnuP/nqrdDwV9+DjD1VfU\n"
"RBsY5Zo2WrrRvcOYm+WsAnWTUvq50mGBqntFjVFTeAZt+r4VYhLw+DnNlbRisAkV\n"
"VwIDAQABo4GRMIGOMBMGCSsGAQQBgjcUAgQGHgQAQwBBMA4GA1UdDwEB/wQEAwIB\n"
"hjAPBgNVHRMBAf8EBTADAQH/MB0GA1UdDgQWBBR42TyJivXOFeptm+dVx3C4Kcgw\n"
"sDASBgkrBgEEAYI3FQEEBQIDAQABMCMGCSsGAQQBgjcVAgQWBBTwWdz8Hj8yFN2n\n"
"2wQpTHI8S6bUzTANBgkqhkiG9w0BAQsFAAOCAgEACaTvpp/Xk1bOqfs82n7SWPpR\n"
"uDctAC/zzreTIR8/UJ5p6tzPkAosK57MQLbpodmFKC+M3DvlYDwOv+cNxPj6y+Ji\n"
"bmFKCIZcMjjjnW2ewTGkN1MjCxcjafd0Y8t+Y0xJJy+bNQLQeiuOxWzjDozJST9k\n"
"S675gvTNQQRxbgqgb5gDiTwVOKj20uIab6GJQk+twZ6lZN+XXyoPybh/hGbpeJrS\n"
"uMhbcAt9rAiR9ouDpvDInyt2mXCCtpDqxCM/jUzzsXdaGRCVC18dHj8rMEnIwQCf\n"
"uVres0bU6bxSseU5BWx23ErEglru77bvJ3Y4nEcuP8ODTK5ntuBwY8E9o6uyqe0Q\n"
"cca/rKFecqfHIBckEtoECCWxfzDK1z52MYqsO4R/XfS8L+KJZAe4at7ARcywRhUk\n"
"aa9TvZi8wc8f4rNxOX5mvIh3AZjTFgNGG2/B2gpvdOAPNiDyzmRk3kq18kd0T4WP\n"
"QnDYZ8fSlAu+5BHvwNRxW4NSQwsh+R8A64ovM7p98dctTh0ogkXjRYJyDqxlE1GO\n"
"wDhSBdOM+w5cr5XGYAph5cBpGEgKM0jrtMMO2KiPNtDjJGO1bwYm9Y5J58oSU3LA\n"
"U5Vhq6YlArz8cpziydZw3YkRi9FXwYx+v5Bnaq16FE3MJuVTNxer0l7+FBrEnh0/\n"
"9CcdA/blDouY6bGULa4=\n"
"-----END CERTIFICATE-----\n\0";

void setup() {
  WiFi.mode(WIFI_MODE_STA); 
  esp_log_level_set("*", ESP_LOG_DEBUG);
  
  Serial.begin(115200);
  delay(10);
  Serial.println();
  Serial.print("Connecting to network: ");
  Serial.println(ssid);
[... 210 lines cut ...]
I (372587) wifi: state: init -> auth (b0)
I (372604) wifi: state: auth -> assoc (0)
I (372619) wifi: state: assoc -> run (10)
I (372620) wpa: wpa2_task prio:2, stack:6656

............................................................I (402620) wifi: state: run -> init (200)
I (402621) wifi: new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:357] _eventCallback(): Reason: 204 - HANDSHAKE_TIMEOUT
I (402632) wifi: flush txq
I (402634) wifi: stop sw txq
I (402638) wifi: lmac stop hw txq
I (402648) wifi: mode : sta (24:0a:c4:1c:9b:88)
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 2 - STA_START
.I (402771) wifi: new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
I (402771) wifi: state: init -> auth (b0)
I (402773) wifi: state: auth -> assoc (0)
I (402780) wifi: state: assoc -> run (10)
I (402780) wpa: wpa2_task prio:2, stack:6656

.I (403877) wpa: >>>>>wpa2 FAILED

```

lbernstone, 2019-03-01:
The example in the WiFi library was pretty specifically written around a specific network (Eduroam).  You may want to take a look at the [esp-idf example](https://github.com/espressif/esp-idf/tree/master/examples/wifi/wpa2_enterprise) to see how to use user certs.

Basstrom87, 2019-03-01:
> The example in the WiFi library was pretty specifically written around a specific network (Eduroam). You may want to take a look at the [esp-idf example](https://github.com/espressif/esp-idf/tree/master/examples/wifi/wpa2_enterprise) to see how to use user certs.

Thanks for the quick reply, I only need the Root CA installed, other than that I don't need any certificates as what I'm planning on using them for doesn't require access to outside our network.

I'm attempting now to work through the example and I'll report back.

Ideally I would like to have this in an Arduino sketch form as a template for my students to use these devices.

zeroby0, 2019-03-28:
Hey @Basstrom87, did you figure out how to do it?

martinius96, 2019-04-29:
@Basstrom87 
Hey, I have published some examples for connection to eduroam networks there:
https://github.com/espressif/arduino-esp32/tree/master/libraries/WiFi/examples/WiFiClientEnterprise
And also for multiple IoT clouds, HTTP, MQTT, HTTPS requests examples in my repo: https://github.com/martinius96/ESP32-eduroam

Basstrom87, 2019-04-30:
> @Basstrom87
> Hey, I have published some examples for connection to eduroam networks there:
> https://github.com/espressif/arduino-esp32/tree/master/libraries/WiFi/examples/WiFiClientEnterprise
> And also [...]

Hey Martinius96,
Thanks for the reply. I had tried your WiFiClientEnterprise example, unfortunately it didn't work, won't get past part of the authentication process, pretty sure its to do with the self signed certificate we have here. 
Thankfully the IT department have set up a new VLAN just for devices like the ESP32 with MAC address filtering on it, it doesn't have access to the internet through it but can access what I need it to through it and it's just WPA2 Personal security on it so that's how I'm getting around it.

Thanks again for your suggestions :)

closed 2019-08-14 by stale[bot]

hopkinskong, 2019-08-20:
@Basstrom87 I am using `esp-idf` and I forgot to turn on WPA2 Enterprise in component config. After turning it on, it works flawlessly.

Basstrom87, 2019-08-20:
@hopkinskong
I tried that and still didn’t get any joy 😔

What kind of enterprise network are you trying to connect to?

Jonel-Bacroya, 2020-12-10:
@hopkinskong 
how do you " turn on WPA2 Enterprise in component config" ? 
I'm new to ESP32 and clueless on WPA2. 
I'm getting this error in the radius server:
""EAP-TLS failed SSL/TLS handshake because the client rejected the ISE local-certificate""

TIA!