espressif/arduino-esp32 #2525: Using strtok() crashes the prg

mdathersajjad, 2019-03-01:
Using strok causes the board to crash
Here is the backtrace

0x400d3aa9: strtok at ../../../.././newlib/libc/string/strtok.c line 103
0x400d0b2e: setup() at /Users/balabhaskar/Documents/Arduino/esp32helloworld/esp32helloworld.ino line 6
0x400d16fb: loopTask(void*) at /Users/balabhaskar/Library/Arduino15/packages/esp32/hardware/esp32/1.0.1/cores/esp32/main.cpp line 20
0x4008950d: vPortTaskWrapper at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/port.c line 143

Here is the sample program
`#include<string.h>
void setup() {

  Serial.begin(9600);
  char* variable = "one:two:three\0";
  strtok(variable,":");
  Serial.println(variable);
}

void loop () {
  
}
`

bertmelis, 2019-03-01:
I don't see a semicolon, so `strtok` returns `nullptr` which cannot be printed.

`strtok` doesn't crash you program, you are doing that yourself.

```C++
if (variable) Serial.println(variable);
```

stickbreaker, 2019-03-01:
try this
```c++
void setup() {

  Serial.begin(9600);
  char * variable = "one:two:three\0";
  Serial.printf(" Origonal String =%s\n");
  char * token = strtok(variable,":");
  while( token){
     Serial.println(token);
     token = strtok(NULL,":");
  }
  Serial.printf("all Done!\n");
}

void loop () {}
```

Chuck.

closed 2019-08-15 by stale[bot]

fluppie, 2021-09-27:
I have this working code when using "good" data like "HelloWorld 12 24.7" or "Publish MainsMeter/Current/L1 6.0"
When you use "Publish MainsMeter/Current/L1" or "HelloWorld 12" or even "Hello" it crashes the ESP32. 
On an Arduino Uno/Nano, you don't have this. Any idea why that is?

 ```
const byte numChars = 128;
char receivedChars[numChars];
char tempChars[numChars];        // temporary array for use when parsing
boolean newData = false;

// variables to hold the parsed data
char msgCmd[numChars] = {0};
char msgTopic[numChars] = {0};
char msgPayload[numChars] = {0};

void parseData() {      // split the data into its parts
// HelloWorld 12 24.7
// Publish MainsMeter/Current/L1 6.0

    char * strtokIndx; // this is used by strtok() as an index

    strtokIndx = strtok(tempChars," ");      // get the first part - the string // \040 = SPACE
    strcpy(msgCmd, strtokIndx); // copy it to msgCmd
 
    strtokIndx = strtok(NULL, " "); // this continues where the previous call left off
    //msgTopic = atoi(strtokIndx);     // convert this part to an integer
    strcpy(msgTopic, strtokIndx); // copy it to msgTopic

    strtokIndx = strtok(NULL, " ");
    //msgPayload = atof(strtokIndx);     // convert this part to a float
    strcpy(msgPayload, strtokIndx); // copy it to msgPayload

}
```