espressif/arduino-esp32 #2529: link to Espressif.com not wroking

sixbacon, 2019-03-01:
Is the Espressif link for the ESP32 package to be inserted in the Arduino IDE preferences working?
I have been using the Arduino IDE with an ESP32 but yesterday it stopped recognising the ESP32. The error listed is 
_Error downloading https://dl.espressif.com/dl/package_esp32_index.json_

I have tried reentering the link but no joy. 

I can read the file when I type the link into a browser. I do not know enough to check if there is a problem with the file.

Thanks

me-no-dev, 2019-03-01:
the file itself is OK and has not been changed in a few weeks. Maybe your IDE is acting up?

sixbacon, 2019-03-02:
Thanks for the reply. I reinstalled the Arduino IDE and still got the error. 

I am using a windows 10 machine. I went into AppData\Local\Arduino15 and removed everything with ESP32 in the title. I was then able to reload the board file. Seems to be sorted now.

closed 2019-08-15 by stale[bot]