espressif/arduino-esp32 #2530: SOftap

williamesp2015, 2019-03-01:
Make your question, not a Statement, inclusive. Include all pertinent information:

What you are trying to do?
Describe your system( Hardware, computer, O/S, core version, environment).
Describe what is failing.
Show the shortest possible code that will duplicate the error.
Show the EXACT error message(it doesn't work is not enough).
All of this work on your part shows us that you have worked to solve YOUR problem. The more complete your issue posting is, the more likely someone will volunteer their time to help you.

If you have a Guru Meditation Error or Backtrace, ***please decode it***:
https://github.com/me-no-dev/EspExceptionDecoder

----------------------------- Remove above -----------------------------

### Hardware:
Board:							ESP32 Dev Module 
Core Installation version:			espressif32                     @ 1.7.0 UPTODATED
IDE name:						Arduino Platform.io

Computer OS:                                        Windows 10
PlatformIO 4.0.0a3
### Description:
Wifi

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp
) event: default event handler failed!

me-no-dev, 2019-03-01:
fixed in Master already :)

closed 2019-03-01 by me-no-dev

williamesp2015, 2019-03-01:
I just updated my platformIo code and libraries , even I deleted espriff32 directory but it did not solved

danielfabro, 2019-03-26:
I have the same problem!