espressif/arduino-esp32 #2533: Station not connecting to AP in a special way ( When there is 3 ESP32)

akshar001, 2019-03-01:
It' simple.
I will share two ino files one is small_sta and other is main_ap.
Now to replicate the same problem you atleast need 3 esp32s.
What i am trying to do is make one ( main_ap )esp32 a softAP and other two esp32 STA( small_sta ).
Other two STA esp32 is getting connected to esp32 softAP.
Now the problem arises when the two of the STA esp32 also becomes softAP with the same SSID as MAIN_AP. One of the ESP32 sta can not connect to softap. 
Actually when you will keep log as verbose level you can see the event log in main_ap that it gets connected but dosen't assigning ip to sta.it will be the same for more sta's also.

To exactly replicate my problem try resetting ESP32 main_ap module so you will know the results.
So the problem arises when all of the ESP32 becomes SOFT_AP with SAME SSID. But as you can see my code i have put down the mac address in WiFi.begin(),

 Sometimes it connects to false SOFT_AP. ( Means other mac address softap)

// You will upload main_ap in one esp32 and small_sta in another two esp32.
// Now first run the main_ap coded esp32 you will get the Mac Adress of it in serial monitor.
// Now in small_sta change the mac address to the mac address that you have saved.
// upload this small_sta code to other two esp32s.

I am using macosx. have facing this problem since two months now

akshar001, 2019-03-01:
[ISSUE ESP32.zip](https://github.com/espressif/arduino-esp32/files/2919632/ISSUE.ESP32.zip)

williamesp2015, 2019-03-01:
Do you have these errors in AP Mode after connect to ESP?
E (8854) event: mismatch or invalid event, id=63
E (8855) event: default event handler failed!
It seems new changes in the wifi.cpp that corrected in the Master but so far is persistent

akshar001, 2019-03-02:
No it's not that error which fails from connectinng.
As you can see the log by completley setting up what i have shared.
AP actually gets connected to STA but than there is no event of assigning_ip from both side.
Set the log at verbose level.

closed 2019-08-06 by akshar001