espressif/arduino-esp32 #2535: cannot specify board variant when compiling under esp-idf

ecc1, 2019-03-01:
### Hardware:
Board:							ttgo-lora32-v1
Core Installation version:			1.0.1-git
IDE name:						IDF component

### Description:

I want to include arduino-esp32 as a component in a project using the esp-idf build system. My target board's pins are different than those defined in variants/esp32/pins_arduino.h (I want to use the definitions in variants/ttgo-lora32-v1/pins_arduino.h).

The problem is that only the variants/esp32 directory is added to the include dirs (see line 33 of component.mk and line 176 of CMakeLists.txt). So when, for example, the SPI library is compiled, it is built with the wrong definitions for SS and MOSI, and doesn't work.

I can brute-force this a couple of ways, but I'd prefer a supported way to specify the board variant that is respected by the make and CMake build systems.

ecc1, 2019-03-05:
These are my current workarounds, suggestions for improvements would be welcome.
For the make-based build system:
```
PROJECT_NAME := xyzzy

BOARD_VARIANT := ttgo-lora32-v1

# Make sure board-specific pin definitions are used instead of the ones in variants/esp32
CPPFLAGS += -I $(abspath components/arduino-esp32/variants/${BOARD_VARIANT})

include $(IDF_PATH)/make/project.mk
```
and for the CMake-based build system:
```
cmake_minimum_required(VERSION 3.5)

set(board_variant ttgo-lora32-v1)

get_filename_component(board_variant_dir "components/arduino-esp32/variants/${board_variant}" ABSOLUTE)

include($ENV{IDF_PATH}/tools/cmake/project.cmake)

project(xyzzy)

# Make sure board-specific pin definitions are used instead of the ones in variants/esp32

function(add_board_variant comp)
    target_include_directories(idf_component_${comp} BEFORE PUBLIC ${board_variant_dir})
endfunction()

file(GLOB components components/*)
foreach(path ${components})
    get_filename_component(c ${path} NAME)
    add_board_variant(${c})
endforeach()
```

closed 2019-08-15 by stale[bot]

atanisoft, 2019-08-15:
@me-no-dev this is still an issue that should be resolved for those compiling as component via CMake. In PIO this is covered via the IDF+Arduino support added in the dev version.

Jkallus, 2023-03-30:
This is still an issue I ran into while compiling for ESP32-S3 using arduino-esp32 as a component. The SPI pins are being pulled from variants\esp32\pins_arduino.h instead of variants\esp32\pins_arduino.h

Jkallus, 2023-03-30:
Solution is to set the variable CONFIG_ARDUINO_VARIANT in the sdkconfig via the ESP-IDF Menuconfig's Arduino section.