espressif/arduino-esp32 #2536: iPhone can't connect to wifi server in AP mode

judge2005, 2019-03-02:
### Hardware:
Board:							ESP32-PICO-KIT V4
Core Installation version:			master
IDE name:						Sloeber
Flash Frequency:					?
PSRAM enabled:                                     ?
Upload Speed:					115200
Computer OS:                                         Windows 10

### Description:
I'm using AsyncWifiManager to handle setting up an AP and a captive portal. I am finding that an iPhone is only occasionally redirected to the captive portal. The problem seems to lie with DNS requests - I hve instrumented both DNServer and WiFiUDP. In the cases where connection is established, DNServer responds to a DNS request. In the case where it doesn't, no DNS request is registered. However, I was getting the following printed out in the console when I attempted to connect to the AP (whether or not it succeeded):

E (22491) event: mismatch or invalid event, id=63
E (22491) event: default event handler failed!

So it seems that the iPhone was actually sending a request of some sort.

I updated to the latest (i.e. master) as of March 2, 2019. The event errors go away, but the problem doesn't.

### Sketch:
```cpp
#include "Arduino.h"
#include "WiFi.h"
#include "AsyncTCP.h"
#include "ESPAsyncWebServer.h"
#include "ESPAsyncWiFiManager.h"
#include "config.h"

AsyncWebServer server(80);
DNSServer dns;
AsyncWiFiManager wifiManager(&server,&dns);

String ssid = "ESP32-APTest";

void setup()
{
    Serial.begin(115200);
    Serial.setDebugOutput(true);

	wifiManager.setConnectTimeout(10);
    wifiManager.startConfigPortalModeless(ssid.c_str(), "secretsauce");
}

void loop()
{
	wifiManager.loop();
}
```

### Debug Messages:
```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:928
ho 0 tail 12 room 4
load:0x40078000,len:9400
ho 0 tail 12 room 4
load:0x40080400,len:5928
entry 0x40080698
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 13 - AP_START
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 0 - WIFI_READY
*WM: SET AP STA[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 13 - AP_START
*WM: Connecting as wifi client...
*WM: Try to connect with saved credentials
[E][WiFiSTA.cpp:211] begin(): connect failed!
*WM: Connection timed out
*WM: Connection result: 
*WM: 255
*WM: 
*WM: Configuring access point... 
*WM: ESP32-APTest
*WM: secretsauce
*WM: AP IP address: 
*WM: 192.168.4.1
*WM: HTTP server started
*WM: About to scan()
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 1 - SCAN_DONE
*WM: Scan done
*WM: DUP AP: FiOS-3RA3E

>>>>>> At this point, the ESP32 is waiting for a connection
>>>>>> Above is all that is output when an iPhone tries to connect (usually)
>>>>>> Below is a successful connection from a Windows 10 PC

[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 15 - AP_STACONNECTED
dhcps: send_offer>>udp_sendto result 0
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 17 - AP_STAIPASSIGNED
*WM: Handle not found
*WM: Request redirected to captive portal
*WM: Handle not found
*WM: Request redirected to captive portal
*WM: Handle not found
*WM: Request redirected to captive portal
*WM: Handle root
*WM: About to scan()
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 1 - SCAN_DONE
*WM: Scan done
*WM: DUP AP: FiOS-3RA3E
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 15 - AP_STACONNECTED
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 17 - AP_STAIPASSIGNED
[W][AsyncTCP.cpp:630] _poll(): rx timeout 4
*WM: Handle not found
*WM: Request redirected to captive portal
*WM: Handle not found
*WM: Request redirected to captive portal

```

judge2005, 2019-03-02:
Here's what happens with 1.0.1:

```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:928
ho 0 tail 12 room 4
load:0x40078000,len:9280
load:0x40080400,len:5848
entry 0x40080698
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 13 - AP_START
*WM: SET AP STA[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 13 - AP_START
*WM: Connecting as wifi client...
*WM: Try to connect with saved credentials
[E][WiFiSTA.cpp:211] begin(): connect failed!
*WM: Connection timed out
*WM: Connection result: 
*WM: 255
*WM: 
*WM: Configuring access point... 
*WM: ESP32-APTest
*WM: secretsauce
*WM: AP IP address: 
*WM: 192.168.4.1
*WM: HTTP server started
*WM: About to scan()
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 1 - SCAN_DONE
*WM: Scan done
*WM: DUP AP: FiOS-3RA3E
E (38374) event: mismatch or invalid event, id=63
E (38374) event: default event handler failed!
E (31046) event: mismatch or invalid event, id=63
E (31046) event: default event handler failed!
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400014dc  PS      : 0x00060f30  A0      : 0x800ec4b4  A1      : 0x3ffb8f20  
A2      : 0x070518ff  A3      : 0x070518fb  A4      : 0x000000ff  A5      : 0x0000ff00  
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x00000000  A9      : 0x00000000  
A10     : 0x00000000  A11     : 0x3ffb9122  A12     : 0x00000000  A13     : 0x00000000  
A14     : 0x60000000  A15     : 0x00000000  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x070518ff  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffc  

Backtrace: 0x400014dc:0x3ffb8f20 0x400ec4b1:0x3ffb8f30 0x400f39e2:0x3ffb9240 0x400f3a1a:0x3ffb92d0 0x400dbbaa:0x3ffb9310 0x400d24a2:0x3ffb9370 0x400d27fd:0x3ffb9490 0x4008e74d:0x3ffb94c0

Rebooting...

```
This is the stack trace decoded:
```
PC: 0x400014dc
EXCVADDR: 0x070518ff

Decoding stack results
0x400ec4b1: _svfprintf_r at ../../../.././newlib/libc/stdio/vfprintf.c line 1529
0x400f39e2: _vsnprintf_r at ../../../.././newlib/libc/stdio/vsnprintf.c line 72
0x400f3a1a: vsnprintf at ../../../.././newlib/libc/stdio/vsnprintf.c line 41
0x400dbbaa: log_printf at C:\Users\mpand\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.1\cores\esp32\esp32-hal-uart.c line 473
0x400d24a2: WiFiGenericClass::_eventCallback(void*, system_event_t*) at C:\Users\mpand\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.1\libraries\WiFi\src\WiFiGeneric.cpp line 342
0x400d27fd: _network_event_task(void*) at C:\Users\mpand\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.1\libraries\WiFi\src\WiFiGeneric.cpp line 67
0x4008e74d: vPortTaskWrapper at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/port.c line 143
```
So probably not really saying much, but the console output does demonstrate that network activity occurred.

judge2005, 2019-03-02:
Is this related to https://github.com/espressif/esp-idf/issues/2915?

judge2005, 2019-03-03:
Looks like I closed it by mistake!

stickbreaker, 2019-03-03:
@judge2005 looks like the line that exploded is [Wifigeneric.cpp:342](https://github.com/espressif/arduino-esp32/blob/7df50a97d17b0953ea01cad0355410a66bd8b8b4/libraries/WiFi/src/WiFiGeneric.cpp#L342)
```c++
if(event->event_id < 26) log_d("Event: %d - %s", event->event_id, system_event_names[event->event_id]);
```
base on the crash error, it looks like `event->event_id` value was out of range of the `system_event_names[]` construct.  The `if()` would limit overrange, but not **UnderRange**.  To me it looks like `event_id` evaluated to a negative value.

```c++
#if ARDUHAL_LOG_LEVEL >= ARDUHAL_LOG_LEVEL_DEBUG
const char * system_event_names[] = { "WIFI_READY", "SCAN_DONE", "STA_START", "STA_STOP", 
"STA_CONNECTED", "STA_DISCONNECTED", "STA_AUTHMODE_CHANGE", "STA_GOT_IP",
 "STA_LOST_IP", "STA_WPS_ER_SUCCESS", "STA_WPS_ER_FAILED", "STA_WPS_ER_TIMEOUT",
 "STA_WPS_ER_PIN","AP_START", "AP_STOP", "AP_STACONNECTED",
 "AP_STADISCONNECTED", "AP_STAIPASSIGNED", "AP_PROBEREQRECVED", "GOT_IP6",
 "ETH_START", "ETH_STOP", "ETH_CONNECTED", "ETH_DISCONNECTED",
 "ETH_GOT_IP", "MAX"};
#endif
```
Chuck.

judge2005, 2019-03-03:
That was on v1.0.1, which doesn't have any range checking there at all. I added +ve and -ve range checking, the result is that there is nothing printed to the console with this version after:
```
E (31046) event: mismatch or invalid event, id=63
E (31046) event: default event handler failed!
```

judge2005, 2019-03-03:
Talk about timing. The latest build works fine, other than the Pico refuses to start up properly on power-on. Closing...

closed 2019-03-03 by judge2005