espressif/arduino-esp32 #2537: Hostname not sent via DHCP request

adydeac, 2019-03-02:
### Hardware:
Board:							ESP32 Dev Module
Core Installation version:			1.0.1
IDE name:						Arduino IDE
Flash Frequency:					80Mhz
PSRAM enabled:                                    no
Upload Speed:					115200
Computer OS:                                        Debian

### Description:
Hi,
I've noticed that somehow (I guess recently?) the hostname is not sent when making DHCP calls:
# tcpdump -Anev -s 1024 -i br-lan port 67 or port 68 or port 69
tcpdump: listening on br-lan, link-type EN10MB (Ethernet), capture size 1024 bytes
01:22:43.011475 30:ae:a4:14:3f:e0 > ff:ff:ff:ff:ff:ff, ethertype IPv4 (0x0800), length 350: (tos 0x0, ttl 255, id 0, offset 0, flags [none], proto UDP (17), length 336)
    0.0.0.0.68 > 255.255.255.255.67: BOOTP/DHCP, Request from 30:ae:a4:14:3f:e0, length 308, xid 0xaedfb4c4, Flags [none]
	  Client-Ethernet-Address 30:ae:a4:14:3f:e0
	  Vendor-rfc1048 Extensions
	    Magic Cookie 0x63825363
	    DHCP-Message Option 53, length 1: Request
	    MSZ Option 57, length 2: 576
	    Requested-IP Option 50, length 4: 172.16.0.195
	    Parameter-Request Option 55, length 12: 
	      Subnet-Mask, Default-Gateway, BR, Domain-Name-Server
	      Domain-Name, Netbios-Name-Server, Netbios-Node, Netbios-Scope
	      Router-Discovery, Static-Route, Classless-Static-Route, Vendor-Option
E..P.................D.C.<.}............................0...?...........................................................................................................................................................................................................c.Sc5..9..@2.....7......,./.!y+.........................................
01:22:43.014082 e8:de:27:6d:29:35 > 30:ae:a4:14:3f:e0, ethertype IPv4 (0x0800), length 342: (tos 0x0, ttl 64, id 7494, offset 0, flags [none], proto UDP (17), length 328)
    172.16.0.1.67 > 172.16.0.195.68: BOOTP/DHCP, Reply, length 300, xid 0xaedfb4c4, Flags [none]
	  Your-IP 172.16.0.195
	  Server-IP 172.16.0.2
	  Client-Ethernet-Address 30:ae:a4:14:3f:e0
	  sname "tftp"
	  file "pxelinux.0"
	  Vendor-rfc1048 Extensions
	    Magic Cookie 0x63825363
	    DHCP-Message Option 53, length 1: ACK
	    Server-ID Option 54, length 4: 172.16.0.1
	    Lease-Time Option 51, length 4: 259200
	    RN Option 58, length 4: 129600
	    RB Option 59, length 4: 226800
	    Subnet-Mask Option 1, length 4: 255.255.255.0
	    BR Option 28, length 4: 172.16.0.255
	    Default-Gateway Option 3, length 4: 172.16.0.1
	    Domain-Name-Server Option 6, length 4: 172.16.0.1
	    Domain-Name Option 15, length 3: "lan"
E..H.F..@..{.........C.D.4Z*............................0...?...........tftp............................................................pxelinux.0......................................................................................................................c.Sc5..6.....3.....:....@;...u...........................lan....

This is the on the same router, but with an request from an android phone:

I am using the ESP-WROOM-32 chipset and Arduino IDE on Linux.

The code snippet is:

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

//Change the code below by your sketch
#include <Arduino.h>

void setup() {
  WiFi.begin(wifiSSID, wifiPass);
  nodeName = "shower-" + WiFi.macAddress();
  nodeName.replace(":", "");
  char _nodeName[20]; nodeName.toCharArray(_nodeName, 20);
  WiFi.setHostname(_nodeName);
  WiFi.mode(WIFI_STA);
}

void loop() {
}
```

luisgcu, 2019-03-07:
@adydeac  we have detected the same issue in project where we use a web server application,  I think we detected  the problem because  we  have several ESP32 in the same wifi network , we can't access the esp32's web-server application and we  can't ping them on the network until  they are connected  some time on the network(3-5minutes)  and we believe that is caused because the esp32 exposes all the same host name "espressif"  ..  Anyone that can guide us on how to properly set the ESP32 host names ?

lbernstone, 2019-03-07:
This probably should be looked at upstream. https://github.com/espressif/esp-idf/blob/master/components/lwip/port/esp32/netif/wlanif.c#L205.  If you add this file into your sketch folder, you could write some code to give each machine a unique name.

beegee-tokyo, 2019-03-07:
Not a 100% sure if that is what you are looking for, but I give my ESP32's different names with
```
char _hostname[20];
uint8_t mac[6];
WiFi.macAddress(mac);
sprintf(_hostname, "esp32-%02x%02x%02x%02x%02x%02x", 
          mac[0], mac[1], mac[2], mac[3], mac[4], mac[5]);
MDNS.begin(_hostname);
```

atanisoft, 2019-03-07:
You can set the hostname before calling wifi.begin(). You just need to set the mode first. Then you should get a unique hostname.

atanisoft, 2019-03-07:
@beegee-tokyo that only sets the mDNS advertised hostname and not the wifi hostname that shows up on the router. But that is good practice to set a unique hostname.

adydeac, 2019-03-07:
As you can see from the tcpdump on the router (running openwrt, the gateway, the only DHCP server in the network), there is no hostname sent. Please see tha ascii dump as well. No "espressif" or anything. Something has changed maybe in the latest code? Or we need to enable the hostname sending?
@beegee-tokyo setting up the mac in that way will definitely screw things up on your network in certain cases:
https://en.wikipedia.org/wiki/MAC_address#Universal_vs._local

asetyde, 2019-03-07:
also with 1.0.2 RC1 ?

adydeac, 2019-03-07:
Same.

[before update]
Thu Mar  7 11:07:45 2019 daemon.info hostapd: wlan0: STA 30:ae:a4:14:3f:e0 IEEE 802.11: authenticated
Thu Mar  7 11:07:45 2019 daemon.info hostapd: wlan0: STA 30:ae:a4:14:3f:e0 IEEE 802.11: associated (aid 2)
Thu Mar  7 11:07:45 2019 daemon.info hostapd: wlan0: STA 30:ae:a4:14:3f:e0 WPA: pairwise key handshake completed (RSN)
Thu Mar  7 11:07:45 2019 daemon.info dnsmasq-dhcp[21458]: DHCPREQUEST(br-lan) 172.16.0.195 30:ae:a4:14:3f:e0 
Thu Mar  7 11:07:45 2019 daemon.info dnsmasq-dhcp[21458]: DHCPACK(br-lan) 172.16.0.195 30:ae:a4:14:3f:e0 
Thu Mar  7 11:08:01 2019 cron.info crond[893]: crond: USER root pid 22271 cmd /scripts/check_snmpd

[after update]
Thu Mar  7 11:09:29 2019 daemon.info hostapd: wlan0: STA 30:ae:a4:14:3f:e0 IEEE 802.11: authenticated
Thu Mar  7 11:09:29 2019 daemon.info hostapd: wlan0: STA 30:ae:a4:14:3f:e0 IEEE 802.11: associated (aid 2)
Thu Mar  7 11:09:29 2019 daemon.info hostapd: wlan0: STA 30:ae:a4:14:3f:e0 WPA: pairwise key handshake completed (RSN)
Thu Mar  7 11:09:29 2019 daemon.info dnsmasq-dhcp[21458]: DHCPREQUEST(br-lan) 172.16.0.195 30:ae:a4:14:3f:e0 
Thu Mar  7 11:09:29 2019 daemon.info dnsmasq-dhcp[21458]: DHCPACK(br-lan) 172.16.0.195 30:ae:a4:14:3f:e0 

The code:
void startWiFi() {
  WiFi.begin(wifiSSID, wifiPass);
  nodeName = "shower-" + WiFi.macAddress();
  nodeName.replace(":", "");
  char _nodeName[20]; nodeName.toCharArray(_nodeName, 20);
#ifdef DEBUG
  Serial.print("WiFi: MAC: "); Serial.println(WiFi.macAddress());
  Serial.print("WiFi: setting hostname "); Serial.print(nodeName); Serial.println("...");
#endif
  WiFi.setHostname(_nodeName);
  tcpip_adapter_set_hostname(TCPIP_ADAPTER_IF_STA, _nodeName);
  WiFi.mode(WIFI_STA);
}

Debug messages:
WiFi: MAC: 30:AE:A4:14:3F:E0
WiFi: setting hostname shower-30AEA4143FE0...

atanisoft, 2019-03-07:
Move wifi.mode and wifi.sethostname before WiFi.begin. the call to tcpip adapter shouldn't be necessary as that is done inside the wifi library

adydeac, 2019-03-08:
WiFi.mode(WIFI_STA);
  nodeName = "shower-" + WiFi.macAddress();
  nodeName.replace(":", "");
  char _nodeName[20]; nodeName.toCharArray(_nodeName, 20);
#ifdef DEBUG
  Serial.print("WiFi: MAC: "); Serial.println(WiFi.macAddress());
  Serial.print("WiFi: setting hostname "); Serial.print(nodeName); Serial.println("...");
#endif
  WiFi.setHostname(_nodeName);
  //tcpip_adapter_set_hostname(TCPIP_ADAPTER_IF_STA, _nodeName);
  WiFi.begin(wifiSSID, wifiPass);

Same:
Fri Mar  8 09:37:31 2019 daemon.info hostapd: wlan0: STA 30:ae:a4:14:3f:e0 IEEE 802.11: authenticated
Fri Mar  8 09:37:31 2019 daemon.info hostapd: wlan0: STA 30:ae:a4:14:3f:e0 IEEE 802.11: associated (aid 2)
Fri Mar  8 09:37:31 2019 daemon.info hostapd: wlan0: STA 30:ae:a4:14:3f:e0 WPA: pairwise key handshake completed (RSN)
Fri Mar  8 09:37:31 2019 daemon.info dnsmasq-dhcp[21458]: DHCPREQUEST(br-lan) 172.16.0.195 30:ae:a4:14:3f:e0 
Fri Mar  8 09:37:31 2019 daemon.info dnsmasq-dhcp[21458]: DHCPACK(br-lan) 172.16.0.195 30:ae:a4:14:3f:e0 

Probably you have noticed that not even the default "espressif" hostname is sent...

lbernstone, 2019-03-08:
I don't know if toCharArray is mangling it somehow,but I have definitely seen invalid names (non-printable chars) come through while watching a tcpdump on the router.  This is working for me:
```
  Serial.begin(115200);
  char hname[19]; // 5+12+1 - don't forget the \0
  snprintf(hname, 19, "ESP32-%012llX", ESP.getEfuseMac());  
  Serial.println(hname);
  WiFi.disconnect();
  WiFi.mode(WIFI_STA);
  WiFi.setHostname(hname);
  WiFi.begin("ssid","passwd");
  WiFi.waitForConnectResult();
  Serial.print("IP Address: ");
  Serial.println(WiFi.localIP());
```

adydeac, 2019-03-10:
Same result:
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1100
load:0x40078000,len:10312
load:0x40080400,len:6460
entry 0x400806a4
ESP32-E03F14A4AE30
IP Address: 172.16.0.195

# tcpdump -Anev -s 1024 -i br-lan port 67 or port 68 or port 69
10:15:57.942822 30:ae:a4:14:3f:e0 > ff:ff:ff:ff:ff:ff, ethertype IPv4 (0x0800), length 350: (tos 0x0, ttl 255, id 0, offset 0, flags [none], proto UDP (17), length 336)
    0.0.0.0.68 > 255.255.255.255.67: BOOTP/DHCP, Request from 30:ae:a4:14:3f:e0, length 308, xid 0x31888ad2, Flags [none]
	  Client-Ethernet-Address 30:ae:a4:14:3f:e0
	  Vendor-rfc1048 Extensions
	    Magic Cookie 0x63825363
	    DHCP-Message Option 53, length 1: Request
	    MSZ Option 57, length 2: 576
	    Requested-IP Option 50, length 4: 172.16.0.195
	    Parameter-Request Option 55, length 12: 
	      Subnet-Mask, Default-Gateway, BR, Domain-Name-Server
	      Domain-Name, Netbios-Name-Server, Netbios-Node, Netbios-Scope
	      Router-Discovery, Static-Route, Classless-Static-Route, Vendor-Option
E..P.................D.C.<].....1.......................0...?...........................................................................................................................................................................................................c.Sc5..9..@2.....7......,./.!y+.........................................
10:15:57.945361 e8:de:27:6d:29:35 > 30:ae:a4:14:3f:e0, ethertype IPv4 (0x0800), length 342: (tos 0x0, ttl 64, id 7492, offset 0, flags [none], proto UDP (17), length 328)
    172.16.0.1.67 > 172.16.0.195.68: BOOTP/DHCP, Reply, length 300, xid 0x31888ad2, Flags [none]
	  Your-IP 172.16.0.195
	  Server-IP 172.16.0.2
	  Client-Ethernet-Address 30:ae:a4:14:3f:e0
	  sname "tftp"
	  file "pxelinux.0"
	  Vendor-rfc1048 Extensions
	    Magic Cookie 0x63825363
	    DHCP-Message Option 53, length 1: ACK
	    Server-ID Option 54, length 4: 172.16.0.1
	    Lease-Time Option 51, length 4: 259200
	    RN Option 58, length 4: 129600
	    RB Option 59, length 4: 226800
	    Subnet-Mask Option 1, length 4: 255.255.255.0
	    BR Option 28, length 4: 172.16.0.255
	    Default-Gateway Option 3, length 4: 172.16.0.1
	    Domain-Name-Server Option 6, length 4: 172.16.0.1
	    Domain-Name Option 15, length 3: "lan"
E..H.D..@..}.........C.D.4Z*....1.......................0...?...........tftp............................................................pxelinux.0......................................................................................................................c.Sc5..6.....3.....:....@;...u...........................lan....

I am using the latest dev libraries.

Am I doing something wrong? Is my board broken in any way?

rpannekoek, 2019-03-10:
I'm having the same issue (using Wemos LOLIN D32 board).  I have also tried every imaginable position of doing the setHostName call:
* After `Wifi.mode(WIFI_STA)` but before `WiFi.begin`
* After `WiFi.begin`
* Using an event handler (`WiFi.onEvent`) which triggers on the `STA_START` event

Whatever I try, the DHCP request never contains a Host Name (DHCP Option 12) in the DHCP Request (unlike other devices, including ESP8266 boards which do properly set their hostname).

As a result, the router (FritzBox) lists the WiFi device as "espressif" instead of the desired hostname  (it's not clear to me where this "espressif" name comes from, because I don't see that anywhere in the DHCP Request either) and the built-in DNS server does not create a record for the hostname.

rpannekoek, 2019-03-10:
Another (less important) issue: I'm creating libraries which work on both ESP8266 and ESP32.  The API for setting hostname is currently different between ESP8266 and ESP32:
* ESP8266:  `WiFi.hostname()`  should be called before `WiFi.begin`
* ESP32: `WiFi.setHostName()`  unclear where it should be called, but most reports mention *after* `WiFi.begin`

atanisoft, 2019-03-10:
@adydeac this may need to be moved to https://github.com/espressif/esp-idf as the DHCP implementation is implemented there (in the lwip stack actually).

adydeac, 2019-03-10:
I'm not very familiar with esp-idf project. Is it another fork? Is it an substitute? Is it part of this project?

atanisoft, 2019-03-10:
> I'm not very familiar with esp-idf project. Is it another fork? Is it an substitute? Is it part of this project?

It provides most of the underlying functionality of this project.

maxdd, 2019-03-14:
As Ibernstone suggested i was able to make it work like this
```
    char hname[19];
    snprintf(hname, 12, "ESP%d-LIGHT", 32);
    WiFi.disconnect();
    WiFi.mode(WIFI_STA);
    WiFi.setHostname(hname);
    WiFi.begin(WIFI_SSID, WIFI_PSK);

```
It really seems to be related to '\0'.
I've also tried like this 
```
char hname[12] = "ESP32-LIGHT";
hname[11] = '\0';
```
and it also works

stickbreaker, 2019-03-14:
@maxdd you should not need this two step process to declare a string:
```c++
char hname[12] = "ESP32-LIGHT";
hname[11] = '\0';
```
this is all you need:
```c++
char hname[] = "ESP32-LIGHT"; // the compiler will append a null automagically. 
```

Your declaration worked because the compiler did append a null, else it would complain about type mismatch.  If you changed the declaration to 13 characters (`char hname[13]`) the compiler would have complained because of a type mis-match. eleven characters between the quotes and the automagic null is only 12, so you get a type mismatch.

Chuck.

maxdd, 2019-03-15:
That's curious, last time a tried your "normal/standard" solution it didnt work.
Now it does, probably i overlooked something due to the late hours.

rpannekoek, 2019-03-17:
Still no success for me if I perform the `WiFi` calls in the sequence given by @maxdd 
BTW, `WiFi.disconnect()` returns `false` if I execute it before `WiFi.mode(WIFI_STA)`,  which makes sense if I look at its implementation (it does basically nothing in that case):

lbernstone, 2019-03-17:
WiFi.disconnect is there to make sure the device releases its lease (and clears the info out of nvs).  hostname is not given (or requested) on a lease renewal/validation.  Both the esp32 and the dhcp server need to see this as a new event.  Ultimately, it is up to the dhcp server to report new dns names, so there will be many situations where this will never work.

atanisoft, 2019-03-17:
Wifi.mode() is required for the disconnect method to do any clearing of NVS IIRC

nightvisi0n, 2019-03-28:
**I could successfully reproduce that this is a real bug and I created a PR in order to fix it: espressif/esp-lwip#6**

*The underlying issue has nothing to do with any `\0` escaping (just because this was told to be a fix in some comments above) and is rather a different bug/side-effect.*

This bug is a little bit tricky, because it doesn't appear in every operational state. If the in-memory stored previous dhcp lease is e.g. no longer valid there will also be a fresh dhcp request (and hostname propagation). That's maybe the reason why it worked for some of you "sometimes".

@luisgcu @rpannekoek You can verify if you are really experiencing the same bug by erasing the complete flash of the esp32 (which in turn clears any previous dhcp lease, leading to a fresh dhcp lease and an updated hostname). This can be done e.g. with [esptool](https://github.com/espressif/esptool#erase-flash-erase_flash--erase-region) or [platformio](https://docs.platformio.org/en/latest/userguide/cmd_run.html#cmdoption-platformio-run-t) (`pio run -t erase`).

jeroenst, 2019-07-04:
I can confirm this bug.

dnsmasq logging with ESP8266:
Jul  4 20:53:53 server dnsmasq-dhcp[940]: 1777019338 DHCPACK(enp3s0.2) 192.168.10.143 84:f3:eb:58:29:63 ESP_SMARTMETER

dnsmasq logging with ESP32:
Jul  4 20:51:58 server dnsmasq-dhcp[940]: 758037265 DHCPACK(enp3s0.2) 192.168.10.106 24:0a:c4:96:48:ac

See the missing hostname in the logging of dnsmasq.... Hostname is set for sure btw.

jeroenst, 2019-07-04:
WORKAROUND:

Add       
WiFi.config(INADDR_NONE, INADDR_NONE, INADDR_NONE);
WiFi.setHostname(_hostname.c_str());
before
WiFi.begin();

I hope issue will be resolved soon.

evert-arias, 2019-07-04:
The workaround suggested by @jeroenst worked for me!

BozoTClown, 2019-07-20:
After working on this issue for over 4 hours @jeroenst suggestion work for me also. My ESP8266 was not this difficult to do. Thank you very much.

WiFi.config(INADDR_NONE, INADDR_NONE, INADDR_NONE);
WiFi.setHostname(mqtt_device_id);
WiFi.begin(ssid, password);

stacklover, 2019-08-01:
I just resolved same issue by using @jeroenst suggestion! Sigh!

nightvisi0n, 2019-09-30:
Veto. This is a real bug and there is an open PR to fix it (see https://github.com/espressif/arduino-esp32/issues/2537#issuecomment-477812782)

jeroenst, 2019-11-29:
It's not solved yet, keep open please.

atanisoft, 2019-11-29:
@jeroenst Until this is solved upstream by ESP-IDF it won't be solved here.

jeroenst, 2019-11-29:
ok, thank you

nightvisi0n, 2019-11-29:
> **I could successfully reproduce that this is a real bug and I created a PR in order to fix it: [espressif/esp-lwip#6](https://github.com/espressif/esp-lwip/pull/6)**
>
> _The underlying issue has [...]

It is already solved, but nobody is interested in my PR 😕

atanisoft, 2019-11-29:
> It is already solved, but nobody is interested in my PR

I don't think they are uninterested in it but perhaps it has gone unnoticed. @me-no-dev can you raise this to the respective devs?

jeremypoulter, 2019-12-12:
I have been looking into this issue and the change in espressif/esp-lwip#6 looks to be already included in the ESP IDF 3.3 and indeed building using that IDF rather than the prebuilt IDF (I am using Platform IO) looks to solve this issue for me.

jeremypoulter, 2020-01-23:
Ok, so I was wrong in my above comment, espressif/esp-lwip#6 is not in the IDF 3.3, I must have looked at the wrong function, however it was indeed working for me when I built using the IDF+Arduino vs vanilla Arduino. 

I have now been able to trace this down to the `CONFIG_LWIP_DHCP_RESTORE_LAST_IP` option being enabled in the Arduino's `sdkconfig`. With this option enabled the IP address is stored in flash and restored on reboot, causing `dhcp_reboot` (which is the function fixed with espressif/esp-lwip#6) to be called. 

With `CONFIG_LWIP_DHCP_RESTORE_LAST_IP` disabled, that code is skipped causing `dhcp_discover` to be used instead, which correctly sets the hostname. 

@me-no-dev would you be opposed to disabling `CONFIG_LWIP_DHCP_RESTORE_LAST_IP` in the Arduino `sdkconfig` until espressif/esp-lwip#6 is merged?

jeremypoulter, 2020-01-23:
FYI here are the relevant bits of code:

https://github.com/espressif/esp-lwip/blob/release/v3.3/src/core/ipv4/dhcp.c#L850:L854
https://github.com/espressif/esp-lwip/blob/release/v3.3/src/core/ipv4/dhcp.c#L932
https://github.com/espressif/esp-idf/blob/release/v3.3/components/lwip/port/esp32/include/lwipopts.h#L232

mvadu, 2020-02-17:
> I have now been able to trace this down to the `CONFIG_LWIP_DHCP_RESTORE_LAST_IP` option being enabled in the Arduino's `sdkconfig`. With this option enabled the IP address is stored in flash and [...]

I think you are on to something.. I followed this [folder structure in Platform.io doc](https://docs.platformio.org/en/latest/platforms/espressif32.html#ulp-coprocessor-programming) and added an edited copy of sdkconfig.h with `#define CONFIG_LWIP_DHCP_RESTORE_LAST_IP 0` but still my DHCP server is not getting hostname as part of DHCP request.

```
Feb 16 23:33:19 dnsmasq-dhcp[260]: DHCPREQUEST(br0) a.b.c.d x:xx:xx:xx:xx:xx 
Feb 16 23:33:19 dnsmasq-dhcp[260]: DHCPACK(br0) a.b.c.d x:xx:xx:xx:xx:xx 
``

jeremypoulter, 2020-02-17:
Unfortunately you have to rebuild the IDF. Fortunatly with Platform IO this is really easyjust change the framework to 

```
framework = arduino, espidf
```
see https://github.com/OpenEVSE/ESP32_WiFi_V3.x/blob/master/platformio.ini#L205 and you can also see my [sdkconfig.h](https://raw.githubusercontent.com/OpenEVSE/ESP32_WiFi_V3.x/master/src/sdkconfig.h)

danielfaust, 2020-02-21:
The suggestion from @jeroenst worked for me, **but** `WiFi.disconnect(true)` does reset the hostname.

In my case, I have a reconnect loop which used to contain
```
WiFi.disconnect(true);
WiFi.begin(ssid, password);
```
and `setHostname` works if it is called in the following way
```
WiFi.disconnect(true);
WiFi.config(INADDR_NONE, INADDR_NONE, INADDR_NONE);
WiFi.setHostname(device);
WiFi.begin(ssid, password);
```
as well as
```
WiFi.disconnect(true);
WiFi.config(INADDR_NONE, INADDR_NONE, INADDR_NONE);
WiFi.begin(ssid, password);
WiFi.setHostname(device);
```
and
```
WiFi.disconnect(true);
WiFi.begin(ssid, password);
WiFi.config(INADDR_NONE, INADDR_NONE, INADDR_NONE);
WiFi.setHostname(device);
```

Swapping the order between `WiFi.config` and `WiFi.setHostname` sets the hostname to `espressif`, not calling `WiFi.config` results in a blank hostname.

So calling `WiFi.config` is a requirement before ``WiFi.setHostname`, and `WiFi.disconnect(true)` resets the effect of `WiFi.config`.

mvadu, 2020-02-23:
I can confirm it too.. With `WiFi.config(INADDR_NONE, INADDR_NONE, INADDR_NONE);` before `WiFi.begin(ssid, password);` works as expected, and on DHCP side I see these requests now
```
23 00:26:32 dnsmasq-dhcp[260]: DHCPDISCOVER(br0) xx:xx:xx:xx:xx:xx 
Feb 23 00:26:32 dnsmasq-dhcp[260]: DHCPOFFER(br0) a.b.c.d xx:xx:xx:xx:xx:xx 
Feb 23 00:26:32 dnsmasq-dhcp[260]: DHCPREQUEST(br0) a.b.c.d xx:xx:xx:xx:xx:xx 
Feb 23 00:26:32 dnsmasq-dhcp[260]: DHCPACK(br0) a.b.c.d xx:xx:xx:xx:xx:xx HostName
```

eoun, 2020-04-22:
Hello

I am trying to connect my ESP32 to my HASSIO MQTT Server.

The Problem I have is that when I connect to my MQTT Server after a short time, the connection (to WIFI) cuts.

I saw on my Router that there was so Name. I fixed this with "iFi.config(INADDR_NONE, INADDR_NONE, INADDR_NONE);". Now the Names shows up. I had hoped that yould fix my problem. But sadly no.

The interresting thing is that this had worked befor. I have switch from the Router that my Internet provieder gave me to a "Netgear RAX40", Now it stoped working. Also interresting is that all of my "normal" ESP8266 are still working normal. 

When I only connected to my Wifi all works fine. But as soon as I add to my code the part where I connect to my MQTT it stoped working.

Any Idee? Do I have to set something in the Router?

kaichunchou, 2020-06-21:
The workaround suggested by @jeroenst worked, but I also use `Wifi.config();` to set up static IP. Has anyone successfully set up both static IP and hostname at the same time?

vshymanskyy, 2020-10-20:
Wake up, Neo.

mvadu, 2020-12-20:
knock knock

closed 2021-02-03 by me-no-dev

SuGlider, 2022-04-29:
Check PR #6659