espressif/arduino-esp32 #2540: WiFiClient::read does not handle EAGAIN

OttoWinter, 2019-03-03:
### Hardware:
Board:							nodemcu-32s / any
Core Installation version:			1.0.1-git
IDE name:						platformio
Flash Frequency:					80MHz
PSRAM enabled:                                    no
Upload Speed:					115200
Computer OS:                                        macOS

### Description:
Describe your problem here

Recently I tried upgrading my application to v1.0.1 and latest version from git, but ever since some calls to `read()` in WiFiClient were breaking with return code -1 (example code below). This would oc be expected when not enough data is available but my code explicitly checked for that, and only read data in when enough was available. Thus, a call to `read()` should never fail (unless socket closes).

With some more logging, I found out it was this line that breaks things:

https://github.com/espressif/arduino-esp32/blob/7df50a97d17b0953ea01cad0355410a66bd8b8b4/libraries/WiFi/src/WiFiClient.cpp#L74

Even though enough data is already available, this call sometimes returns EAGAIN (11 = EWOULDBLOCK) when pumping lots of data through the interface (in this case OTA).

Simply adding a few retries to the `read` call fixed things, but ultimately I think the fix should be in the core.

### Sketch:

```cpp
WiFiClient client;

void setup() {
  // setup wifi client
}
void loop() {
  // pump a bunch of data through the client
  int avail = client.available();
  if (avail < 1024) { yield(); return; }
  uint8_t buf[2019];
  int res = client.read(buf, 1024);
  if (res != avail) {
    ESP_LOGE("main", "read failed: %d", res);
    // message: "read failed: -1", errno is 11 (EAGAIN)
    // note: if we retry the call after a short delay it works again
  }
}
```

### Fix

I think this behavior is not expected for the usual user of this project - at least I was under the impression that calling available before all reads would put me on the safe side.

I'd like to contribute a fix for this, but I'm not sure where - the could would have to block _somewhere_, but I'm not sure where. Suggestions?

CC: @me-no-dev (`git blame`)

starlino, 2019-04-06:
I am getting a similar error when sending continuous data to the device
[D][ICWebSocketEvent.h:37] webSocketEvent(): [0] WStype_BIN length: 4096
[D][ICWebSocketEvent.h:37] webSocketEvent(): [0] WStype_BIN length: 4096
[D][ICWebSocketEvent.h:37] webSocketEvent(): [0] WStype_BIN length: 4096
[D][ICWebSocketEvent.h:37] webSocketEvent(): [0] WStype_BIN length: 4096
[D][ICWebSocketEvent.h:37] webSocketEvent(): [0] WStype_BIN length: 4096
[D][ICWebSocketEvent.h:37] webSocketEvent(): [0] WStype_BIN length: 4096
[D][ICWebSocketEvent.h:37] webSocketEvent(): [0] WStype_BIN length: 4096
[E][WiFiClient.cpp:454] flush(): 11
D][ICWebSocketEvent.h:13] webSocketEvent(): [0] Disconnected!

Error seems to be here  https://github.com/espressif/arduino-esp32/blob/7df50a97d17b0953ea01cad0355410a66bd8b8b4/libraries/WiFi/src/WiFiClient.cpp#L452

Shouldn't this be handled properly rather than brutally disconnecting the client ???

h2zero, 2019-04-25:
Noticed this issue over the last week as well when testing MQTT with qos 2. Sending packets would fail while waiting to read an ack response. 

I think the best way to handle this is the same way the write() function works, a retry loop with a timer and counter. I’ll try to hack up some code tonight and submit a PR for testing.

closed 2019-08-14 by stale[bot]