espressif/arduino-esp32 #2541: Guru Meditation Error: Core  0 panic'ed (InstrFetchProhibited)

GeorgeFlorian, 2019-03-04:
### Hardware:

Board: ESP32 DEVKIT1
Core Installation version: 1.0.1-git [this one](https://github.com/espressif/arduino-esp32)
IDE name: Arduino IDE
Flash Frequency: 80Mhz
PSRAM enabled: no
Upload Speed: 921600
Computer OS: Linux Mint 19.1 Mate

Hello !

I am trying to check for long presses of a button but I keep getting `Guru Meditation Error: Core  0 panic'ed (InstrFetchProhibited)` .

I thought of a simple system using the delay function `delay();`. I always get this error. The ESP is in AP Mode. But I can't do nothing with him because it keeps restarting because of the error.

```
#include <Arduino.h>
#include <WiFi.h>
#include <ESPAsyncWebServer.h>
#include <SPIFFS.h>

#define LED 2
#define PRESSED LOW
#define NOT_PRESSED HIGH

int apFlag = 1;

typedef struct Buttons {
  const byte pin = 27;
  const int debounce = 10;
  unsigned long counter = 0;
  bool previousState = NOT_PRESSED;
  bool currentState;
} Button; 

Button pushButton;

void setup() {  
  Serial.begin(115200);
  delay(2000);
  pinMode(LED, OUTPUT);
  pinMode(pushButton.pin,INPUT_PULLUP);
  digitalWrite(LED,LOW);
  //...rest of code
}

void handlePress() {
  apFlag = 1;
  //--- Do something..
}

void loop() {
  digitalWrite(LED, HIGH);
  delay(2000);
  Serial.println((String)"apFlag = " + apFlag);

  if(digitalRead(pushButton.pin) == PRESSED && apFlag == 2) {
    digitalWrite(LED, LOW);
    delay(500);
    digitalWrite(LED, HIGH);
    delay(2000);
  }
  if(digitalRead(pushButton.pin) == PRESSED && apFlag == 2) {
    digitalWrite(LED, LOW);
    delay(2000);
    handlePress();
  }
  digitalWrite(LED, LOW);
  delay(500);
  if(apFlag == 2) {ESP.restart();}
}
```
## This is one output:
### It can output more than one `apFlag = 1' and it doesn't always results in error. But it is unreliable.

```
apFlag = 1
Guru Meditation Error: Core  0 panic'ed (InstrFetchProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x00000000  PS      : 0x00060e30  A0      : 0x800fa3dd  A1      : 0x3ffb5070  
A2      : 0x3ffcba68  A3      : 0x3ffcbcbc  A4      : 0x3ffcb3cc  A5      : 0x3ffcbc9c  
A6      : 0x73706c6d  A7      : 0x7d706c6d  A8      : 0x800fa280  A9      : 0x3ffb5030  
A10     : 0x3ffcba78  A11     : 0x3ffcbcbc  A12     : 0x3ffb507c  A13     : 0x00000044  
A14     : 0x00000001  A15     : 0x00000006  SAR     : 0x00000010  EXCCAUSE: 0x00000014  
EXCVADDR: 0x00000000  LBEG    : 0x4000c349  LEND    : 0x4000c36b  LCOUNT  : 0x00000000  

Backtrace: 0x00000000:0x3ffb5070 0x400fa3da:0x3ffb50b0 0x40104ae1:0x3ffb50d0 0x40108479:0x3ffb5110 0x4010d536:0x3ffb5130 0x400fcf7f:0x3ffb5150 0x4008ecd1:0x3ffb5180

Rebooting...
ets Jun  8 2016 00:22:57
```

## This is the decoded exception:

```
PC: 0x00000000
EXCVADDR: 0x00000000

Decoding stack results
0x400fa3da: get_arg at ../../../.././newlib/libc/stdio/vfprintf.c line 2242
0x40104ae1: tcp_listen_input at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/lwip/src/core/tcp_in.c line 613
0x40108479: udp_new_port at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/lwip/src/core/udp.c line 110
0x4010d536: ip6_input at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/lwip/src/core/ipv6/ip6.c line 703
0x400fcf7f: mbedtls_internal_sha256_process at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/mbedtls/mbedtls/library/sha256.c line 230
0x4008ecd1: vPortTaskWrapper at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/port.c line 143

```

lbernstone, 2019-03-05:
```
#define RESET_PIN 0
bool restartNow = false;
bool resetPressed = false;

void IRAM_ATTR isrReset() {resetPressed = true;}

void resetAction() {
  for (int x=0; x<20; x++) {
    if (digitalRead(RESET_PIN)) {
      restartNow = true;
      resetPressed = false;
      return; 
    }
    delay(50);
  }
  resetPressed = false;
  Serial.println("Long press");
}

void setup() {
  Serial.begin(115200);
  // switch is normally closed
  attachInterrupt(RESET_PIN, isrReset, FALLING);
}

void loop() {
  if(restartNow) ESP.restart();  
  if(resetPressed) resetAction();
  delay(50);
}
```

GeorgeFlorian, 2019-03-05:
> ```
> #define RESET_PIN 0
> bool restartNow = false;
> bool resetPressed = false;
>
> void IRAM_ATTR isrReset() {resetPressed = true;}
>
> void resetAction() {
>   for (int x=0; x<20; x++) {
>     if [...]

First of all, thank you for your quick response.
Would it be fine for you to explain what's happening in your code ?
Or why isn't mine working ? What is the bug ?
And how can I avoid this problem (whatever the bug is) in the future ?
Thank you !

closed 2019-08-01 by GeorgeFlorian