espressif/arduino-esp32 #2547: Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.

roger-zacharczyk, 2019-03-06:
After two days of running as a controller of PMS7003 my ESP32 stopped working. Now every time it tries to connect to my wifi network it responds:
`
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1100
load:0x40078000,len:10088
load:0x40080400,len:6380
entry 0x400806a4
[D][Configuration.cpp:28] Configuration(): Initializing configuration
[D][Configuration.cpp:45] loadConfiguration(): Loading configuration
[D][WiFiConnector.cpp:13] connect(): Setting up Wi-Fi
[D][WiFiConnector.cpp:24] connectToNetwork(): Connecting to Wi-Fi
Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4017239c: 751cffec bc0d0add 114801f8
Core 0 register dump:
PC      : 0x401723a2  PS      : 0x00060630  A0      : 0x80172776  A1      : 0x3ffaf180  
A2      : 0x00000013  A3      : 0x00004b49  A4      : 0x3ffaf18c  A5      : 0x00000017  
A6      : 0x00000072  A7      : 0x00000009  A8      : 0x00000072  A9      : 0x6000e0c0  
A10     : 0x3ffc7324  A11     : 0x04b411eb  A12     : 0x00000100  A13     : 0x3ffc7324  
A14     : 0x0000000f  A15     : 0x00000000  SAR     : 0x00000019  EXCCAUSE: 0x00000000  
EXCVADDR: 0x00000000  LBEG    : 0x401780d0  LEND    : 0x401780dd  LCOUNT  : 0x00000000  

Backtrace: 0x401723a2:0x3ffaf180 0x40172773:0x3ffaf1b0 0x40172901:0x3ffaf1f0 0x40177a9c:0x3ffaf210 0x40177c17:0x3ffaf260 0x401757d6:0x3ffaf280 0x40175cc8:0x3ffaf2a0 0x400ebab5:0x3ffaf360 0x400ebd76:0x3ffaf390 0x4014b665:0x3ffaf3c0 0x4014b768:0x3ffaf3f0 0x4014ba6a:0x3ffaf420 0x4014bbdf:0x3ffaf450 0x40089378:0x3ffaf470 0x4008fc9d:0x3ffaf4b0

Rebooting...`

What can I do about it? Is the hardware somehow dead? It crashes in WiFi.begin()/WiFi.enableSTA() method.

The same code works on other board.

stickbreaker, 2019-03-06:
Verify your Wifi credentials. It looks like NVS storage is corrupt.

roger-zacharczyk, 2019-03-06:
What's NVC?
Even with hardcoded and correct credentials passed to WiFi.begin it happens.

stickbreaker, 2019-03-06:
Non Volatile Storage(NVS).
Do a Google search on "Erase Flash ESP32"

roger-zacharczyk, 2019-03-06:
Tried to erase flash with esptool. Didn't help.

stickbreaker, 2019-03-06:
@roger-zacharczyk well, it could be a hardware fault:

>Guru Meditation Error: Core 0 panic'ed (IllegalInstruction). Exception was unhandled.

This is saying that a program instruction was not valid.  Usually that is a hardware issue,  makes sure you don't have anything connected to to FLASH buss pins 6..11.   It could be your FLASH has a bad byte that is not correctly storing the program.

lbernstone, 2019-03-06:
Run the [backtrace decoder](https://github.com/me-no-dev/EspExceptionDecoder). If it points to a problem with esp_phy, then you are not providing enough power to initialize the WiFi.  Try to have peripherals turned off until WiFi is connected and the power requirements go down.

closed 2019-08-15 by stale[bot]

leikoilja, 2020-03-21:
Thank you @lbernstone! 
Did a decoding on the backtrace to find out it's `phy` related. You helped me to realize that my ESP connected to a USB hub(into a 2.1A socket thou) is struggling to turn on when my phone is charging from the same hub. Just unplugging the phone from charging has fixed it. Good to know, thanks a lot!

Falkon99, 2020-05-11:
I have this error in ESP32-Cam PCB.
_"Guru Meditation Error: Core 0 panic'ed (IllegalInstruction). Exception was unhandled."_
The problem was weak power supply. 
Better power has solved it.

Ugurugur45, 2021-03-19:
@Falkon99 I get the same error with ESP32 CAM module Ai-Thinker. I supply the module with 5V but battery drops off sometimes. What voltage you supply the module ?

Thanks