espressif/arduino-esp32 #2548: Changings

arnold-b, 2019-03-06:
Hello,

where can i always find the Changes which are made. I´m quit frustrated to migrade from esp8266 to esp32. The most time i need is to find who made which change,  Don´t missunderstand. You all doing a great job. But it´s useless if the people getting tired to search and change there sources.

Jeroen88, 2019-03-06:
Maybe you should invest some time and look at this repository's home page? At releases you'll find this [link](https://github.com/espressif/arduino-esp32/releases/tag/1.0.1)

arnold-b, 2019-03-06:
Thanks for the link.

 

Before it was easy. Normally the old function still exist but the complier will mention its desperate. 

Then I know I have to change it. 

I miss this. The manly advantage of Arduino was that it is compatible to every kind of hardware. That makes hardware changes easy. 

Now the change, need weeks.

 

I already spend 3 days and still have compiler errors. It’s a few thousand lines.

Jeroen88, 2019-03-06:
@arnold-b A lot of libraries are the same between the ESP8266 and the ESP32, or (almost) the same api is offered. However, the ESP32 is another processor and it's own libraries are maintained in this repository.

arnold-b, 2019-03-06:
For me it’s the same core. Just 2 of it. 

But finaly I have only the choice to stay with Arduino or do the more worse step and change from c++ to c. with edpidf

arnold-b, 2019-03-06:
Take for example Fs.h its totally changed. Dir class don´t even exist anymore

Jeroen88, 2019-03-06:
The ESP32 differs completely from the ESP8266, all of the differences are masqueraded for you by the higher level API's. I think you should really view it from a different angle: look at what the community already gave to you without asking anything in return. You are part of that community too, and, if you need certain functionality, create it for yourself, and propose a PR...

You are giving 2 excellent examples. The following snippets already take you where you want to go:

```
#if defined(ESP8266)
#include <FS.h>
#elif defined(ESP32)
#include <SPIFFS.h>
#else
#error "Unsupported board class"
#endif
```

The SPIFFS api and the FS api are very similar (if not the same). I am running complicated software for both the ESP8266 and the ESP32 from one code base, using occasional #if defined's

A directory is a little more complicated. You could use something like this to show the root directory:

```
void showDirectory() {
#if defined(ESP8266)
  Dir dir = SPIFFS.openDir("/");
  while (dir.next()) {
      Serial.print(dir.fileName());
      File f = dir.openFile("r");
      Serial.printf("%*s%d\n", 24 - dir.fileName().length(), "", f.size());
      f.close();
  }
#else
    listDir(SPIFFS, "/", 0);
#endif
}

```

Sometimes things get even more complicated. I needed HTTPUpdate.cpp, which wasn't there. So I ported it from the ESP8266 and created a PR. In this way I still have one clean code base.

arnold-b, 2019-03-06:
Thanks. For Help

I just was very angry about so many changes

closed 2019-03-06 by arnold-b

arnold-b, 2019-03-06:
HTTPUpdate.cpp is also one part i need. and many other