espressif/arduino-esp32 #2552: Adding IRAM_ATTR for getCycleCount

Super169, 2019-03-07:
I'm not sure if this is a common usage, but for some application, it may need to count the time using getCycleCount inside the interrupt (e.g. softwareserial).
It need to have fast response and it may not allow executing from flash. So, IRAM_ATTR is needed to make sure the code is located in RAM instead of flash,  and now I have to modify my local copy in order to make it work.

I'd like to know if it's possible to modify getCycleCount by adding IRAM_ATTR in the official library.

lbernstone, 2019-03-07:
micros is defined as IRAM_ATTR, so microsecondsToClockCycles should be able to do this for you.

Super169, 2019-03-07:
Thanks a lot.  However, ESP32 can be running up to 160MHz, the result of getCycleCount can provide more accurate measurement.  As the connection can be running in 115200bps or above, getCycleCount would be better, and it works fine in 115200 so far.

I will try to modify the program and test with micros.
I'd like to confirm if those esp_timer functions are loaded in RAM, as micros will just call esp_timer_get_time.  Thanks!

closed 2019-08-15 by stale[bot]