espressif/arduino-esp32 #2553: Can i add my own partition scheme

arnold-b, 2019-03-07:
Hello, can i add my own Additional partition scheme ?. 
I tried to add in the board conf and made my own .csv but i still can not see it 

Thanks

beegee-tokyo, 2019-03-07:
Maybe this helps.
Step by step:
[Change partition size (Arduino IDE)](https://desire.giesecke.tk/index.php/2018/04/20/change-partition-size-arduino-ide/)

iwanraflis, 2019-03-08:
Hi I am currently new. But I managed to add a new partition earlier by adding some stuffs in the boards.txt

For example:

[image]

Result:
[image]

Make sure you edit the correct config. I am using EPS32 Dev  Module. So i add some new config to **esp32.menu.PartitionScheme**
Lets say if you are using esp32 wrover, they you have to add new config too **esp32wrover.menu.PartitionScheme**
Hope this helps

closed 2019-08-15 by stale[bot]