espressif/arduino-esp32 #2554: esp32 stop working

atlask1, 2019-03-07:
### Hardware:
Board:							ESP32 DevKit v1 & ttgo_lora
Core Installation/update date:		        latest
IDE name:					       Arduino IDE
Flash Frequency:					40Mhz
PSRAM enabled:                                     no
Upload Speed:				        115200
Computer OS:                                         Ubuntu

### Description:
Hi, same problem with ESP32 DEVKIT v1 and TTGO LORA, tested with some sketch: with devkit I tested sketch with http.post() every 10 sec and with or without Serial.print, after undefined time, stop trasmission or stop printing, all without error or guru meditation.
With ttgo the sketch is simply LoRa receiver that print the message from trasmitter every 10 sec: after 30-45 min stop printing.
Is maybe a uffer problem? 
If I reset all start again until the next stop.
thanks

beats0126, 2019-03-07:
do u have add in any delay()?
last time i try with lora.receiver hangging after few hours. within the lora receive data loop, i put in delay(1) then solve my problem

atlask1, 2019-03-07:
No, there isn't any delay, this is the sketch on TTGO , the sender is an atmega328 with LoRa sender example from library.
The strange is that at 3pm I reboot the TTGO and now is still active....
But I have see this strange behaviour also with esp32 devkit v1 with a sketch without delay , with millis() and a task that simply send http.post every 10 second. Sometimes ( too much time for me :) ) hang and don't print nothing , I see the connection active(wifi) but don't send. Sometimes loose connection , but this is another issue that I'm trying to understand...

'
/* Example from Sandeep Mistry 
 * With mods from AliExpress/LilyGo docs
 * For TTGo ESP32 LoRa-OLED board
 * http://www.lilygo.cn/down_view.aspx?TypeId=11&Id=78&Fid=t14:11:14
 * Based on SX1276 LoRa Radio
 * http://www.semtech.com/apps/product.php?pn=SX1276
 * RMB 29Nov2017
 */

#include <SPI.h>
#include <LoRa.h>       // https://github.com/sandeepmistry/arduino-LoRa
#include <U8g2lib.h>   // https://github.com/olikraus/U8g2_Arduino
// #include <U8x8lib.h>
#ifdef U8X8_HAVE_HW_SPI
#include <SPI.h>
#endif
#ifdef U8X8_HAVE_HW_I2C
#include <Wire.h>
#endif
U8G2_SSD1306_128X64_NONAME_F_SW_I2C Display(U8G2_R0, /* clock=*/ 15, /* data=*/ 4, /* reset=*/ U8X8_PIN_NONE); 

#define OFF 0   // For LED
#define ON 1

// SPI LoRa Radio
#define LORA_SCK 5        // GPIO5 - SX1276 SCK
#define LORA_MISO 19     // GPIO19 - SX1276 MISO
#define LORA_MOSI 27    // GPIO27 - SX1276 MOSI
#define LORA_CS 18     // GPIO18 - SX1276 CS
#define LORA_RST 14   // GPIO14 - SX1276 RST
#define LORA_IRQ 26  // GPIO26 - SX1276 IRQ (interrupt request)

// I2C OLED Display works with SSD1306 driver
#define OLED_SDA 4
#define OLED_SCL 15
#define OLED_RST 16

/* Pick One. Hardware I2C does NOT work! This article helped: https://robotzero.one/heltec-wifi-kit-32/ 
* TTGo boards similar to Heltec boards, LED_BUILTIN = 2 instead of pin 25
* Some OLED displays don't handle ACK correctly so SW I2C works better. Thank you Olikraus!
* TTGo OLED has pin 16 reset unlike other OLED displays
*/

// UNCOMMENT one of the constructor lines below
//U8X8_SSD1306_128X64_NONAME_SW_I2C Display(/* clock=*/ OLED_SCL, /* data=*/ OLED_SDA, /* reset=*/ OLED_RST); // Unbuffered, basic graphics, software I2C
//U8G2_SSD1306_128X64_NONAME_1_SW_I2C Display(U8G2_R0, /* clock=*/ OLED_SCL, /* data=*/ OLED_SDA, /* reset=*/ OLED_RST); // Page buffer, SW I2C
//U8G2_SSD1306_128X64_NONAME_F_SW_I2C Display(U8G2_R0, /* clock=*/ OLED_SCL, /* data=*/ OLED_SDA, /* reset=*/ OLED_RST); // Full framebuffer, SW I2C

const int blueLED = LED_BUILTIN; 

String rssi = "";
String packet = "";

void setup() {
  Serial.begin(115200);
  while (!Serial);

  Serial.println("LoRa Receiver");

  Display.begin();
  Display.enableUTF8Print();    // enable UTF8 support for the Arduino print() function
  Display.setFont(u8g2_font_ncenB10_tr);

  // Very important for SPI pin configuration!
  SPI.begin(LORA_SCK, LORA_MISO, LORA_MOSI, LORA_CS); 
  
  // Very important for LoRa Radio pin configuration! 
  LoRa.setPins(LORA_CS, LORA_RST, LORA_IRQ);         

  pinMode(blueLED, OUTPUT); // For LED feedback
  digitalWrite(blueLED, ON);
  delay(1000);
  digitalWrite(blueLED,OFF);
  if (!LoRa.begin(866E6)) {
    Serial.println("Starting LoRa failed!");
    while (1);
  }
  
  // The larger the spreading factor the greater the range but slower data rate
  // Send and receive radios need to be set the same
//  LoRa.setSpreadingFactor(12);  // ranges from 6-12, default 7 see API docs
}

void loop() {
  // try to parse packet
  int packetSize = LoRa.parsePacket();
  if (packetSize) {
    // received a packet
    
    Serial.print("Received packet '");

    digitalWrite(blueLED, ON);  // Turn blue LED on

    // read packet
    packet = "";                   // Clear packet
    while (LoRa.available()) {
      packet += (char)LoRa.read(); // Assemble new packet
    }
    rssi = LoRa.packetRssi();

    // Display Info
    Display.clearBuffer();  
    Display.setCursor(0,12); Display.print("LoRa Receiver");
    Display.setCursor(0,26); Display.print("Received :");
    Display.setCursor(0,42); Display.print("    '" + packet + "'\n");
    Display.drawFrame(40,44,26,17);
    Display.setCursor(0,58); Display.print("RSSI " + rssi);
    Display.sendBuffer();

    digitalWrite(blueLED, OFF); // Turn blue LED off
    
    Serial.println(packet + "' with RSSI " + rssi); 

  }
}
`

lbernstone, 2019-03-07:
For troubleshooting, it might be useful to add something like this to ensure that your loop is still running:
```
if(!millis() % 600000) {
  digitalWrite(blueLED, ON);
  delay(500);
  digitalWrite(blueLED, OFF);
```

stickbreaker, 2019-03-07:
@lbernstone what do you thing about this code sequence:
```c++
static uint32_t heartbeat=0;
static bool heartbeatOn=false;
if(millis()-heartbeat>5000){ // turn led on every 5 seconds
  digitalWrite(blueLED, ON);
  hearbeat = millis();
  heartbeatOn = true;
}

if(heartbeatOn && (millis()-heartbeat>500)){ // turn off led after 500ms
  digitalWrite(blueLED,OFF);
  heartbeatOn = false;
}
```
I know it uses more storage than your example, but it doesn't introduce the `delay()` which could mask the problem, and your `if()` has to exactly occur on the millisecond boundary. So, unless the loop executes multiple times per millisecond, the chances of it occurring consistently become vanishingly small.  If the average loop executes in 2ms you only have a 50% chance of it lighting the led every 12 minutes. 

Chuck.

lbernstone, 2019-03-07:
I wrote the code on the fly, Chuck.  I wouldn't use it in production; I'd use a timer.  It just is an indicator that things are still running :smile:.

stickbreaker, 2019-03-08:
Yep, production is different than diag. I was just wanting to interject my 2cents about unintended consequences. I have been caught when adding `Serial()` debug lines "fixed" timing issues because of the 115200 baud pacing. 

I'll have to check into `Ticker()` object. Never used it. 

Do use lambda functions very often?

Chuck

lbernstone, 2019-03-08:
I don't do anything complicated with lambdas, but I think they are much clearer than a one liner function in a callback like this.  It shows you what the timer is supposed to do right in the spot where you instantiate it.

atlask1, 2019-03-09:
@beats0126 : 
Hi, why you think I solve with a delay?   
@Ibernstone:
In my sketch led blink every time I receive something , so I know if loop is still running, but when it hang, no blink and no print on oled, and no error . 
Seems not crash, is random block, last time is working for 2 days, other time for 45 minutes.
This happen with and without serial connection .
This happen also with esp32 devkit v1 with a simply http.post every 10 seconds...
I don’t understand if is a sketch problem or core limitation.
I need 24/24 and 7/7 working device, but I don’t understand if is possibile or not.
Same problem with wifi connection, sometimes loose connection after 30 min and sometimes after 2h, and re connection not always work, so I need to reboot via sw(I add a counter for a max number  of error connection that cause reboot).
Thanks!

stickbreaker, 2019-03-09:
@atlask1 your problem description is too vague. We were just trying to give you additional ideas for diagnostic tools.

If I told you my program stopped after 10minutes, how would you tell me to fix it?
What changes should I make?
Should I change the battery?
Is Windows 7 right?
Do I need a new USB cable?

We know you are having problems, but, our un-informed guesses are worthless.  You need to isolate the problem into the smallest code example so we have a chance to identify the problem.

My programs problem ended up being an integer promotion of a uint16 to a int32 that was treated as a negative value in a modulus calculation.

Was that your Guess?

Chuck.

atlask1, 2019-03-09:
@stickbreaker 
You are right! sorry for the vague information... 
It isn't a power problem, I'm 99% sure of this, I'm on Ubuntu but I don't believe is a computer problem.
Maybe could be :

>  an integer promotion of a uint16 to a int32 that was treated as a negative value in a modulus calculation

I write about because happen also with other sketch without a clear reason, and if happen without GURU meditation or error in script, it's hard to debug or to solate problem.
thank you , I try to use your valuable advice , I will change the led usage and the declaration of counter, just to try some solution.
thanks again :)

ogass81, 2019-03-20:
I have experienced similar symptoms, and similar to @atlask1  it is very hard to nail it down to a particular part of your sketch.

The skeetch / loop keeps running, not core crash. However, the HttpServer / WifiConnection is not responding to incoming requests anymore. This can happen after 30min or a week. A reboot resolves the issue.

closed 2019-04-29 by atlask1

OmegaWolfProjects, 2019-09-15:
I have the same problem. That is to say the 'ESP32 TTGO LoRa' is freezing up randomly over night. I'm now using a **watchdog timer** which should reset the board if it freezes up, hopefully this will solve the problem.