espressif/arduino-esp32 #2555: client.write config retry and timeout

ChristianSasso, 2019-03-07:
On the file "WiFiClient.cpp", it would be nice if was possible to configure the defines WIFI_CLIENT_MAX_WRITE_RETRY and WIFI_CLIENT_SELECT_TIMEOUT_US without changing the library file.

Maybe by changing this part of the code:

"WiFiClient.cpp":
```
...

//DELETE:
//size_t WiFiClient::write(const uint8_t *buf, size_t size)

//ADD:
size_t WiFiClient::write(const uint8_t *buf, size_t size, int retry, int timeout_us)

{
    int res =0;

    //DELETE:
    //int retry = WIFI_CLIENT_MAX_WRITE_RETRY;

    int socketFileDescriptor = fd();
    size_t totalBytesSent = 0;
    size_t bytesRemaining = size;

    if(!_connected || (socketFileDescriptor < 0)) {
        return 0;
    }

    while(retry) {
        //use select to make sure the socket is ready for writing
        fd_set set;
        struct timeval tv;
        FD_ZERO(&set);        // empties the set
        FD_SET(socketFileDescriptor, &set); // adds FD to the set
        tv.tv_sec = 0;

        //DELETE:
        //tv.tv_usec = WIFI_CLIENT_SELECT_TIMEOUT_US;

        //ADD
        tv.tv_usec = timeout_us;

        retry--;

        if(select(socketFileDescriptor + 1, NULL, &set, NULL, &tv) < 0) {
            return 0;
        }

        if(FD_ISSET(socketFileDescriptor, &set)) {
            res = send(socketFileDescriptor, (void*) buf, bytesRemaining, MSG_DONTWAIT);
            if(res > 0) {
                totalBytesSent += res;
                if (totalBytesSent >= size) {
                    //completed successfully
                    retry = 0;
                } else {
                    buf += res;
                    bytesRemaining -= res;
                }
            }
            else if(res < 0) {
                log_e("%d", errno);
                if(errno != EAGAIN) {
                    //if resource was busy, can try again, otherwise give up
                    stop();
                    res = 0;
                    retry = 0;
                }
            }
            else {
                // Try again
            }
        }
    }
    return totalBytesSent;
}

//ADD:
size_t WiFiClient::write(const uint8_t *buf, size_t size)
{
    return write(buf, size, WIFI_CLIENT_MAX_WRITE_RETRY, WIFI_CLIENT_SELECT_TIMEOUT_US);
}

...

```

"WiFiClient.h":
```
...

   size_t write(uint8_t data);

   //ADD:
   size_t write(const uint8_t *buf, size_t size, int retry, int timeout_us);

   size_t write(const uint8_t *buf, size_t size);

...   
```

xbary, 2019-03-11:
This is how the client is going to disappear (eg by putting the laptop into sleep mode), eg telnet putty client.
In my repo I set 1 retry rigidly.

ChristianSasso, 2019-03-15:
It can be rigidly, maybe some set_retry function, done in your setup.
Should have some way of changing it without editing the library file
Otherwise, every time you update for the latest version, you must go and change that library file again

alonsoac, 2019-07-24:
It would be very usefull to be able to control this on a per client basis. Some connections need to be quick and not very reliable and will be better with different settings as another that needs many retries.
The default settings add up to a total of 10 seconds where the code is stuck in retries and this affects all other functions.

closed 2019-10-06 by stale[bot]