espressif/arduino-esp32 #2556: unable to read efuse

Adrianotiger, 2019-03-07:
### Hardware:
Board:							ESP32 Dev Module
Core Installation/update date:			07/mar/2019
IDE name:							Arduino IDE
Flash Frequency:					40Mhz
PSRAM enabled:                                          no
Upload Speed:						115200
Computer OS:                                            Windows 10

### Description:
I am unable to read the efuse block from my sketch.
In the ESP-IDF documentation there is a `esp_efuse_read_block` function, but this is not available from the Arduino IDE. I found fuse.h with some functions like `ets_efuse_read_op` but I am unable to understand it and can't find any application using it.

If I understand it correctly, block1, 2 and 3 can used for some unique ID. But once I wrote the blocks with the esptools, it should be possible to read it from the uploaded sketch right?

So the question is: how can I read the block using Arduino IDE?

Thanks!

lbernstone, 2019-03-07:
There is no such function in esp-idf either.  You need to directly read registers, using the locations in <soc/efuse_reg.h>. For example,
`REG_GET_FIELD(EFUSE_BLK0_RDATA4_REG, EFUSE_RD_ADC_VREF)`

Adrianotiger, 2019-03-08:
Thank you. I am now able to read it.

Simple Arduino example to read MAC (just to verify it with the internal function) and read block3:

```
#include <soc/efuse_reg.h>
void setup()
{
  uint32_t mac = REG_GET_FIELD(EFUSE_BLK0_RDATA2_REG, EFUSE_RD_WIFI_MAC_CRC_HIGH);
  Serial.print("MAC: 0x"); 
  Serial.print(mac, HEX);
  mac = REG_GET_FIELD(EFUSE_BLK0_RDATA1_REG, EFUSE_RD_WIFI_MAC_CRC_LOW);
  Serial.println(mac, HEX);

  Serial.print("Block 3: ");
  char h[10];
  for(int32_t block3Address=EFUSE_BLK3_RDATA0_REG;block3Address<=EFUSE_BLK3_RDATA7_REG;block3Address+=4)
  {
    uint32_t block = REG_GET_FIELD(block3Address, EFUSE_BLK3_DOUT0);
    sprintf(h, "%08X ", block);
    Serial.print(h);
  }
}
void loop() {}
```

What happens if I read-protect this block? If I read the documentation, it should not be possible to read this block anymore. Is it only for the espefuse python script or also the sketch? Can I continue to read it trough the sketch?

lbernstone, 2019-03-08:
I do not mess with the efuses much, so I can only point you at the documentation.  AFAIK, the python tool has the same sort of access to them as running code.  The bootloader obviously needs to be able to read them.

closed 2019-08-01 by Adrianotiger

CharlesMcBaker, 2020-07-09:
```
#include <soc/efuse_reg.h>
uint32_t mac;

void setup()
{
Serial.begin(9600);
}

void loop() 
{
    char macadress[13]; //12 MAC ADRESS
    
    mac = REG_GET_FIELD(EFUSE_BLK0_RDATA2_REG, EFUSE_RD_WIFI_MAC_CRC_HIGH);
    String first_part = String(mac, HEX);
    mac = REG_GET_FIELD(EFUSE_BLK0_RDATA1_REG, EFUSE_RD_WIFI_MAC_CRC_LOW);
    String second_part = String(mac, HEX);
    
    String stringcombine = first_part + second_part ;

      for (int i = 0; i <= 11; i++)
      {
      macadress[i] = stringcombine[i+2];
      Serial.println(macadress);
      }      
}
```