espressif/arduino-esp32 #2557: Problem sending ADC values over bluetooth

Hetric, 2019-03-07:
### Hardware:
Board:							ESP32 Dev Module
Core Installation version:			na
IDE name:						Arduino IDE
Flash Frequency:					na
PSRAM enabled:                                      na
Upload Speed:					115200
Computer OS:                                        Windows 10

### Description:
I programmed the ESP32 with Arduino IDE.
I'm trying to send ADC values to my PC using bluetooth method, but it's always sending 52 48 57 53 13 10, which is 4095 in decimal).  
Please help!
Here is my code:

#include "BluetoothSerial.h" //Header File for Serial Bluetooth, will be added by default into Arduino
#define ANALOG_PIN_0 4
BluetoothSerial SerialBT; //Object for Bluetooth

int incoming;
int analog_value = 0;

void setup() 
 {
//    Serial.begin(115200);
    SerialBT.begin("ESP32");
 }
 
void loop() {
   analog_value = analogRead(ANALOG_PIN_0);
   delay(1000);   
//   Serial.println(analog_value);
    
  if (SerialBT.available()) //Check if we receive anything from Bluetooth
  {
    incoming = SerialBT.read(); //Read what we recevive 
    
    if (incoming == 49)
        {        SerialBT.println(analog_value);    // Why is it always sending these numbers 52 48 57 53 13 10??
        }
        
    if (incoming == 48)
        {        SerialBT.println("IJKLMNOP");
        }     
  }
  delay(20);
}

stickbreaker, 2019-03-08:
```c++
#include  "BluetoothSerial.h" //Header File for Serial Bluetooth, will be added by default into Arduino
#define ANALOG_PIN_0 4
BluetoothSerial SerialBT; //Object for Bluetooth

void setup(){
  Serial.begin(115200);
  SerialBT.begin("ESP32");
  pinMode(ANALOG_PIN_0,INPUT);
}

void loop() {
  int analog_value = 0;
  analog_value = analogRead(ANALOG_PIN_0);
  delay(1000);
  Serial.printf("Current Reading on Pin(%d)=%d\n",ANALOG_PIN_0,analog_value);

  if (SerialBT.available()){ //Check if we receive anything from Bluetooth
#define MAXBUFF 50
    static char inputBuffer[MAXBUFF+1]; // fifty character input buffer to store digits of input from blueTooth, persistent
    static int bufPos = 0; // current position in buffer 
     
    inputBuffer[bufPos++] = SerialBT.read(); //Read next character from BlueTooth
    if(bufPos >= MAXBUFF){ // buffer filled before newline '\n' encountered, problem.
      inputBuffer[MAXBUFF] = '\0'; // mark end of "cstring" buffer
      Serial.printf("buffer overflow from bluetooth=%s|\n",inputBuffer);
      bufPos = 0; // start over
    }
    else { // terminate cString
      inputBuffer[bufPos] = '\0';
    }

    if((bufPos>0)&& (inputBuffer[bufPos] == '\n')){ // found end of println() sent over bluetooth crlf(13,10)
      // now we have all the reading and can convert from human readable(println()) format to an integer value.
      char * nextValue;  // storage space for strtol() to record end of conversion position
      int convertedValue = strtol(inputBuffer,&nextValue,0); // convert from human readable to integer, standard bases (0xnn = hex value, 0nn = octal value +-nn = decimal)
      if(convertedValue==0) {// possible conversion error
        if( inputBuffer == nextValue){// conversion error, input buffer does not contain a number
           Serial.printf("Error Decoding Bluetooth, data=%s|\n",inputBuffer);
        }
        else {
            Serial.printf(" Decoded value is zero? data from bluetooth =%s|\n",inputBuffer);
        }
      }
      else { // successful decode of bluetooth data
         Serial.printf(" Decoded bluetooth data=%s| = %d\n",inputBuffer, convertedValue); 
      }
      bufPos = 0; // ready for next reading;
    }
    else { // still waiting for end of line, wait for next bluetooth character
    }
  }
}
```

Try this code.   Your code was displaying the individual character (ASCII) values generated by the `SerialBT.println()` 

Chuck.

Hetric, 2019-03-08:
Chuck,
The code didn't fix it. I found out that ACD2 from ESP2 are reserved for wifi. I didn't turn on the wifi hoping that I would be able to use ADC2, and that didn't sovlve it. 
Is there a way to use ADC2 for Bluetooth to read analog values?

[image]

stickbreaker, 2019-03-08:
Sorry, didn't check to see that you were using ADC2.  I assumed you were miss understanding the human readable transmission format of `println()`.
The ADC2 hardware is used by the radio subsystem.  So, if you use WiFI or Bluetooth the ADC2 converter is unavailable.  only pins connected to ADC1 can be used for analog inputs if the radio subsystem is used.

here is a good [ESP32 DevKit pinout graphic](https://i1.wp.com/randomnerdtutorials.com/wp-content/uploads/2018/08/ESP32-DOIT-DEVKIT-V1-Board-Pinout-36-GPIOs-updated.jpg?w=750&ssl=1) that shows ADC1 and ADC2 pins.

It looks like you are limited to ADC on pins 32..36,39.

Chuck.

copercini, 2019-03-23:
Related with https://github.com/espressif/arduino-esp32/issues/102

closed 2019-08-15 by stale[bot]