espressif/arduino-esp32 #2558: Port not showing on Ardunio IDE, but sketch is running

phrfpeixoto, 2019-03-07:
### Hardware:
Board:							ESP32 Dev Module
Core Installation/update date:		04/mac/2019
IDE name:						Arduino IDE 1.8.8
Flash Frequency:					80Mhz
PSRAM enabled:                                       no
Upload Speed:					115200
Computer OS:                                           Mac OSX High Sierra

### Description:
Can you help me to fix/debug/workaround this? My ESP32 Devkit 1 stopped showing the port on my Arduino IDE setup (mac) after I uploaded the following sketch

Here's pictures of my board: https://imgur.com/a/ms9QAAs

If i connect the wires to GPIO19 and 21 I see that the sketch is indeed running, since the led and relay does change states.

I'm now trying to flash another sketch to the board, but the serial port simply won't show up.
I did try holding EN, BOOT and even both before connecting the USB cable, but I get the same results.

Is it possible that the pins in use are interfering with the serial usage? If so, is there a way to force this board into serial upload mode? 

I'm using MacOS High Sierra on a macbook pro, w/ Arduino IDE 1.8.8
ESP32 addons v. 1.0.1

I get the following error when trying to upload:
```esptool.py v2.6-beta1
Traceback (most recent call last):
Serial port /dev/cu.SLAB_USBtoUART
  File "esptool.py", line 2865, in <module>
  File "esptool.py", line 2858, in _main
  File "esptool.py", line 2565, in main
  File "esptool.py", line 213, in __init__
  File "serial/__init__.py", line 88, in serial_for_url
  File "serial/serialposix.py", line 268, in open
serial.serialutil.SerialException: [Errno 2] could not open port /dev/cu.SLAB_USBtoUART: [Errno 2] No such file or directory: '/dev/cu.SLAB_USBtoUART'
Failed to execute script esptool
An error occurred while uploading the sketch
```

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp
#define PIN_READ 19
#define LED 2
#define RELAY 21

unsigned long previousMillis = 0;
unsigned long i = 0;

void setup() {
  Serial.begin(115200);
  pinMode(PIN_READ, INPUT_PULLDOWN);
  pinMode(LED, OUTPUT);
  pinMode(RELAY, OUTPUT);
}

void loop() {
  unsigned long currentMillis = millis();
  

  if (currentMillis - previousMillis >= 1000) {
    i++;
    // save the last time you blinked the LED
    previousMillis = currentMillis;

    // if the LED is off turn it on and vice-versa:
    int value = 44;
    value = digitalRead(PIN_READ);
    delay(10);
    Serial.printf("%d - Pin %d current state is %d\n", i, PIN_READ, value);
    digitalWrite(LED, value);
    digitalWrite(RELAY, value);
 }
}
```

lbernstone, 2019-03-08:
https://desire.giesecke.tk/index.php/2018/01/30/esp32-not-showing-in-serial-port-mac-desktop-laptop-problem/

phrfpeixoto, 2019-03-08:
The issue seems to be faulty hardware. Same sketch works on another board. Closing

closed 2019-03-08 by phrfpeixoto

kelvinmead, 2020-04-29:
> https://desire.giesecke.tk/index.php/2018/01/30/esp32-not-showing-in-serial-port-mac-desktop-laptop-problem/

wrong issue

> The issue seems to be faulty hardware. Same sketch works on another board. Closing

but can you upload twice? he says he can upload the sketch once, and then it locks out the esp, so he cant upload a second sketch.