espressif/arduino-esp32 #2559: Task WatchDog gets Triggered immediately after SerialBT.begin

baaridunnasr, 2019-03-08:
### Hardware:
Board:							DOIT ESP32 DEV KIT
Core Installation version:			1.0.1
IDE name:						PIO
Flash Frequency:					40Mhz
PSRAM enabled:                                    No
Upload Speed:					921600
Computer OS:                                        Ubuntu

### Description:
The WDT gets triggered everytime I call the Serial BT. It does not trigger after any noticeable delay so somehow just calling it triggers a WDT. If the SerialBT.begin is commented out, it works fine. Wifi is enabled all the time and works pretty fine. I have tried a sample sketch where they work together... As you can see I have added delays in many places to make sure WDT resets. 

### Sketch: 
```cpp

void IRAM_ATTR BT_interrupt_handler()
{
    BaseType_t xHigherPriorityTaskWoken = pdFALSE;
    delay(1500);
    int pinState = digitalRead(interrupt_pin);
    if (pinState == LOW)
    {
        portENTER_CRITICAL_ISR(&mux);
        BT_interrupt = true;
        portEXIT_CRITICAL_ISR(&mux);

        Serial.println("got into interrpt");

        xTaskNotifyFromISR(core_0_task_BT_handler, 1, eSetValueWithOverwrite, &xHigherPriorityTaskWoken);
        portYIELD_FROM_ISR();
    }
}

//-------------------------------------------------------------------------------------------------

void core_0_function_BT_handler(void *parameter)
{
    while (true)
    {
        if (ulTaskNotifyTake(pdTRUE, portMAX_DELAY) > 0)
        {

            portENTER_CRITICAL(&mux);
            BT_interrupt = false;
            portEXIT_CRITICAL(&mux);
            delay(2000);

            Serial.println("got into bt");

            bt.connect("ESP_32");
        }
    }
}
//-------------------------------------------JOBS ON CORE 0 ---------------------------------------

void setup()
{
    Serial.begin(115200);
    Serial.println("STARTED");

    DB9.setup_connection();
    POST.init();

    UPSI_delay = micro_sd.UPSI_logging_delay;

    data_UPSI_queue = xQueueCreate(50, sizeof(struct data_UPSI));

    pinMode(LED_BUILTIN, OUTPUT);
    attachInterrupt(digitalPinToInterrupt(interrupt_pin), BT_interrupt_handler, FALLING);

    Serial.printf("Task UPSI on core 1 : %d \n", (xTaskCreatePinnedToCore(&core_1_function_UPSI, "UPSI", 5000, NULL, 1, &core_1_task_UPSI, 1)));
    Serial.printf("Task COMMS on core 0 : %d \n", (xTaskCreatePinnedToCore(&core_0_function_comms_and_logs, "Comms and Logs", 50000, NULL, 1, &core_0_task_comms_and_logs, 0)));
    Serial.printf("Task BT on core 0 : %d \n", (xTaskCreatePinnedToCore(&core_0_function_BT_handler, "BT Interrupt", 50000, NULL, 3, &core_0_task_BT_handler, 0)));
}
void loop()
{
    delay(10);
}
```
#### The BT connect function called inside core_0_function_BT_handler

```cpp
void ESP32_BT::connect(String ap_name)
{
    _ap_name = ap_name;
    long timeout = millis();

    while (((millis() - timeout) <= 10000) && (!connected_status))
    {
        Serial.println("into bt init loop");
        delay(1);
        if (!SerialBT.begin(_ap_name))
        {
            Serial.println("bt failed");
            delay(500);
        }
        else
        {
            Serial.println("bt success");
            monitor_mode = true;
        }
        connected_status = SerialBT.hasClient();
    }

    if (connected_status)
    {
        while (SerialBT.available() == 0)
        {
            delay(100);
        }

        user_entered_password = SerialBT.readString();

        while (strcmp(user_entered_password.c_str(), _password.c_str()) != 0)
        {
            delay(100);
            SerialBT.println("wrong_password");
            while (SerialBT.available() == 0)
            {
                delay(500);
            }
            user_entered_password = SerialBT.readString();
        }
    }
}
```

### Debug Messages:
```
[W][sd_diskio.cpp:143] sdCommand(): crc error
[W][sd_diskio.cpp:143] sdCommand(): crc error
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 4 - STA_CONNECTED
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 7 - STA_GOT_IP
[D][WiFiGeneric.cpp:385] _eventCallback(): STA IP: 10.90.1.39, MASK: 255.255.255.0, GW: 10.90.1.1
Task UPSI on core 1 : 1 
Task COMMS on core 0 : 1 
Task BT on core 0 : 1 
got into interrpt
got into bt
into bt init loop
E (27749) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (27749) task_wdt:  - IDLE0 (CPU 0)
E (27749) task_wdt: Tasks currently running:
E (27749) task_wdt: CPU 0: BT Interrupt
E (27749) task_wdt: CPU 1: IDLE1
E (27749) task_wdt: Aborting.
abort() was called at PC 0x4015e444 on core 0

Backtrace: 0x40091e8c:0x3ffbe540 0x400920bd:0x3ffbe560 0x4015e444:0x3ffbe580 0x40081a9d:0x3ffbe5a0 0x400dfca4:0x00000000
```
### Decoded Stack Results
```
Decoding stack results
0x40091e8c: invoke_abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/panic.c line 155
0x400920bd: abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/panic.c line 170
0x4015e444: task_wdt_isr at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/task_wdt.c line 174
```

### Additional things I've tried
1) Disabling HSPI before trying with no luck
2) A simple sample sketch with WIFI + BT which works without any issues.
3) Adding the following code in the void loop() [not sure if it is correct]
```

TIMERG0.wdt_wprotect=TIMG_WDT_WKEY_VALUE;
TIMERG0.wdt_feed=1;
TIMERG0.wdt_wprotect=0;
```
from this
```
#include "soc/timer_group_struct.h"
#include "soc/timer_group_reg.h"
```
This TIMERG0 makes some weird stuff happen. There seems to be no errors but then I am not able to see the BT device on my phone. And there are no events on verbose debug to indicate BT started as there usually is from the test sketch.

baaridunnasr, 2019-03-08:
```
 STARTED
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 4 - STA_CONNECTED
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 7 - STA_GOT_IP
[D][WiFiGeneric.cpp:385] _eventCallback(): STA IP: 10.90.1.39, MASK: 255.255.255.0, GW: 10.90.1.1
Task UPSI on core 1 : 1 
Task COMMS on core 0 : 1 
Task BT on core 0 : 1 
got into interrpt
got into bt
into bt init loop
```
The above^ is what I get when I add the below into the void loop. It just sits there. Nothing happens
```
TIMERG0.wdt_wprotect=TIMG_WDT_WKEY_VALUE;
TIMERG0.wdt_feed=1;
TIMERG0.wdt_wprotect=0;
```

stickbreaker, 2019-03-08:
@baaridunnasr 
* Don't `delay()` inside interrupt Routine
* Don't use `Serial()` inside interrupt Routine, It can and will stall if output buffer fills.
* prefix all data shared between interrupt context and task with `volatile`
* only call `portYIELD_FROM_ISR()` when indicated.

```c++
volatile bool BT_Interrupt = false;
volatile portMUX_TYPE mux = portMUX_INITIALIZER_UNLOCKED;

void IRAM_ATTR BT_interrupt_handler()
{
    BaseType_t xHigherPriorityTaskWoken = pdFALSE;
    int pinState = digitalRead(interrupt_pin);
    if (pinState == LOW)
    {
        portENTER_CRITICAL_ISR(&mux);
        BT_interrupt = true;
        portEXIT_CRITICAL_ISR(&mux);
        xTaskNotifyFromISR(core_0_task_BT_handler, 1, eSetValueWithOverwrite, &xHigherPriorityTaskWoken);

        if(xHigherPriorityTaskWoken) portYIELD_FROM_ISR();
    }
}
```

>E (27749) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (27749) task_wdt:  - IDLE0 (CPU 0)
E (27749) task_wdt: Tasks currently running:
E (27749) task_wdt: CPU 0: BT Interrupt
E (27749) task_wdt: CPU 1: IDLE1
E (27749) task_wdt: Aborting.
abort() was called at PC 0x4015e444 on core 0

This is telling you that the `IDLE0{CPU 0)` task never got any processing time.  Process `BT Interrupt` never released the CPU.  `IDLE0` runs at priority 0 (zero), so it only runs when there is nothing ready to run with a higher priority. The higher priority task need to not be so glutenous.  Figure out how you can insert a `delay()` with a value >= 2 into your high priority tasks.

I think the default wdt is 8 seconds. 

Chuck.

baaridunnasr, 2019-03-08:
Dear @stickbreaker 

### 1)
>     Don't `delay()` inside interrupt Routine
There were no delays inside ISR. Trying to keep it as short as possible

-----------------------------------------------------------------------
### 2)
>     Don't use Serial() inside interrupt Routine, It can and will stall if output buffer fills.
Okay, makes sense. I removed them.

-----------------------------------------------------------------------
### 3)
>    prefix all data shared between interrupt context and task with volatile

`volatile bool BT_Interrupt = false; ` / /<--- This was already done. 

`volatile portMUX_TYPE mux = portMUX_INITIALIZER_UNLOCKED;` 

 The IDE highlights w/ the following error message : `argument of type "volatile portMUX_TYPE *" is incompatible with parameter of type "portMUX_TYPE *"`
 wherever I call `portENTER_CRITICAL_ISR(&mux);` or `portENTER_CRITICAL(&mux);`

-----------------------------------------------------------------------
### 4)
>    only call portYIELD_FROM_ISR() when indicated.

Implemented as per your suggestion.

-----------------------------------------------------------------------
### 5)
> This is telling you that the IDLE0{CPU 0) task never got any processing time. Process BT Interrupt never released the CPU. IDLE0 runs at priority 0 (zero), so it only runs when there is nothing [...]

Will Look into it ASAP. 

-----------------------------------------------------------------------
### 6)
>  p.s. Are you sure you can allocate 105k of RAM as stack to these tasks? BlueTooth is really piggy on RAM. with WiFi and BlueTooth, I don't think you have that much RAM to spare.

Trying with a lower stack gives me a LoadStoreProhibited Error. I assume that the Tasks that call the BT function are the ones that need a high stack.

### Further Additional observations
Also, Upon opening the BlueToothSerial.cpp and examining the `begin` and further the `_init_bt` function I find a 

```cpp
xTaskCreate(_spp_tx_task, "spp_tx", 4096, NULL, 2, &_spp_task_handle);
```

Is this perhaps causing some sort of clash with the task priorities?

I noticed this and then changed my `core_0_function_BT_handler` to priority 3.

```cpp
TIMERG0.wdt_wprotect=TIMG_WDT_WKEY_VALUE;
TIMERG0.wdt_feed=1;
TIMERG0.wdt_wprotect=0;
```
The above causes task to hang while init BT. No debug messages. No crashes. BT not initialized too. Just sits there.

Many thanks
Nasr.

baaridunnasr, 2019-03-08:
## Additional observations

```
void core_0_function_BT_handler(void *parameter)
{
    while (true)
    {
        if (ulTaskNotifyTake(pdTRUE, portMAX_DELAY) > 0)
        {

            portENTER_CRITICAL(&mux);
            BT_interrupt = false;
            portEXIT_CRITICAL(&mux);
            delay(2000);  //<--- Already had this delay in to make sure WDT resets. 

            Serial.println("got into bt");

            // bt.connect("ESP_32"); //<--- Has been commented out so no BT call
        }
    }
}
```

As I said before, somehow the BT call seems to seize core 0 and hold it there. Not allowing IDLE0 to reset at all.

stickbreaker, 2019-03-08:
@baaridunnasr 
#### 1) from your posted example
```c++
void IRAM_ATTR BT_interrupt_handler()
{
    BaseType_t xHigherPriorityTaskWoken = pdFALSE;
    delay(1500);
    int pinState = digitalRead(interrupt_pin);
    if (pinState == LOW)
```

#### 3) volatile portMUX_TYPE mux = portMUX_INITIALIZER_UNLOCKED;
you are correct. I did not look at the data type.

One other consideration on [Critial Sections](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/freertos-smp.html?highlight=critical#critical-sections-disabling-interrupts), understand that a critical section will cause the other CPU to spin, and disable this CPU's interrupts, so don't call any function from within a critical section that takes any significant processing time.

#### other
  You use `SerialBT()`, what is / where is `bt()` defined? are you using them in multiple tasks?

#### additional observations
  that `delay()` should not be necessary , the `xTaskNotifyTake()` will suspend this task, I would set your priorities to be dissimilar 2,3,4, you might want to read about how the ESP32 implementation of the Scheduler can skip tasks with equivalent priorities [round-robin-scheduling](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/freertos-smp.html?highlight=critical#round-robin-scheduling)

Chuck.

baaridunnasr, 2019-03-08:
@stickbreaker 
### 1) 
> from your posted example

Oh damnit. I was trying to debounce the key to check for a value after 1.5 seconds.

---------
### 2)
> You use SerialBT(), what is / where is bt() defined? are you using them in multiple tasks?

Serial BT is defined in another .cpp file which I have included using it's header file. To be clear, all BT functions are defined in another header file + cpp file to keep them clean and debuggable if necessary.

---------
### 3)
> at delay() should not be necessary , the xTaskNotifyTake() will suspend this task

 You are correct. However removing the delay throws an interrupt wdt timeout rather than a task wdt timeout as seen before. Not sure why this is happening.

[image]

--------

### 4) 

>  I would set your priorities to be dissimilar 2,3,4. 

All my tasks on a particular core are of different priorities. However the inspection of the in built functions is what brought about that doubt.

stickbreaker, 2019-03-08:
#### 2)
I don't have any experience with bluetooth, But, does the hardware driver support multiple instances? 

Wifi doesn't. I know Wifi and Bluetooth had to be specially written to take and release the single radio.

Chuck.

baaridunnasr, 2019-03-08:
@stickbreaker 

Okay, coexist isnt perfect. But the access is synchronous alternating so I'm not worried. 

I have another doubt. Let's say I have a task with some stack 'x'. Now if I call Wifi/BT inside it does the allocated stack for those peripherals come from inside the calling task or the total Heap memory?

It may seem like a very bad memory issue in my program that I've found out with @tobozo 's help using the `ESP.getFreeHeap()`. The Wifi and BT running together just gobble stellar chunks of my RAM. 

Trying to optimise this at the moment. However this definitely did not seem like a memory error from the debug messages.

Many thanks,
Nasr.

baaridunnasr, 2019-03-08:
@stickbreaker  
Right now I decided to call BT from the setup function instead of another task and initialises and I saw that it takes a 99k stack from the RAM. Wifi is super hungry too. 

Any way possible to reconfigure RAM limits from modifying the ld files?
I've had some luck with ESP8266 in increasing DRAM size. Flashed and tested too, without any issues.

Nasr.

stickbreaker, 2019-03-08:
Most of the memory allocation come from heap. The default arduino task only has 8192 stack and can run both wifi and bluetooth.

stickbreaker, 2019-03-08:
All tasks share One global heap, each task has private stack. A new task allocates its stack from heap.

baaridunnasr, 2019-03-08:
@stickbreaker so this looks the BT and WiFi tasks have their own stack which is completely independent of the main Arduino stack, right?

stickbreaker, 2019-03-08:
Bluetooth and WiFi create their own tasks which each have their own stack.  The each tasks stack is only used for local, non persistent data structures. When you call the `.begin()` functions you cause the creation of these background tasks. the `.end()` function start the shutdown sequence,  it may return before the background tasks have shutdown and cleaned up after themselves. So, calling `.begin()`, use it, `.end()` does not guarantee that all background tasks and data structures have been freed, some of the garbage collection is not done until the `IDLE0, IDLE1` tasks have executed.  Rapid fire `.begin()`, `.end()` sequences will exhaust system resources.

Chuck.

baaridunnasr, 2019-03-09:
So, I'll look into returning control to IDLE0/IDLE1 once in a while to finish the garbage collection. Do you have any idea why the lack of memory for BT wasn't apparent? LoadStoreProhibited Error should have immediately shown that BT doesnt have enough stack. 

Anywho, thanks for the help Chuck. Much appreciate you coming to help me out everytime I post such absurd issues on here.

Many thanks,
Nasr.

closed 2019-03-09 by baaridunnasr