espressif/arduino-esp32 #2560: net::ERR_CONNECTION_RESET

arnold-b, 2019-03-08:
Hello,

I´m using spiffs for serving the files over html. 
if i try to open one html page i get a few files libraries but not all. on the files which i dont get ican see in the browser: net::ERR_CONNECTION_RESET 
In my serial log on ESP 32 i dont see even the request for this file. Normaly i see every html request.

Before i used this program on esp8266 and there are no problem. 

If i try to load the file over url request one by one. i get all.

I think there must be some limitation which i don´t know now. 

thanks for help

arnold-b, 2019-03-08:
Hello, 

after i had the Problem with my solution and i didnt find any problem.
I just installed ESP3D and had the same Problems. 
If i try to run esp3d with the index.html.gz everything runs, but the first loading is very slow. around 92K. (1 minute),
I tried with diffrent ESP32 and with diffrend browser.

If i expand the gz file, download to spiffs and try to run just index.html there are a lot js and css to download.
It´s randomly wich file esp is serving and which one will be ignored. it looks like esp is to busy with accessing the small files, that he even ignore the html requests totaly.  Because in the serial log you can not see any request of this files. 
Chrome Error: net::ERR_CONNECTION_RESET on this file.

Every file can be loaded if you try it one by one. Then ESP have enough time.

if you try to load 1-5 small files everything looks good. but don´t load one big file or even 20 files.

just try it yourself. downloadhttps://github.com/luc-github/ESP3D/tree/2.0. 
and then https://github.com/luc-github/ESP3D-WEBUI/tree/2.0.

if you install it on his small slow brother esp8266, then everything works fast. may its still to early to use esp32. i should have wait till version 3 before i try to make hardware for this.

arnold-b, 2019-03-08:
Now i did this and ignored many files 
			if (SPIFFS.exists(uri()) && (uri().indexOf("css/") == -1) && (uri().indexOf("js/") == -1))
			{

					Files.streamFile(&HoTWebserver);
}
else send back 404 to client

then i still can not see the requests, which i am missing.

the whole download is no only 16kb. it looks for me more and more that wifi or webserver can not handle 49 requests

lbernstone, 2019-03-09:
https://github.com/espressif/arduino-esp32/blob/master/libraries/SD/src/sd_diskio.cpp#L728
https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/storage/fatfs.html#_CPPv320esp_vfs_fat_registerPKcPKc6size_tPP5FATFS

arnold-b, 2019-03-09:
Hello,

how should this link help ?. I dont even get the html request. 

thanks

arnold-b, 2019-03-10:
Now at least i get errormessage from WiFiClient.cpp
Stream File
/sy.html
text/html
[D][WiFiClient.cpp:481] connected(): Disconnected: RES: 0, ERR: 128
[D][WiFiClient.cpp:481] connected(): Disconnected: RES: 0, ERR: 128
[V][WebServer.cpp:284] handleClient(): New client
[V][Parsing.cpp:113] _parseRequest(): method: GET url: /css/menu.css search: 
[V][Parsing.cpp:219] _parseRequest(): headerName: Host
[V][Parsing.cpp:220] _parseRequest(): headerValue: 192.168.1.109
[V][Parsing.cpp:219] _parseRequest(): headerName: Connection
[V][Parsing.cpp:220] _parseRequest(): headerValue: keep-alive
[V][Parsing.cpp:219] _parseRequest(): headerName: User-Agent
[V][Parsing.cpp:220] _parseRequest(): headerValue: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/72.0.3626.121 Safari/537.36
[V][Parsing.cpp:219] _parseRequest(): headerName: Accept
[V][Parsing.cpp:220] _parseRequest(): headerValue: text/css,*/*;q=0.1
[V][Parsing.cpp:219] _parseRequest(): headerName: Referer
[V][Parsing.cpp:220] _parseRequest(): headerValue: http://192.168.1.109/sy.html?
[V][Parsing.cpp:219] _parseRequest(): headerName: Accept-Encoding
[V][Parsing.cpp:220] _parseRequest(): headerValue: gzip, deflate
[V][Parsing.cpp:219] _parseRequest(): headerName: Accept-Language
[V][Parsing.cpp:220] _parseRequest(): headerValue: de-DE,de;q=0.9,en-US;q=0.8,en;q=0.7
[V][Parsing.cpp:247] _parseArguments(): args: 
[V][Parsing.cpp:230] _parseRequest(): Request: /css/menu.css
[V][Parsing.cpp:231] _parseRequest():  Arguments: 
[E][WebServer.cpp:602] _handleRequest(): request handler not found
-1
Stream File
/css/menu.css
text/css
[D][WiFiClient.cpp:481] connected(): Disconnected: RES: 0, ERR: 128
[D][WiFiClient.cpp:481] connected(): Disconnected: RES: 0, ERR: 128
[D][WiFiClient.cpp:481] connected(): Disconnected: RES: 0, ERR: 128
[D][WiFiClient.cpp:481] connected(): Disconnected: RES: 0, ERR: 128
[D][WiFiClient.cpp:481] connected(): Disconnected: RES: 0, ERR: 128
[V][WebServer.cpp:284] handleClient(): New client
[V][Parsing.cpp:113] _parseRequest(): method: GET url: /css/style.css search: 
[V][Parsing.cpp:219] _parseRequest(): headerName: Host
[V][Parsing.cpp:220] _parseRequest(): headerValue: 192.168.1.109
[V][Parsing.cpp:219] _parseRequest(): headerName: Connection
[V][Parsing.cpp:220] _parseRequest(): headerValue: keep-alive
[V][Parsing.cpp:219] _parseRequest(): headerName: User-Agent
[V][Parsing.cpp:220] _parseRequest(): headerValue: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/72.0.3626.121 Safari/537.36
[V][Parsing.cpp:219] _parseRequest(): headerName: Accept
[V][Parsing.cpp:220] _parseRequest(): headerValue: text/css,*/*;q=0.1
[V][Parsing.cpp:219] _parseRequest(): headerName: Referer
[V][Parsing.cpp:220] _parseRequest(): headerValue: http://192.168.1.109/sy.html?
[V][Parsing.cpp:219] _parseRequest(): headerName: Accept-Encoding
[V][Parsing.cpp:220] _parseRequest(): headerValue: gzip, deflate
[V][Parsing.cpp:219] _parseRequest(): headerName: Accept-Language
[V][Parsing.cpp:220] _parseRequest(): headerValue: de-DE,de;q=0.9,en-US;q=0.8,en;q=0.7
[V][Parsing.cpp:247] _parseArguments(): args: 
[V][Parsing.cpp:230] _parseRequest(): Request: /css/style.css
[V][Parsing.cpp:231] _parseRequest():  Arguments: 
[E][WebServer.cpp:602] _handleRequest(): request handler not found
-1
Stream File
/css/style.css
text/css
[D][WiFiClient.cpp:481] connected(): Disconnected: RES: 0, ERR: 128
[D][WiFiClient.cpp:481] connected(): Disconnected: RES: 0, ERR: 128
[D][WiFiClient.cpp:481] connected(): Disconnected: RES: 0, ERR: 128
[D][WiFiClient.cpp:481] connected(): Disconnected: RES: 0, ERR: 128
[D][WiFiClient.cpp:481] connected(): Disconnected: RES: 0, ERR: 128
[V][WebServer.cpp:284] handleClient(): New client
[V][Parsing.cpp:113] _parseRequest(): method: GET url: /js/app.js search: 
[V][Parsing.cpp:219] _parseRequest(): headerName: Host
[V][Parsing.cpp:220] _parseRequest(): headerValue: 192.168.1.109
[V][Parsing.cpp:219] _parseRequest(): headerName: Connection
[V][Parsing.cpp:220] _parseRequest(): headerValue: keep-alive
[... 193 lines cut ...]
[V][Parsing.cpp:113] _parseRequest(): method: GET url: /js/smoothie.js search: 
[V][Parsing.cpp:219] _parseRequest(): headerName: Host
[V][Parsing.cpp:220] _parseRequest(): headerValue: 192.168.1.109
[V][Parsing.cpp:219] _parseRequest(): headerName: Connection
[V][Parsing.cpp:220] _parseRequest(): headerValue: keep-alive
[V][Parsing.cpp:219] _parseRequest(): headerName: User-Agent
[V][Parsing.cpp:220] _parseRequest(): headerValue: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/72.0.3626.121 Safari/537.36
[V][Parsing.cpp:219] _parseRequest(): headerName: Accept
[V][Parsing.cpp:220] _parseRequest(): headerValue: */*
[V][Parsing.cpp:219] _parseRequest(): headerName: Referer
[V][Parsing.cpp:220] _parseRequest(): headerValue: http://192.168.1.109/sy.html?
[V][Parsing.cpp:219] _parseRequest(): headerName: Accept-Encoding
[V][Parsing.cpp:220] _parseRequest(): headerValue: gzip, deflate
[V][Parsing.cpp:219] _parseRequest(): headerName: Accept-Language
[V][Parsing.cpp:220] _parseRequest(): headerValue: de-DE,de;q=0.9,en-US;q=0.8,en;q=0.7
[V][Parsing.cpp:247] _parseArguments(): args: 
[V][Parsing.cpp:230] _parseRequest(): Request: /js/smoothie.js
[V][Parsing.cpp:231] _parseRequest():  Arguments: 
[E][WebServer.cpp:602] _handleRequest(): request handler not found
-1
Stream File
/js/smoothie.js
application/javascript
[D][WiFiClient.cpp:481] connected(): Disconnected: RES: 0, ERR: 128
[D][WiFiClient.cpp:481] connected(): Disconnected: RES: 0, ERR: 128

schreibfaul1, 2019-04-10:
Hello,
i have the same problem. Disable AMPDU in sdkconfig.h as in #2382 described.
#define CONFIG_ESP32_WIFI_AMPDU_TX_ENABLED 0
#define CONFIG_ESP32_WIFI_AMPDU_RX_ENABLED 0
That works for me.

closed 2019-08-15 by stale[bot]

nikbakhtenb4, 2023-06-28:
I have wo system. both successfully build esp.
 when I buil esp with one of them every thing works fine, but when I try to build with other system I have problem to routing and loading webserver of esp after upload.
first try to update the platforms in platformIO of 'vs Code', for example I updated Espressif 32 , or you can just uninstall and install again. I try to add lots of extention of my 'vs code', I dont know which one exactly help, but I bleive they effect,here is the some name of them: 'C/C++','C/C++ Extension Pack','C/C++ Theme','ESP8266FS', 'Espressif IDF', 'ESP-Engine',...
then restart your vscode.
'clean full' project from platformIO and build again.
I hope this help you