espressif/arduino-esp32 #2568: Serial.available() issues when changingsetRxBufferSize

mars000, 2019-03-11:
Make your question, not a Statement, inclusive. Include all pertinent information:

What you are trying to do?
Describe your system( Hardware, computer, O/S, core version, environment).
Describe what is failing.
Show the shortest possible code that will duplicate the error.
Show the EXACT error message(it doesn't work is not enough).
All of this work on your part shows us that you have worked to solve YOUR problem. The more complete your issue posting is, the more likely someone will volunteer their time to help you.

If you have a Guru Meditation Error or Backtrace, ***please decode it***:
https://github.com/me-no-dev/EspExceptionDecoder

----------------------------- Remove above -----------------------------

### Hardware:
Board:							LOILIN Pro 32
Core Installation version:			1.0.1
IDE name:						Arduino IDE
Flash Frequency:					240Mhz
PSRAM enabled:                                    ?
Upload Speed:					115200
Computer OS:                                        Windows 10

### Description:
Describe your problem here

You need two devices connected to each other via Serial connections to replicate this issue.

I am using Serial2 hardware to TX/RX and I notice that ifI use Serial2.setRxBufferSize() to change the hardware serial buffer ta value like 15000; then when I use Serial2.available() it slowly increases up to the set buffer size i.e. 15000 in this case. When I don't use Serial2.setRxBufferSize() (i.e. leave to default) it all works as per normal.

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

// Receiving ESP32 

setup()
{
 Serial2.begin(500000, SERIAL_8N1, RXD2, TXD2);
 
 Serial2.setRxBufferSize(15000);  // If I do not do this function call ... all work well in my read_serial().
}

void read_serial()
{
int serial_available = Serial2.available();
Serial.println (serial_available); // for each loop slowly goes up and up each time, and flat lines to 15000 even through I read all serial items in this loop.  
while (Serial2.available() < 0) // slowly increases up to 15000 each time i go through this loop if SerRxBufferSize is set to 15000; even though data being sent through Serial write ESP32 is approx 500bytes every 1 second.
 {
     char c = ((char) Serial2.read());  // read correctly...
  }
}

loop()
{
  static read_event = millis() + 2000;
 if (millis() > read_event)
 {
  serial_read();
  read_event = millis() + 2000;
 }
}

```

### Debug Messages:
```
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
```

stickbreaker, 2019-03-11:
`Serial2.available()` will ALWAYS be >=0,  It can never be negative.

```c++
while (Serial2.available() < 0) // slowly increases up to 15000 each time i go through this loop if SerRxBufferSize is set to 15000; even though data being sent through Serial write ESP32 is approx 500bytes every 1 second.
 {
     char c = ((char) Serial2.read());  // read correctly...
  }
```

I think you actually want:
```c++
while (Serial2.available() > 0) // slowly increases up to 15000 each time i go through this loop if SerRxBufferSize is set to 15000; even though data being sent through Serial write ESP32 is approx 500bytes every 1 second.
 {
     char c = ((char) Serial2.read());  // read correctly...
  }

```

Chuck.

mars000, 2019-03-11:
apologies - that was a typo when I copy/pasted it ... it is and should be > 0.

closed 2019-08-15 by stale[bot]