espressif/arduino-esp32 #2569: "[HTTP] GET... failed, error: connection refused" when using Static IP

GeorgeFlorian, 2019-03-11:
### Hardware:

Board: ESP32 DEVKIT1
Core Installation version: 1.0.1-git [this one](https://github.com/espressif/arduino-esp32)
IDE name: Arduino IDE
Flash Frequency: 80Mhz
PSRAM enabled: no
Upload Speed: 921600
Computer OS: Linux Mint 19.1 Mate

Hello there !

I'm having troubles receiving the response from a URL when using Static IP. **All works perfectly fine using DHCP IP.**
I assume the ESP32 connects to the internet because I can access the webpage (which has a HTML input form where one can enter the URL) using the Static IP.

Long story short I am reading lines from a SPIFFS file and storing them in an String array x[].

```
void setup () {
  if(x[0] != NULL &&      // SSID
     x[0].length() != 0 &&
     x[1] != NULL &&      // Password
     x[1].length() != 0 &&
     x[2] != NULL &&      //Local IP
     x[2].length() != 0 &&
     x[3] != NULL &&      // Gateway
     x[3].length() != 0 &&
     x[4] != NULL &&      // Subnet
     x[4].length() != 0) {
       IPAddress local_IP_STA, gateway_STA, subnet_STA;
       local_IP_STA.fromString(x[2]);
       gateway_STA.fromString(x[3]);
       subnet_STA.fromString(x[4]);
       IPAddress primaryDNS(8, 8, 8, 8);
       IPAddress secondaryDNS(8, 8, 4, 4);
       WiFi.begin(x[0].c_str(),x[1].c_str());
       delay(50);
       if(!WiFi.config(local_IP_STA, gateway_STA, subnet_STA))
       Serial.println("Couldn't configure STATIC IP ! Starting DHCP IP !");
       delay(50);
}
void loop() {
if ((WiFi.status() == WL_CONNECTED)) {
HTTPClient http;
http.begin(URL);
      int httpCode = http.GET(); //Make the request
  
      if (httpCode > 0) { //Check for the returning code
          Serial.printf("[HTTP] GET... code: %d\n", httpCode);
          if(httpCode == HTTP_CODE_OK) {
              String payload = http.getString();
              Serial.println(payload);
            }
        } else {
            Serial.printf("[HTTP] GET... failed, error: %s\n", http.errorToString(httpCode).c_str());
          }
  
      http.end();
   } // while
delay(10000);
}
```

`WiFi.config(local_IP_STA, gateway_STA, subnet_STA)` returns 
```
[E][WiFiGeneric.cpp:658] hostByName(): DNS Failed for ...
f 
```

It will randomly output **f** and **d**.

`WiFi.config(local_IP_STA, gateway_STA, subnet_STA, primaryDNS)` or 
`WiFi.config(local_IP_STA, gateway_STA, subnet_STA, primaryDNS, secondaryDNS)` returns 
```
[E][WiFiGeneric.cpp:658] hostByName(): DNS Failed for ...
[HTTP] GET... failed, error: connection refused
```

What do you think the problem is ?

lbernstone, 2019-03-11:
Your assumption that you can reach the internet is not a good one.  Are you using the exact same addressing, including subnet mask and gateway, that you were getting from dhcp?  Please try that first.

GeorgeFlorian, 2019-03-11:
> Your assumption that you can reach the internet is not a good one. Are you using the exact same addressing, including subnet mask and gateway, that you were getting from dhcp? Please try that first.

I've tried with exact same local IP, Gateway and SubnetMask as the DHCP and it works. I don't get the error anymore.

It also works only when I pass the primaryDNS as follows: `WiFi.config(local_IP_STA, gateway_STA, subnet_STA, primaryDNS)`

But what does this mean ?

lbernstone, 2019-03-11:
Uh, that means that [TCP/IP subnetting](https://en.wikipedia.org/wiki/Subnetwork) is working the way it is supposed to.  DNS is needed to resolve names to addresses.  A DNS server is information that your DHCP server can give to you along with a valid address....

GeorgeFlorian, 2019-03-12:
> Uh, that means that [TCP/IP subnetting](https://en.wikipedia.org/wiki/Subnetwork) is working the way it is supposed to. DNS is needed to resolve names to addresses. A DNS server is information that your DHCP server can give to you along with a valid address.... Please close this issue, since it is not a error in ESP code.

So it's my fault for not finding working IP, gateway and subnet, right ?

closed 2019-03-12 by GeorgeFlorian