espressif/arduino-esp32 #2570: Status of Ping for the ESP32

bill-orange, 2019-03-11:
For some time I have been successfully using Ping.h on the ESP32 core on various boards.  Using the 1.0.0 core all is well.  If I switch to the newer 1.0.1 core, compiler errors result ('inet_addr_to_ipaddr' was not declared in this scope).  This issue has been beat into the ground elsewhere. example: https://forum.arduino.cc/index.php?topic=591282.0

Is this problem on the roadmap for repair in future core releases?  I would not want to be stuck on the 1.0.0 core forever and I also don't want to abandon my current use of ping.

If Ping.h functionality will be restored soon then I will stick with the 1.0.0 core for now.  Otherwise, I will have to work toward a more elaborate work-around.

atanisoft, 2019-03-11:
Perhaps open an issue with the maintainer of the Ping library? It looks like they are using an API that was replaced in LwIP itself, most likely the correct replacement for inet_addr_to_ipaddr is ip4_addr_set or ip4_addr_copy.

bill-orange, 2019-03-13:
I am not sure if Ping is still being actively supported by the original developer or if he is still around would wish to work on an ESP32 problem.  I tried the two replacements you suggested.  No dice.  They had different structures.

I any event, it would be good to keep Ping portable between ESP8266 and ESP32 applications.

Thanks for the suggestions.

atanisoft, 2019-03-14:
Add:
```
#include <lwip/inet.h>
```
to the top of Ping.cpp and it should work just fine. This shouldn't have even compiled with v1.0.0 as this function is not defined in any of the headers that Ping.cpp includes.

You can close this issue and instead track this here: https://github.com/pbecchi/ESP32_ping/issues/8

bill-orange, 2019-03-15:
Unfortunately adding the #include statement did not help. I will close this thread and track on the one mentioned above.

closed 2019-03-15 by bill-orange