espressif/arduino-esp32 #2571: PROVIDE ( GPIO = 0x60004000 ); in tools/sdk/ld/esp32.peripherals.ld

marcmerlin, 2019-03-11:
I have the latest git checkout of esp32-arduino
saruman:~/arduino/hardware/espressif/esp32$ git log
commit 7df50a97d17b0953ea01cad0355410a66bd8b8b4 (HEAD -> master, tag: 1.0.2-rc1, origin/master, origin/HEAD)
Author: Me No Dev <me-no-dev@users.noreply.github.com>
Date:   Sun Mar 3 17:19:11 2019 +0100

I had to reapply this patch:
tools/sdk/ld/esp32.peripherals.ld
-PROVIDE ( GPIO = 0x3ff44000 );
+PROVIDE ( GPIO = 0x60004000 );

I got it from there https://github.com/NeoCat/ESP32-P3RGB64x32MatrixPanel/blob/master/README.md 'You should patch the tools/sdk/ld/esp32.peripherals.ld to avoid random pixel noise on the display as following'

also present here http://gitlab.studiopresent.rs/ESP32-dotlab/framework/blob/master/tools/sdk/ld/esp32.peripherals.ld

and also found it here https://github.com/mrfaptastic/ESP32-RGB64x32MatrixPanel-I2S-DMA

https://qiita.com/h_nari/items/d52734440a7256072e8e when translated by google, says " I encountered GPIO 's unintelligible behavior. After examining the cause, it turned out to be a CPU bug (Eratta)"

I'm not sure if any of those people opened a bug or tried to get it changed, but in case none of them did, I'm filing this bug.
I am not knowledgeable about why this fix is needed, although the last link does give details that should help

marcmerlin, 2019-08-01:
@projectgus @me-no-dev 
https://github.com/espressif/arduino-esp32/blob/master/tools/sdk/ld/esp32.peripherals.ld
is still showing the apparent incorrect value. Is it something that should be fixed, or the patch given by several on the net, is incorrect?

atanisoft, 2019-08-01:
@marcmerlin please open an issue against https://github.com/espressif/esp-idf/ as this file comes from their repo and it is set the same as in IDF: https://github.com/espressif/esp-idf/blob/release/v3.2/components/esp32/ld/esp32.peripherals.ld#L4

closed 2019-10-14 by stale[bot]