espressif/arduino-esp32 #2573: failed to upload error

crunchysteve, 2019-03-12:
new esp32 dev, installed via boards manager, Arduino IDE 1.8.8, Mac OS Sierra, following error lines...
"Arduino: 1.8.8 (Mac OS X), Board: "ESP32 Dev Module, Disabled, Default, 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 921600, None"

Sketch uses 192048 bytes (14%) of program storage space. Maximum is 1310720 bytes.
Global variables use 12816 bytes (3%) of dynamic memory, leaving 314864 bytes for local variables. Maximum is 327680 bytes.
Traceback (most recent call last):
esptool.py v2.6-beta1
  File "esptool.py", line 2865, in <module>
Serial port /dev/cu.usbmodem1421
  File "esptool.py", line 2858, in _main
  File "esptool.py", line 2565, in main
  File "esptool.py", line 213, in __init__
  File "serial/__init__.py", line 88, in serial_for_url
  File "serial/serialposix.py", line 268, in open
serial.serialutil.SerialException: [Errno 2] could not open port /dev/cu.usbmodem1421: [Errno 2] No such file or directory: '/dev/cu.usbmodem1421'
Failed to execute script esptool
An error occurred while uploading the sketch

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences."

Any idea what's wrong?

closed 2019-08-15 by stale[bot]

andybrunner91, 2021-02-01:
I had this issue and it was because the port was not set in my Arduino IDE.

I fixed it by going to Tools -> Port, and selecting my usb device. In my case it was called /dev/cu.SLAB_USBtoUART

carmuno, 2023-05-18:
@andybrunner91 yes, its correct. thanks bro